canboat/canboat #259: PGN:130820 Yet Another Way To Encode Strings In N2K

dbeckwith01, 2021-10-01:
I'm trying to reverse engineer a digital switching platform controller from BEP Marine.  I have been able to dump the following PGN which seems to be used to provide the names of the configured circuits to populate display menus.

Both of these strings seem to be standard C null terminated strings with no length data unfortunately which doesn't seem to fit any of the existing canboat string formats if I'm not mistaken.

I have come up with the following PGN example but it doesn't allow me to read the bytes after the string for the first example.  

Am I correct in assuming that another form of string handling needs to be implemented?

```
  {"BEP: String Message",
     130820,
     PACKET_INCOMPLETE,
     PACKET_FAST,
     0x2E,
     0,
     {{"Manufacturer Code", 11, RES_MANUFACTURER, false, "=295", "BEP Marine"},
      {"Reserved", 2, RES_NOTUSED, false, 0, ""},
      {"Industry Code", 3, RES_LOOKUP, false, "=4", "Marine Industry"},
      {"DIP Switch", BYTES(1), 1, false, 0, ""},
      {"Counter", BYTES(1), 1, false, 0, ""},
      {"String Code", BYTES(1), RES_LOOKUP, false, "=128", "Request"},
      {"Circuit Name", BYTES(33), RES_ASCII, false, 0, ""},
      {"Circuit Type", BYTES(1), RES_LOOKUP, false, "0=Stan,1=Mode1,2=Mode2, 3=Mode 3,4=Mode4", ""},
      {"Reserved", BYTES(1), RES_NOTUSED, false, 0, ""},
      {"Reserved", BYTES(1), RES_NOTUSED, false, 0, ""},
      {"Reserved", BYTES(1), RES_NOTUSED, false, 0, ""},
      {"Reserved", BYTES(1), RES_NOTUSED, false, 0, ""},
      {"Reserved", BYTES(1), RES_NOTUSED, false, 0, ""},
      {"Reserved", BYTES(1), RES_NOTUSED, false, 0, ""},
      {"Reserved", BYTES(1), RES_NOTUSED, false, 0, ""},
      {0}}},
```

The first has example below has the circuit name of Circuit 2.
```
2021-09-29-22:47:47.919,7,130820,2,255,8,40,18,27,99,01,00,80,43
2021-09-29-22:47:47.919,7,130820,2,255,8,41,49,52,43,55,49,54,20
2021-09-29-22:47:47.920,7,130820,2,255,8,42,32,00,00,00,00,00,00
2021-09-29-22:47:47.920,7,130820,2,255,8,43,00,20,00,00,ff,ff,ff
```

For this example I used the BEP programming tool to try and set the longest string which seems to be 32 characters in length. 
```
2021-09-30-23:19:04.685,7,130820,2,255,8,a0,2f,27,99,01,01,80,31
2021-09-30-23:19:04.686,7,130820,2,255,8,a1,32,33,34,35,36,37,38
2021-09-30-23:19:04.686,7,130820,2,255,8,a2,39,30,31,32,33,34,35
2021-09-30-23:19:04.687,7,130820,2,255,8,a3,36,37,38,39,30,31,32
2021-09-30-23:19:04.687,7,130820,2,255,8,a4,33,34,35,36,37,38,39
2021-09-30-23:19:04.688,7,130820,2,255,8,a5,30,31,32,00,01,00,00
2021-09-30-23:19:04.688,7,130820,2,255,8,a6,00,00,00,20,00,00,ff
```

keesverruijt, 2021-10-04:
Interesting. It certainly looks like there is no lead byte.

I am actually not that sure that the current definition of RES_STRING is OK. The RES_STRINGLZ and RES_STRINGLAU are okay (and clearly not this!)

For now I am OK with you adding a RES_STRINGZ field that supports your current belief in that the string is ASCII and zero terminated.

webmasterkai, 2021-12-18:
What BEP programming tool are you using to program what products? Are you changing individual circuit names on the network or are you saving a complete configuration to the network?

dbeckwith01, 2022-12-09:
@webmasterkai Sorry for the delay, I'm using the NAVICO CAN to USB  Converter.  The part number is 000-00020-001.  I was using the CZone software and saving an entire configuration to the device.

patg, 2024-03-17:
Hi Darren

Have you made any progress in decoding(RE) the BEP CZONE PGNs ?
Did some work a two years ago to integrate CZONE support in canboat and build a
signalk-extension to control the circuits and especially dimming, but ended with too much
unknown and loose ends. Parsing the CZone-cfg would also be a nice thing, but out of scope
here.