canboat/canboat #265: Proposal : Move pgn definitions from .h file to Kaitai Struct

TwoClocks, 2021-12-18:
First off, I'd like to say that I've been using this project and its data for years. I am very appreciative of the time and energy that has gone into creating and maintaining it.

For those unfamiliar, [Kaitai Struct](https://kaitai.io/) is a  declarative language for parsing binary data. It can output parsing code in 15-ish different programming languages. It's used fairly extensively for network protocols and file formats. You can see a list of published parsers here : [https://formats.kaitai.io/](https://formats.kaitai.io/)

I have taken a quick stab at auto-converting CANBoat's JSON data to a kaitai struct's `.ksy` file format. Here is one pgn in the format: [gnss_position_data](https://gist.github.com/TwoClocks/d6c443d7a33b502447d249b3e56cfa3b).
I generated this programmatically. I'm sure my JSON converted isn't complete and doesn't cover all the types correctly. I just wanted to do this as an experiment.

I also created (by-hand) a .ksy file describing yacht devices' [Voyage Recorder file format](https://gist.github.com/TwoClocks/dfacac137a9ac72d9f7523985d729935). 

With these two files and the kaitai struct compiler, I could parse a yatchd data file, and pull out all the position messages. I did it in both java and python w/ no changes to the `.ksy` files, and identical output. Each version ended up being a few lines of code (not counting all the auto-generated code). Here is a kotlin app that runs [though a yachd file](https://gist.github.com/TwoClocks/9c0892480005fb9e18df0813f829fb41).

As a bonus, it can also output graphviz docs for a `.ksy` file : [gnss-image](https://user-images.githubusercontent.com/5883156/146653681-398cca6d-2cf5-4bd6-a15d-2323a654a5f7.png)

Kaitai Struct also has a great [on-line IDE](https://ide.kaitai.io/) that's very helpful for reverse engineering data. That is what I used to create the voyage recorder parser. I was inspired by this article on using the IDE to reverse engineer : [https://gettocode.com/2017/09/18/reverse-engineering-with-kaitai-struct/](https://gettocode.com/2017/09/18/reverse-engineering-with-kaitai-struct/)

So, with all that being said : 

I propose that we permanently move the `pgn.h` to .ksy files, and retire `pgn.h`. It should be easy to port `analyzer` (and the rest of the tools in this project) to use generated C files from the kaitai struct compiler instead of `pgn.h`

We'd then have a source for pgn data that is easily accessible from every major programing language. We'd also have a source format that's easier to follow than the existing `pgn.h` file. Perhaps more people could contribute pgn definitions. As it's easier to reverse engineer pgn's w/ some sample data and the IDE. No need to understand the `analyser` code. To be fair, anyone that's capable of reverse engineering a pgn can likely understand the C code in `analyzer` quire easily.

I understand this is duplicative of the work that's already been done in the caboatjs project. 

I'm happy to contribute to this, but I don't want to do it without your help/support/blessing. Right now `pgn.h` is the defacto source of open source pgn data. I have no desire to create a competing "source of truth".

I think we can programmatically generate more of the `.ksy` from `pgn.h`, but the .h file has some great resources in the comments, so those would need to be moved over by hand.

Or if the desire to replace `pgn.h` isn't there, perhaps `analyzer` could output `.ksy` format as well as JSON/XML? Although, I think `.ksy` is a better "source of truth" than `pgn.h`. For example : It would be easy to support the new string format referenced in #259 by creating a new type in a `.ksy` than writing code (then porting it to various languages).

thoughts?

webmasterkai, 2021-12-18:
Does Kaitai read/write to the DBC Format? https://github.com/canboat/canboat/issues/233 https://github.com/canboat/canboat/pull/257 https://www.csselectronics.com/pages/can-dbc-file-database-intro http://socialledge.com/sjsu/index.php/DBC_Format

DBC is common when dealing with CAN messages. There has been some work to port to it in the above issue and PR. It's used by some automotive projects like openpilot and openvehicles.
https://cabana.comma.ai/?demo=1
https://github.com/commaai/opendbc
https://github.com/GENIVI/CANdevStudio
https://docs.google.com/document/d/1ik4rbaMyAFXXszg3OeMeZ5QGxne66nFdd2u_GT8W8kE/edit

The other format I've seen being used is https://github.com/julietkilo/kcd it also works with https://github.com/cantools/cantools

In terms of pulling definitions out of pgn.h and into a text format of some kind, that's a good idea. The issue I have is often these formats are specific to a messages/sentence. Where each PGN has its own file that decodes every field. This ends up creating a lot of duplication when it's not necessary.

Ultimately a PGN is a collection of values. Each value has a DataField definition. A "Large Time Interval" or "Fixed String" is a DataField that is used in many different PGNs. Ideally it is defined once and referenced by the PGN that uses it. If anything, the source of truth could be in a relational database format that is then machine exported into all the rest since defining a data type once and reusing it makes a fix applied everywhere it is used much easier. It could be the DataField (Signed Distance) is referenced from DataDefinition (Distance, XTE, Geoidal Separation) that are then referenced on the PgnField that is referenced on the PGN. Or something like that... 

That's kinda what #259 is getting at. There are different ways to decode strings. A specific string decode definition can be reused in many places.

TwoClocks, 2021-12-19:
Thanks for the response.

I looked at the links you sent. It seems like they fall into two camps (please correct me if I misunderstood).

1) An intermediate representaion. Basically, another way to store CAN data. That seems like what DBC. It isn't the pgn's as they appear on the wire. It's those translated to another representation for easier use. So it's easier to analyze after the fact.

2)) A format that describes how CAN PGNs look on the wire. This seems like what KCD is.

Kaitai is in the latter camp. It describes the bits as you would expect to see them on the wire. Or in a file if they are stored in the same format as seen on the wire. Like yacht devices' recorder does.

The main benefit I see of kaitai over KCD is that it generates the code to parse the format you've described in different languages. You don't have to write the code that parses the XML, and generates a parser. All that work is done for you.

KCD looks like just another way to describe CAN PGNs (AFAICT). You would still need to write the code that from the KCD XML file to parse bits on the wire. I'm not sure that's better than the XML or JSON files `analyzer` generates now. Just different, and maybe more "standard" but not more "useful".

Yes, you can define 'types' in kaitai, then refer to those types when parsing. So you would define the type for "Large Time Interval" once, then expect that type in a specific PGN. So very similar to the way the `pgn.h` works now.  So a PGN becomes a list of what types you expect in what order. And types can contain other types, so if there is a common set that are always together you can make a type of that set. etc..

So yeah, you would make a new type for #259. and it would appear in whatever PGN's that use it. Or if different PGNs use different types by different vendors, as long as you can detect it programmatically, you can parse the correct type.

Kaitai does not generate "encoders" though, only "decoders". So it can only "deserialize" the format, not serialize it. So you could not use it to automate DBC generation. But to the extent that DBC follows the same types as PGNs creating a DBC writer from a kaitai source would not be hard.

You could also write a `.ksy` to parse the DBC format. Although that format looks much simpler than PGN as they are on the wire (which is likely the point of DBC).

keesverruijt, 2021-12-19:
I am the first to acknowledge that having the "truth" reside in a C header file is not the optimal solution. It was the easiest for me to to start, and I like being able to develop on-board on a simple UNIX system that has only a few dependencies installed, without continuous internet access. Moving away from that into a situation where the truth resides in some better format is only acceptable to me if this brings substantial advantages.

One such advantage could be that it is easier to understand or to generate an easy-to-understand format _that is complete_. The latter part, to me, is the killer feature. It has to describe the N2K PGNs in full, with all intricacies thereof.

The problem with N2K is that it is a binary format that is extended on-the-fly in many different ways, unfortunately. It would have been so much easier if a proper encoding standard such as ASN.1 would have been used. But alas, this is not the case. So we have all the warts and warps that there are, and our describing mechanism should be able to handle these.

A non-exhaustive list of problems (sorry, challenges) with describing PGNs:

1. Same "company specific" PGN used by multiple companies for different purposes. The standard handles this by having a 2 byte prefix that indicates the Manufacturer and Industry code.
2. PGNs having "function code" fields that define the format of the rest of the PGN. Example: PGN 126208 - Group function.
3. Variable length fields. 
4. Repeating fields, usually the last _n_ fields with _n_ a PGN specific value.
5. Repeating fields, followed by fixed fields. Sigh. (even the C code doesn't handle this yet!) Example: PGN 130571 - Library Data Group

Another peculiarity in N2K is that they use an extensible list of data types. Unfortunately, we can only reverse-engineer these after the fact. So for instance, a field might contain an _angle_ of rotation relative to some 0 point. For instance, heading. Such angle data is generally represented in a two byte fixed point notation with a quantisation of 0.0001 radian. I think a new format should make this explicit and state something as "Angle, subtype 1". We need subtypes as there are different resolutions (quantisations) for several types, such as rotation. Sort of like how there seem to be at least 4 four string formats!

My worry is that an "implicit" language such as YAML will not be the most suited to do this. Although XML is a pain in the ass, at least it is fully extendable and fully describing of the data that it contains.

The only "plus" points going for YAML/KSY is that it is the "ml of current vogue" and that KSY has generators. Whoopee. But those generators can also be used if we include an XML-to-KSY converter. 

So, yes, an XML-to-KSY converter would be useful. But I don't see how KSY adds value or is better than -- for instance -- XML.

TwoClocks, 2021-12-19:
I don't think the textual format matters much. XML, or JSON or YAML. Any of them have the capability to contain enough information to parse PGNs. In someways a `.h` is more preferable because it's already parsed into an "AST-like" form. So if the question you are asking is "why is YAML better than XML?", my answer is "it isn't".

So let's just assume KSY can represent PGN data "well" (I'll get back to this at the end).

To me, the reason to use `.ksy` are these 

1) It's made specifically to describe binary formats.
2) It outputs parsing code in 15 different languages. So someone who wants to put a raspberry PI on their N2K network can write in python/lua/nium/rust/whatever w/o having to write a parser.
3) It has tools made specifically to help reverse enginner binary formats, and a nice IDE to do so.
4) it has a compiler that exposes the AST. So even if you want to generate your own parser, it's a much better place to start from than a textual file.

You get most of the bang for the buck from a XML-to-KSY converter, as you point out. And if the .h file, and resulting XML contained all the information, and had all the abstractions needed to parse PGNs, then that's perfect. But you say yourself that the code doesn't contain abstractions for certain things (fixed after repeating groups). So if I want to parse a PGN that has fixed-after-repeating, I'd need to take the following steps
1) Reverse engineer the PGN by hand (no nice GUi tools to help)
2) Alter/add a new abstraction to the .h file and C code. For adding a new PGN this is pretty easy. For adding a new "type" that the C code doesn't currently understand, this is a chunk of work.
3) Make sure that abstraction is accurately represented in the XML output.
4) Alter the XML-to-KSY converter for the new abstraction
5) re-compile KSY to JavaScript.

Whereas is the "source of truth" was a KSY file I would do the following
1) Reverenginer the PGN w/ tools made specifically for doing so.
2) Add a new type/PGN to the KSY file.
3) re-compile KSY to JavaScript.

Of course, this is all predicated on the idea that KSY can accurately represent PGN data. I've used KSY for a bunch of different formats, including ones that have relative offsets, repeating groups, and other craziness. It's all worked fine.

But seeing is believing. Why don't you pick 3-5 PGNs. Ideally, PGNs you have some sample data for, and I'll write the KSY files for them and we can take a look at the results. You can also look at the generated C code, and see if it meet your criteria for embedded work. Feel free to pick a PGN the current C code doesn't support completely support, or that are "problematic" in some other way.

If you don't like the results, I'll go away and quit bothering you.

tkurki, 2021-12-26:
I tried to get my hands a little dirty with Kaitai. One sort of major problem I see is that there is no built in stream processing - unless I am missing something? I created a toy example and the generated code has just a synchronous _parse() function. So parsing a file like your example works, but parsing an endless stream of data will not work out of the box.

Naturally this can be bypassed with a hand crafted outer parser, that passes complete pgns to Kaitai generated parser, but as I said it will not work out of the box.

@TwoClocks your original idea of going all in with Kaitai seems a little drastic at this point. While you provided very illustrative examples none of them were plug and play, end to end working code.

I think a logical next step would be a conversion from XML or JSON to ksy, with parser generation sprinkled on top. In practise that could be a PR to this repo that adds some targets to the Makefile:
1. `generate-ksy`
1. `generate-parsers` //my main interest is in Signal K / JavaScript
1.  a command to read binary n2k data from stdin and output parsed data to stdout
1.  same as previous, but for actisense-serial data, see [sample](https://github.com/SignalK/signalk-server/blob/master/samples/aava-n2k.data)

The conversion need not be perfect at this stage. It sounded like you already had some of that done? I can help with (3) and (4) in js land.

I like the idea of web ide: being able to grab some captured data and quickly iterate on it is enticing. We could add a way to easily capture some sample data to Signal K server and a pointer to the IDE, including the base n2k ksy definition. The SK server could even support changing the ksy on the fly and/or iterating in the browser server admin webapp.

There's a steep learning curve though: errors in the ksy file throw very unhelpful exceptions and just getting started is a bit daunting. At least it is/was for me.

tkurki, 2021-12-26:
Forgot to include my main point: if we make Kaitai available we can gain collective experience with that and see where things go. There's more to the world than C and js, Kaitai would cover for example golang and rust.

TwoClocks, 2021-12-26:
Re : Streams - Not sure which language you were using. On the JVM you can impl the `KaitaiStream` interface (abstract class), and read from a network, or USB port, or whatever.  One thing : Kaitai doesn't check if there is enough data available to parse the msg. It will just fail if it runs out of data mid-message. So likely anything reading "live" data would need to make sure it had a complete message before parsing. And you likely want a different `.kSY` definition for the framing protocol, like I did w/ the yachtd `.dat` file.

Yes, I agree that moving to Kaitai as the conical format is a drastic step at this point. I just wanted to plant the seed that it might be the logical outcome at some point. Having '.KSY` as a new output format to join json and xml is a logical first step.

I'm happy to contribute to the effort. I do have a XML -> KSY converter now that understand a few different types but by no means all.

My question would be : Do you want a XML -> KSY converter, or would you prefer the current C code to emit KSY directly? If you do want an XML -> KSY converter, what is your preferred language? I did mine in scala, but it's because I was interested in learning scala at the time. I assume you don't want to add a JVM dependency to the current project. I can port it to whatever you like as a starting point. I see there is already some python in the project, that might be a good option.

tkurki, 2021-12-26:
> you can impl the KaitaiStream interface (abstract class), and read from a network, or USB port, or whatever. One thing : Kaitai doesn't check if there is enough data available to parse the msg. It [...]

Well, this was kind of my point. Creating a stream that reads from a network or USB port is easy enough, but as the stream never ends Kaitai parser never finishes and afai could find there is no streaming api where you would get access to the parsed data as messages arrive and are parsed. So we need a wrapper that constructs complete PGNs to pass to a Kaitai generated parser.

As Kees stated earlier less is more in terms of dependencies. Even [xsltproc](https://github.com/canboat/canboat/blob/master/analyzer/Makefile#L38) that does the XML to JSON conversion is fading to obscurity. But I'll let @keesverruijt  take the lead here.

keesverruijt, 2021-12-26:
At the moment I _think_ the XML contains all "knowledge" so you might as well start from XML. If it turns out there is some hidden meaning then adding it to the XML shouldn't be the issue.

Python seems a logical choice as it is well known and used a lot. I am not a big fan but I don't see a better alternative.

I may start redesigning some of the C definitions soonish now that we have a good idea of the PGN structures and in particular the data fields. I never changed it from what I had when I understood only 4-5 PGNs.

aldas, 2021-12-26:
just one note here about DBC.  There are plenty of DBC lexers/parsers out there as @webmasterkai  mentioned.  Some of those I have bookmarked when I researched this stuff to load PGNs though DBC file with Go.
* https://github.com/christoph2/pydbc/blob/master/pydbc/dbc.g4 (ANTLRv4)
* https://github.com/eerimoq/cantools/blob/master/cantools/database/can/formats/dbc.py
* https://github.com/msalau/dbc-parser (yacc)
* https://github.com/GENIVI/CANdb/blob/master/src/dbc_grammar.peg (peg)
* https://github.com/msalau/dbc-parser/blob/master/lemon/dbc_parser.lemon (lemon)

having so many already existing lexers/parsers is very nice as you can port DBC database to many languages as you can generate code to read DBC format.

----

> * Same "company specific" PGN used by multiple companies for different purposes. The standard handles this by having a 2 byte prefix that indicates the Manufacturer and Industry code.
> * PGNs having "function code" fields that define the format of the rest of the PGN. Example: PGN 126208 - Group function.
> * Variable length fields.
> * Repeating fields, usually the last n fields with n a PGN specific value.
> * Repeating fields, followed by fixed fields. Sigh. (even the C code doesn't handle this yet!) Example: PGN 130571 - Library Data Group

I do not know if all those concerns could be handled in DBC tools.  But some are probably handled but with different terminology. 

In DBC terminology "PGN field" is called signal. Each CAN ID (=PGN) can have list of enum values for signals.

About NMEA2000 and PGN 126208 multipacket PGNs. I think in CAN world they use term "multiframe message" and J1939 world they use term multipacket message. Here is someone describing how to define these messages with popular  DBC tool https://www.youtube.com/watch?v=rKy5RfjrFZo  .  Also this site has quite good DBC description and has block ("Multiplexed signals (OBD2 DBC example)") which describes multipacket stuff https://www.csselectronics.com/pages/can-dbc-file-database-intro

Well I'm no guru of N2k but it is pretty much same as J1939 and this multipacket handling is handled by those automobile related DBC tools. then there should be fine with marine stuff also.

anyway. that list is good start for functional requirements for describing PNGs.  

Talking from my own experience of creating N2k library for Go is that current Canboat JSON format is technically very easy to use (all modern languages can marshal JSON) but knowledge how to decipher all those fields is quite complex task. As to understand some field you need to search canboat c code where those fields are used and understand WHY  ala `RepeatingFieldSet1` and `RepeatingFieldSet2` or  what can `Field.resolution` contain and how to interpret those values - this field meaning is extremely overloaded with different things.   
I think most impactful would be to document XML or JSON schema. Maybe even with background info why some stuff looks a little bit weird.

TwoClocks, 2021-12-26:
I'm not a huge fan of python either. But it seems convenient for this.

When I did my simple XML -> KSY parser, I had to go look at the C code to figure out how to parse some types, as @aldas points out. I think that's fine for now.

I think the XML has everything in it, but it's hard to get at sometimes. For example, in my toy version, I noticed that enums w/ the same name are defined multiple times in the XML file. Sometimes they are the same definition, sometimes they aren't, and sometimes they aren't defined at all. So in my demo I gave up making enums re-usable types, and just defined them again for every PGN that used them, and post-pended the PGN # to the name. Which was fine at the time.

But now I'd like to produce a more polished KSY file. It'd be nice to know which enums can use the same definition, or which ones are unique.
So I can do it the hard way in python, which is to do a 2-pass thing. Get all the enum defs, and find all the ones that are identical, then alter the names of the ones that are unique.

There are also PGN's that have the same # that are defined multiple times in the XML file. It's unclear to me which one to use, or how to programmatically determine which one to use at runtime. I assume that magic is in the C code someplace.

So, I'm happy to "do it the hard way" and see where we get. Unless @keesverruijt s' pending C PGN-reDef rework is going to address these types of issues? In which case should I hold off a bit?

aldas, 2021-12-27:
Those "PGNs defined multiple times" are multipacket ones where content of packet actually defines its meaning. This is similar to NMEA0183 MTK and AIS (!AIVDM/!AIVDO) packets which is different protocol embedded into same NMEA0183 prefix. 

I have done NMEA0183 stuff lately and from that I have opinion that in NMEA world you probably can not have "classical" enum - fixed set of values. It seems to be more like "list of known values" and assumption that there is always some manufacturer device that could send value that you have not added your enum list or at some situation value is left empty (can happen by two consecutive `,,` delimiters). From that I deduce that very few people actually read/buy NMEA* standards and most of  developers pretty much "improvise" their interpretation.

This is too much to ask but something similar which DBC signals and enumeration types for them (http://socialledge.com/sjsu/index.php/DBC_Format#Enumeration_Types) support would make importing PGNs easier. 
> An enumeration type is used where the user wishes to use or see names, instead of numbers. For example, instead of a state machine showing up as "0, 1, 2", we could see it as "stopped, running, paused". It is accomplished by adding two new lines in the DBC file. A "BA_" field needs to be added, and for the sake of simplicity, you can follow the example below to list an enumeration as a "FieldType" first. Then, you need a "VAL_" field that actually defines the enumeration values.
```
BO_ 500 IO_DEBUG: 4 IO
 SG_ IO_DEBUG_test_enum : 8|8@1+ (1,0) [0|0] "" DBG

BA_ "FieldType" SG_ 500 IO_DEBUG_test_enum "IO_DEBUG_test_enum";

VAL_ 500 IO_DEBUG_test_enum 2 "IO_DEBUG_test2_enum_two" 1 "IO_DEBUG_test2_enum_one" ;
```

In XML would it probably means that you need XSD to define enums
```xml
 <xs:simpleType name="color" final="restriction" >
    <xs:restriction base="xs:string">
        <xs:enumeration value="green" />
        <xs:enumeration value="red" />
        <xs:enumeration value="blue" />
    </xs:restriction>
</xs:simpleType>
<xs:element name="SomeElement">
    <xs:complexType>
        <xs:sequence>
            <xs:element name="Color" type="color" />
        </xs:sequence>
    </xs:complexType>
</xs:element>
```

In JSON world you have JSONSchema and its enums https://json-schema.org/understanding-json-schema/reference/generic.html?highlight=enum#enumerated-values

keesverruijt, 2021-12-28:
> I have done NMEA0183 stuff lately and from that I have opinion that in NMEA world you probably can not have "classical" enum - fixed set of values. It seems to be more like "list of known values" [...]

This is not true in NMEA2000. All enums are properly defined by the NMEA. The on-wire data only contains the identifier, never a name. The problem is that the standard is closed, and cannot be openly discussed or disseminated, so we don't know the exact definition. If I bought the standard I could not write this software as that counts as dissemination.

> About NMEA2000 and PGN 126208 multipacket PGNs. I think in CAN world they use term "multiframe message" and J1939 world they use term multipacket message. 

There are two different implementations of messages with longer-than-8-data-bytes. I don't know how J1939 and most other CAN protocols do this as I haven't delved into that very deep (unfortunately, CAN and J1939 are also "closed" and "non free"). What I _do_ know is that the NMEA2000 "fast packet" method is completely different. Fast packet is only a data level definition, there is nothing in the underlying CAN standard or CAN hardware to facilitate this. It uses the first 2 data bytes in the first packet and the first data byte of the remaining packets to encode such Fast PGNs on the wire. 

Thus one problem with "fast packet" is that DBC doesn't handle this neatly as it is not aware of this 'clever' way of transmitting.

keesverruijt, 2021-12-28:
> So, I'm happy to "do it the hard way" and see where we get. Unless @keesverruijt s' pending C PGN-reDef rework is going to address these types of issues? In which case should I hold off a bit?

Yes, you should hold off. I am going to enrich the XML with proper datatypes and proper unique enums.

aldas, 2021-12-28:
when NMEA0183 enums are characters (ala "A", "V" etc) then in N2k those are enums represented in integers. in that sense char  vs uint8/byte there is no difference. Whether it is 'A'/'V' or 1/0 makes little difference when you plug device into your network and that manufactures has "augmented" their device with additional value 'C' or 2.  Maybe this happens less in N2K but in NMEA0183 it seems to happen (although rarely)

Isnt NMEA2000 "Fast packet" same as CAN-bus "ISO-TP" https://en.wikipedia.org/wiki/ISO_15765-2 or simpler explanation https://piembsystech.com/can-tp-protocol/?

If I am not mistaken then older CANbus spec was published by Bosch and anyway - CAN and J1939 are very well documented/reverse engineered and have huge amount of libraries because of all those cars and even electric bikes today work with CANbus. Some J1939 docs have very good quality https://assets.vector.com/cms/content/know-how/_application-notes/AN-ION-1-3100_Introduction_to_J1939.pdf

Linux kernel has quite extensive docs for
* CANbus driver https://www.kernel.org/doc/html/latest/networking/can.html (supports ISO-TP)
*  J1939 driver https://www.kernel.org/doc/html/latest/networking/j1939.html (does not support fast packet)

keesverruijt, 2021-12-28:
> when NMEA0183 enums are characters (ala "A", "V" etc) then in N2k those are enums represented in integers. in that sense char vs uint8/byte there is no difference. Whether it is 'A'/'V' or 1/0 makes [...]

Bit values are usually encoded in minimal bit numbers in N2K, for instance 1 or 2 bits. As soon as more than 1 bit is used, the values "all ones", "all ones minus one" and sometimes "all ones minus two" are common special values. "all ones" means "no value", "all ones minus one" means "error". 

For instance, look at the current definition of PGN 127347 "Heading Track control" field "Rudder Limit Exceeded" which in the current definition is: 

     {"Rudder Limit Exceeded", 2, RES_LOOKUP, false, LOOKUP_YES_NO, ""},

This results in the following enum:

  * 0 = No
  * 1 = Yes
  * 2 = Error
  * 3 = Unknown

> Isnt NMEA2000 "Fast packet" same as CAN-bus "ISO-TP" https://en.wikipedia.org/wiki/ISO_15765-2 or simpler explanation https://piembsystech.com/can-tp-protocol/?

*NO*, unfortunately. Fast Frame is a simplified but different version of it. And precisely that is the problem with commercial non-N2K analysers since they do not understand Fast Frame.

aldas, 2021-12-28:
> It uses the first 2 data bytes in the first packet and the first data byte of the remaining packets to encode such Fast PGNs on the wire. Thus one problem with "fast packet" is that DBC doesn't handle this neatly as it is not aware of this 'clever' way of transmitting.

I think DBC with extended multiplexing is that magic. 

Taken from https://www.csselectronics.com/pages/can-dbc-file-database-intro (see block named "Multiplexed signals (OBD2 DBC example)")

> ... But CAN DBC files also support extended multiplexing, where a multiplexed signal (in this case S1) can also be a multiplexor and thus control the interpretation of other parts of the data payload. To see this, note that S1 takes the same role as Service in the syntax, adding an M after m1 and being grouped with the two OBD2 PIDs, speed and throttle position.

> Extended multiplexing works as before: If S1 takes the value 13 (HEX 0D), it means that only signals that are A) part of the S1 group and B) have a multiplexer switch value of 13 should be taken into account. In this example, it means that byte 4 reflects data for vehicle speed. If byte 3 equals 17 (HEX 11), byte 4 reflects data for the throttle position instead.

> DBC multiplexing and extended multiplexing is an advanced topic and not supported by all data processing tools. However, you can use e.g. the CANDB++ DBC editor to more easily view and understand DBC files with multiplexing, like the OBD2 DBC:

Another explanation.
* https://assets.vector.com/cms/content/know-how/_application-notes/AN-ION-1-0521_Extended_Signal_Multiplexing.pdf (Extended Signal Multiplexing in DBC Databases)

which means that you can tie "Fast packet" first byte (packet sequence number) to group of signal (nmea fields). ie. If first byte is 0 then other bytes map to following signals. If first byte is 1 then other bytes are these signals.
I think this is how automotive industry has mentally modeled DBC to support similar stuff.

Note: fields that span multiple frames could pose problems. but when you define message (PGN) structure in DBC you provide length for that packet.
[image]
Which probably means that nothing does not stop to use 255 there (for example CAN FD frames can be up to 64 bytes, oposed to regular 8 bytes). You could define DBC with list fast packet PGNs in regards that you assume that these sequence bytes are left out and you deal only with multiplexed data and single combined packet. (low level frame combining is left out of DBC PNGs). So "fast packet" is so low level implementation detail that it left out of PGN database and only reference to "fast packet" existence would be that PGNs length is more than 8 bytes in its definition.

keesverruijt, 2021-12-28:
Can we move DBC discussions to a new issue, it is definitely interesting but no longer directly influences KaiTai?

To get back to providing data to a KaiTai parser, some of the input formats that `analyzer` receives have already lost all PGN data encapsulation, for instance when the data was received by an Actisense NGT-1.

aldas, 2021-12-28:
Sorry for going off topic. Kaitai is more of a question of creating protocol parser definition similar to what ANTLR/Ragel/yacc/bison achieve from which you generate code you your preferred language. DBC/XML/JSON is more of documenting/specification of messages (PGNs) in a machine/tool readable/editable way and nothing about how you implement protocol parsing. I was fixated more of documentation part as implementation is always subjective as sometimes people do not want to be limited by generated code (Not Invented Here syndrome).

keesverruijt, 2021-12-28:
Please look at issue #266 branch [refactor-definitions](/canboat/canboat/tree/refactor-definitions), and note any comments on issue #266.

TwoClocks, 2022-10-05:
oh yeah. This is a massive improvement.
I think I can convert `docs/canboat.xml` fairly easily. I'll take a stab at a python converter.

closed 2022-10-17 by keesverruijt