canboat/canboat #266: Refactor data definitions

keesverruijt, 2021-12-28:
Refactor the data definitions for the PGNs so that:

* The C code becomes easier to maintain and modify.
* The XML data contains structured data so that it is easier to generate other specification languages from it. This then becomes useful for #265 .
* The existing XML and JSON code is not impacted other than new keys are added.

In particular, all regularity that there is to the datatypes possible in PGN fields should be exposed and exploited.

This will include a split of the current `analyzer` executable into an `analyzer` and `analyzer-generate-xml` program.

aldas, 2021-12-28:
Have you considered in addition to have documentation page like GPSD has https://gpsd.gitlab.io/gpsd/NMEA.html ? 
I'm asking this because GPSd page has useful notes along with sentences that help you understand the context. And it seems to be somewhat goto place where to look NMEA0183 related info.

ala some markdown file in repo or in wiki part that contains more human readable form of PGNs / types / resolutions. Although there are atm ~343 PGNs (247 unique) which would be monumental task to write down initially all by hand (unless generated to readable blocks or maybe Doxygen can do something) ... and probably hard to be disciplined enough to keep up to date.

I could help if needed.

keesverruijt, 2021-12-28:
Well, `analyzer -explain` goes a long way. Making this generate markdown, which is readable by humans, is a good plan.

You could help out once #266 is fixed by enhancing the `analyzer-explain -explain` code.

TwoClocks, 2021-12-29:
This is much easier to work with. thanks!
I will give you more feedback as I work with it. Do you want problems I find here in this thread or open issues?

Is the intent to be backward compatible? That's why enums show up both in the enumerations section and in individual PGNs?

TwoClocks, 2021-12-29:
A question. From the XML how would I tell the difference between
`field->resolution == RES_STRINGLZ` and `field->resolution == RES_STRINGLAU` and `field->resolution == RES_STRING` They need to be parsed differently, I think.

Ideally, I'd just like the raw value of `field->resolution`. Or the `#define` name. Hard to do the latter w/o reflection though. And maybe `field->units` as well. I think I need to parse them to generate the KSY.

aldas, 2021-12-29:
It would be useful if resolution would be also enum - so you can detect if some type exists or even has been added between now and last time you fetched xml/json. 

It would be very helpful if all those schema fields could be explained also - like what does `<Field><Match>` mean or `RepeatingFields`, `RepeatingFieldSet1`, `RepeatingFieldSet2`  or `<Missing>` and its contents? Separate XSD which defines the schema and has comments/descriptions for fields ala
```xml
<xsd:element name="Complete">
  <xsd:annotation>
    <xsd:documentation xml:lang="en">
         Is PGN a fast packet or not
    </xsd:documentation>
  </xsd:annotation>
```

also Resolution being used for type and scaling factor is one of those confusing things. in xml/json there is `<Type>` and `<Resolution>` which is easy to grasp  but "negatives are special types and positives are integers with scaling factor" is a little bit different story.  

Just some observations when I had to search in code what all those types do and what their implementations are.

keesverruijt, 2021-12-29:
Good points. Keep them coming. Unfortunately I have to keep the XML backwards compatible, so we will have to add extra keys instead of changing the old ones, and this is why the enums are present at leaf level -- just ignore them.

keesverruijt, 2021-12-29:
Instead of passing through the current "RES_" data, I intend to first put _everything_ in a straight jacket of a well defined data-type. So all fields get a datatype="XYZ", with a separate list of datatypes. This is (hum) some work so will take a while :-)

commit c9e7263e (2022-09-06): Fix #266
closed 2022-09-06 by keesverruijt