canboat/canboat #267: PGN 129793 base station MMSI is missing 00 prefix

tkurki, 2022-01-01:
Base station ids have 00 prefix but it is absent in the analyzer output:

```
echo '2019-01-25T19:00:00.326Z,7,129793,43,255,26,04,77,43,25,00,bd,73,b1,02,ae,39,ef,1e,03,f0,db,c4,28,c9,c2,00,01,46,10,00,fc' | ../../canboat/rel/darwin-arm64/analyzer -json
{"timestamp":"2019-01-25T19:00:00.326Z","prio":7,"src":43,"dst":255,"pgn":129793,"description":"AIS UTC and Date Report","fields":{"Message ID":4,"Repeat Indicator":"Initial","User ID":2442103,"Longitude": 4.5183933,"Latitude":51.8994350,"Position Accuracy":"High","RAIM":"in use", "Position Time": "18:59:59","Communication State":"49865","AIS Transceiver information":"Channel A VDL reception","Position Date":"2019.01.25","GNSS type":"GPS","Spare":"0"}}
```

[002442103 is port of rotterdam](https://www.e-navigation.nl/list-non-mobile-ais?field_mmsi_value_op=starts&field_mmsi_value=002442103&field_operator_value_op=contains&field_operator_value=&field_type_value_op=or&field_operation_value_op=or)

keesverruijt, 2022-03-04:
I guess this depends on your definition of how this filed is formatted, as this is currently a number field 002442103 is more wrong than 2442103, as this could theoretically be interpreted as octal value 002442103, so we would need to format this as a string instead. So this would be another breaking change....

tkurki, 2022-05-09:
Well, if it is an MMSI then in JSON I would expect it to be a string, not a number. It is defined as a nine character string with numeric characters at https://www.itu.int/dms_pubrec/itu-r/rec/m/R-REC-M.585-8-201910-I!!PDF-E.pdf (or that's how I'd interpret that in JSON).

We'll just need to live with breaking changes I think...

commit 60890247 (2022-09-28): Fix #267: Print all MMSI numbers as string, with 9 digits.
closed 2022-10-05 by keesverruijt