canboat/canboat #270: Analyzer / pgn definitions versioning

tkurki, 2022-02-20:
There have been now several cases where fields have been renamed for [one reason](https://github.com/canboat/canboat/commit/aac4d58422b18477a0bc570006e4fc18fe0f31d0) or [another](https://github.com/canboat/canboat/commit/b6bff5c77c5e33521e6b85b3e1bcc9e4c96fcfee) in analyzer, which means that the analyzer output changes. For any 3rd part, such as n2k-signalk conversion and any other code out there this means that analyzer API changes and the change is not backwards compatible. Even [fixing spelling mistakes](https://github.com/canboat/canboat/commit/8fa3fb2973b4d932b8ec1de44b86ba9c7e68aa07) is a breaking change.

Changes like these are usually communicated with [semantic versioning](). There is no clear versioning mechanism for canboat, as canboat as a whole is not really published anywhere - you just pull the latest from Github. Afaik the only publish mechanism is publishing pgns.json in npmjs.org. This is governed by [package.json](https://github.com/canboat/canboat/blob/master/analyzer/package.json) and the version there.

Now if we simply apply strict semantic versionin there we should be good, I think.

Hmm, except in Github the version is [1.4.0](https://github.com/canboat/canboat/blob/900329145e2446b85832cf085cfce9c1df271b6e/analyzer/package.json#L3) but in [npm 1.4.2](https://www.npmjs.com/package/@canboat/pgns)?

[image]

I think it would be a good idea to automate the npm publish process as a Github Action - so that you'd git tag the version and it would publish the version in package.json of that tag automatically. The tag could be just pgn definition specific, like `pgns-v1.4.3` or something.

What do you think?

sbender9, 2022-02-20:
I fixed the GitHub version issue. I forgot to push that.

sbender9, 2022-02-20:
I would love to get this versioning issue resolved. Get @canboat/pgns out of my personal GitHub.

 I'm scared of automating the publishing because of these issues. I always try to catch this stuff before I publish changes. Failed miserably this last time.

keesverruijt, 2022-03-04:
I agree we're very bad at doing this. However... I am still struggling with semantic versioning. According to semantic versioning any breaking change should be a major version increase (1.x.x -> 2.0.0) ? Do you suggest we do this for a field rename?

tkurki, 2022-03-05:
Yes. 

I think we (or at least I do) have a lot of baggage from OOH MAJOR VERSION BIG CHANGE, whereas in semantic versioning it is just a new version that is no longer backwards compatible. And it clearly communicates that _hey you may need to change your software if you are depending on this stuff_.

It does dilute things - how do you make people pay attention to a real major new version, with plenty of new features or new performance characteristics? Let's imagine you rewrite the whole canboat, but keep functionality exactly the same - _whaaat, I worked my ass off and you're saying it's just a patch version_.

You can also version the API separately - in this case analyzer output / pgn definitions and the software (or parts of it) separately. This brings its own complexity to the picture.

I suppose a concrete step forward would be to settle on semantic versioning for pgns.json in package.json  / in npm registry and embed the same version in pgns.xml, to make things explicit there also.

mbj4668, 2023-01-03:
I think it would also be very useful if the schema itself was properly versioned, and that the generated xml files has the correct version.  Currently, there is a `Version` attribute in `PGNDefinitions`, but it is `0.1` both in `pgns.xml` (old format) and `canboat.xml` (new format).

I suggest that this attribute is made required, and changed to `xs:token`, so that semantic versioning can be used.  Further, I suggest the `version` attribute is used in `xs:schema` to declare the schema version in the schema document. 

 Also, I suggest that the current version is bumped to `1.0`, and that `canboat.xml` uses `1.0`.

mbj4668, 2023-01-03:
Hmm, after some more thought, it would probably be better to replace the current `Version` attribute with an element `SchemaVersion`.   Then it can be `SchemaVersion` also in the JSON file.

The current code calls the old format "v1" and the new "2".  So perhaps use "2.0" as the current version.

If you think that this is a good idea, I can create a PR.

keesverruijt, 2023-01-03:
Ah yes, a version on the _schema_. Yes that would be useful. 

Can you refer to this in the CHANGELOG in your PR?