canboat/canboat #271: SRC is not used when writing to device

kaleitec, 2022-03-04:
When the actisense-serial is used to write on the NMEA2K device, the src of the given message is not used. It seems that the actisense-serial use in any case the src of the device, which is used to send the message. Is there the possibility to have an additional flag and the functionality to send messages with the src which is included in the message?

Especially for testing purposes it would be a great extension.

This would mean to insert the given src into the message before it is written to the device. But I can't figure out where the message need to extended to fulfil this functionality.

keesverruijt, 2022-03-04:
Sorry, this is not possible. No NMEA gateway I know of allows this, in order to stay within spec. 

If you really want to do this, you will need to use a direct CAN interface and write your own messages. Functionality to do this is currently not in the CANBoat project.

kaleitec, 2022-03-04:
Ok thanks for the answer!

closed 2022-03-04 by kaleitec