canboat/canboat #272: 130851 definition doesn't work

tkurki, 2022-04-13:
See below snippet of 130850 `Simnet: Event Command: AP command` and  130851 `Simnet: Event Reply: AP command` traffic between my B&G Vulcan 7 and my Simrad TP-32 tillerpilot.

130851 parsing fails with `PGN 130851 field Proprietary ID error`.

It looks like autopilot is just echoing the messages back as they were sent: reordering like messages after each other:
```
2022-04-12T15:08:50.875Z,2,130850,4,255,8,00,0c,41,9f,00,ff,ff,0a
2022-04-12T15:08:50.902Z,7,130851,0,255,8,00,0c,41,9f,00,ff,ff,0a

2022-04-12T15:08:50.876Z,2,130850,4,255,8,01,1c,00,ff,ff,ff,ff,ff
2022-04-12T15:08:50.903Z,7,130851,0,255,8,01,1c,00,ff,ff,ff,ff,ff

2022-04-12T15:08:50.878Z,2,130850,4,255,8,20,0c,41,9f,00,ff,ff,0a
2022-04-12T15:08:50.914Z,7,130851,0,255,8,20,0c,41,9f,00,ff,ff,0a

2022-04-12T15:08:50.879Z,2,130850,4,255,8,21,09,00,ff,ff,ff,ff,ff
2022-04-12T15:08:50.915Z,7,130851,0,255,8,21,09,00,ff,ff,ff,ff,ff
```
but 130851 differs from 130850 def in  this field: https://github.com/canboat/canboat/blob/master/analyzer/pgn.h#L6905

Any idea why? Replacing it with the one from 130850 naturally makes things work. Should it be fixed to be like 130850?

```
2022-04-12T15:08:50.875Z,2,130850,4,255,8,00,0c,41,9f,00,ff,ff,0a
2022-04-12T15:08:50.876Z,2,130850,4,255,8,01,1c,00,ff,ff,ff,ff,ff
2022-04-12T15:08:50.878Z,2,130850,4,255,8,20,0c,41,9f,00,ff,ff,0a
2022-04-12T15:08:50.879Z,2,130850,4,255,8,21,09,00,ff,ff,ff,ff,ff
2022-04-12T15:08:50.902Z,7,130851,0,255,8,00,0c,41,9f,00,ff,ff,0a
2022-04-12T15:08:50.903Z,7,130851,0,255,8,01,1c,00,ff,ff,ff,ff,ff
2022-04-12T15:08:50.914Z,7,130851,0,255,8,20,0c,41,9f,00,ff,ff,0a
2022-04-12T15:08:50.915Z,7,130851,0,255,8,21,09,00,ff,ff,ff,ff,ff
2022-04-12T15:08:51.079Z,2,130850,4,255,8,40,0c,41,9f,ff,ff,64,0a
2022-04-12T15:08:51.079Z,2,130850,4,255,8,41,2b,00,ff,00,00,00,ff
2022-04-12T15:08:57.526Z,2,130850,4,255,8,60,0c,41,9f,00,ff,ff,0a
2022-04-12T15:08:57.526Z,2,130850,4,255,8,61,1a,00,03,ae,00,ff,ff
2022-04-12T15:08:57.535Z,7,130851,0,255,8,40,0c,41,9f,00,ff,ff,0a
2022-04-12T15:08:57.536Z,7,130851,0,255,8,41,1a,00,03,ae,00,ff,ff
2022-04-12T15:08:58.798Z,2,130850,4,255,8,80,0c,41,9f,00,ff,ff,0a
2022-04-12T15:08:58.799Z,2,130850,4,255,8,81,1a,00,03,ae,00,ff,ff
2022-04-12T15:08:58.826Z,7,130851,0,255,8,60,0c,41,9f,00,ff,ff,0a
2022-04-12T15:08:58.826Z,7,130851,0,255,8,61,1a,00,03,ae,00,ff,ff
2022-04-12T15:09:00.277Z,2,130850,4,255,8,a0,0c,41,9f,00,ff,ff,0a
2022-04-12T15:09:00.277Z,2,130850,4,255,8,a1,1a,00,03,ae,00,ff,ff
2022-04-12T15:09:00.298Z,2,130850,4,255,8,c0,0c,41,9f,00,ff,ff,0a
2022-04-12T15:09:00.299Z,2,130850,4,255,8,c1,1a,00,03,ae,00,ff,ff
2022-04-12T15:09:00.301Z,7,130851,0,255,8,80,0c,41,9f,00,ff,ff,0a
2022-04-12T15:09:00.302Z,7,130851,0,255,8,81,1a,00,03,ae,00,ff,ff
2022-04-12T15:09:00.333Z,7,130851,0,255,8,a0,0c,41,9f,00,ff,ff,0a
2022-04-12T15:09:00.334Z,7,130851,0,255,8,a1,1a,00,03,ae,00,ff,ff
2022-04-12T15:09:02.447Z,2,130850,4,255,8,e0,0c,41,9f,00,ff,ff,0a
2022-04-12T15:09:02.448Z,2,130850,4,255,8,e1,1a,00,03,d1,06,ff,ff
2022-04-12T15:09:02.470Z,7,130851,0,255,8,c0,0c,41,9f,00,ff,ff,0a
2022-04-12T15:09:02.471Z,7,130851,0,255,8,c1,1a,00,03,d1,06,ff,ff
2022-04-12T15:09:04.597Z,2,130850,4,255,8,00,0c,41,9f,00,ff,ff,0a
2022-04-12T15:09:04.598Z,2,130850,4,255,8,01,1a,00,02,d1,06,ff,ff
2022-04-12T15:09:04.634Z,7,130851,0,255,8,e0,0c,41,9f,00,ff,ff,0a
2022-04-12T15:09:04.635Z,7,130851,0,255,8,e1,1a,00,02,d1,06,ff,ff
2022-04-12T15:09:06.572Z,2,130850,4,255,8,20,0c,41,9f,00,ff,ff,0a
2022-04-12T15:09:06.573Z,2,130850,4,255,8,21,1a,00,02,d1,06,ff,ff
2022-04-12T15:09:06.587Z,7,130851,0,255,8,00,0c,41,9f,00,ff,ff,0a
2022-04-12T15:09:06.588Z,7,130851,0,255,8,01,1a,00,02,d1,06,ff,ff
2022-04-12T15:09:08.522Z,2,130850,4,255,8,40,0c,41,9f,00,ff,ff,0a
2022-04-12T15:09:08.523Z,2,130850,4,255,8,41,1a,00,02,ae,00,ff,ff
2022-04-12T15:09:08.537Z,7,130851,0,255,8,20,0c,41,9f,00,ff,ff,0a
2022-04-12T15:09:08.538Z,7,130851,0,255,8,21,1a,00,02,ae,00,ff,ff
2022-04-12T15:09:09.976Z,2,130850,4,255,8,60,0c,41,9f,00,ff,ff,0a
2022-04-12T15:09:09.977Z,2,130850,4,255,8,61,1a,00,03,ae,00,ff,ff
2022-04-12T15:09:09.986Z,7,130851,0,255,8,40,0c,41,9f,00,ff,ff,0a
2022-04-12T15:09:09.987Z,7,130851,0,255,8,41,1a,00,03,ae,00,ff,ff
2022-04-12T15:09:11.324Z,2,130850,4,255,8,80,0c,41,9f,00,ff,ff,0a
2022-04-12T15:09:11.324Z,2,130850,4,255,8,81,1a,00,03,d1,06,ff,ff
2022-04-12T15:09:11.335Z,7,130851,0,255,8,60,0c,41,9f,00,ff,ff,0a
2022-04-12T15:09:11.336Z,7,130851,0,255,8,61,1a,00,03,d1,06,ff,ff
2022-04-12T15:09:13.036Z,2,130850,4,255,8,a0,0c,41,9f,00,ff,ff,0a
2022-04-12T15:09:13.037Z,2,130850,4,255,8,a1,06,00,ff,ff,ff,ff,ff
2022-04-12T15:09:13.048Z,7,130851,0,255,8,80,0c,41,9f,00,ff,ff,0a
2022-04-12T15:09:13.049Z,7,130851,0,255,8,81,06,00,ff,ff,ff,ff,ff
2022-04-12T15:09:15.074Z,2,130850,4,255,8,c0,0c,41,9f,00,ff,ff,0a
2022-04-12T15:09:15.074Z,2,130850,4,255,8,c1,09,00,ff,ff,ff,ff,ff
2022-04-12T15:09:15.115Z,7,130851,0,255,8,a0,0c,41,9f,00,ff,ff,0a
2022-04-12T15:09:15.116Z,7,130851,0,255,8,a1,09,00,ff,ff,ff,ff,ff
2022-04-12T15:09:19.922Z,2,130850,4,255,8,e0,0c,41,9f,00,ff,ff,0a
2022-04-12T15:09:19.922Z,2,130850,4,255,8,e1,1a,00,03,ae,00,ff,ff
2022-04-12T15:09:19.938Z,7,130851,0,255,8,c0,0c,41,9f,00,ff,ff,0a
2022-04-12T15:09:19.940Z,7,130851,0,255,8,c1,1a,00,03,ae,00,ff,ff
2022-04-12T15:09:24.446Z,2,130850,4,255,8,00,0c,41,9f,00,ff,ff,0a
2022-04-12T15:09:24.447Z,2,130850,4,255,8,01,0a,00,ff,ff,ff,ff,ff
2022-04-12T15:09:24.476Z,7,130851,0,255,8,e0,0c,41,9f,00,ff,ff,0a
2022-04-12T15:09:24.476Z,7,130851,0,255,8,e1,0a,00,ff,ff,ff,ff,ff
2022-04-12T15:09:34.725Z,2,130850,4,255,8,20,0c,41,9f,00,ff,ff,0a
2022-04-12T15:09:34.726Z,2,130850,4,255,8,21,0a,00,ff,ff,ff,ff,ff
2022-04-12T15:09:34.745Z,7,130851,0,255,8,00,0c,41,9f,00,ff,ff,0a
2022-04-12T15:09:34.746Z,7,130851,0,255,8,01,0a,00,ff,ff,ff,ff,ff
2022-04-12T15:09:50.775Z,2,130850,4,255,8,40,0c,41,9f,00,ff,ff,0a
2022-04-12T15:09:50.776Z,2,130850,4,255,8,41,06,00,ff,ff,ff,ff,ff
2022-04-12T15:09:50.787Z,7,130851,0,255,8,20,0c,41,9f,00,ff,ff,0a
2022-04-12T15:09:50.788Z,7,130851,0,255,8,21,06,00,ff,ff,ff,ff,ff
2022-04-12T15:09:55.097Z,2,130850,4,255,8,60,0c,41,9f,00,ff,ff,0a
2022-04-12T15:09:55.098Z,2,130850,4,255,8,61,0a,00,ff,ff,ff,ff,ff
2022-04-12T15:09:55.115Z,7,130851,0,255,8,40,0c,41,9f,00,ff,ff,0a
2022-04-12T15:09:55.116Z,7,130851,0,255,8,41,0a,00,ff,ff,ff,ff,ff
2022-04-12T15:09:57.297Z,2,130850,4,255,8,80,0c,41,9f,00,ff,ff,0a
2022-04-12T15:09:57.298Z,2,130850,4,255,8,81,0a,00,ff,ff,ff,ff,ff
2022-04-12T15:09:57.310Z,7,130851,0,255,8,60,0c,41,9f,00,ff,ff,0a
2022-04-12T15:09:57.311Z,7,130851,0,255,8,61,0a,00,ff,ff,ff,ff,ff
```

sbender9, 2022-04-13:
Only the first three fields are defined. The rest are proprietary. I'd say change it since you have real data.

keesverruijt, 2022-04-13:
I'll see what I can find with my NAC3 and AC42. But it sounds plausible.

tkurki, 2022-05-09:
@keesverruijt any update on this?

keesverruijt, 2022-10-05:
Well, after the recent refactors (merge commit d88f047e57538a8f6ac31564849f4f80182845dd) and upgrade to v4.0.0 it is cleanly picking up that these are unknown fast packets, and it shows the following:

    2022-04-12T15:08:50.879Z 2   4 255 130850 Manufacturer Proprietary fast-packet non-addressed:  Manufacturer Code = Simrad; Industry Code = Marine; Data = 00 FF FF 0A 09 00 FF FF FF FF
    2022-04-12T15:08:50.903Z 7   0 255 130851 Manufacturer Proprietary fast-packet non-addressed:  Manufacturer Code = Simrad; Industry Code = Marine; Data = 00 FF FF 0A 1C 00 FF FF FF FF

which is about as good as it is gonna get without reverse engineering the actual Navico PGNs.

keesverruijt, 2022-10-05:
So I am closing this and opening a new issue that we should reverse engineer Navico autopilot controls more extensively.

closed 2022-10-05 by keesverruijt

TwoCanPlugIn, 2022-10-25:
I've also submitted some info to the canboat repo, but my analysis of the samples here with the current code under development for TwoCan seems to decode this into something sensible.

Hope this helps

[tp32.txt](https://github.com/canboat/canboat/files/9858839/tp32.txt)