canboat/canboat #274: ROT (PGN 127251) resolution

speters, 2022-07-25:
Is there any reliable information about the resolution of ROT in PGN 127251?

Canboat uses
https://github.com/canboat/canboat/blob/bf2b56cd1763e9a6258b60ea5752a0eacd890c9a/analyzer/pgn.h#L39
which is also found in 
https://github.com/ttlappalainen/NMEA2000/blob/c2de650e9abe943da7c87f6a6c03796c4f7b3234/src/N2kMessages.cpp#L312

As about all free software using NMEA2000 is either based on info from Canboat or ttlapalainen's NME2000 library, it makes no sense to see how others do it...

In my test case  I supply
`180°/ min = 3°/s = 0.0523598776 rad/s`

Candump frame:
` can0  09F11319   [8]  01 FC 90 19 00 FF FF FF`

Analyzer output:
`2022-07-25-13:05:47.438 2  25 255 127251 Rate of Turn:  SID = 1; Rate = 3.00000 deg/s`

But on my Raymarine ST70 0.3°/s is shown:
[image]

This matches some [old but official looking PGN docu](http://web.archive.org/web/20220725124745/https://raw.githubusercontent.com/z-xr/NMEA2000/main/NMEA2000_v1-300_App_B1_PGN_Table(1).pdf) where resolution is said to be `1/32*10E-6`:
[image]

[Yacht Devices CAN Log Viewer 1.40](https://www.yachtd.com/products/can_view.html) reports 300 deg/min which I consider an error:
[image]

keesverruijt, 2022-07-25:
I will double check in the coming weeks, I have some good sources on board but haven't double checked the results as defined in canboat. We tend not to make violent corners 😄

keesverruijt, 2022-09-10:
**Actisense EBL reader** and my **B&G Zeus 3** both agree with CANboat.

Note that your old PGN document is self-contradictory, as a reported range of 67 rad/s is equivalent to a resolution of (1/32)*1e-6, not (1/32)*10e-6. (2^31 - 1)*(1/32)*1e-6 = 67,108864. I think the 10 is a printing error in the document. Note that the reported resolution would mean 30 deg/min, not 0.3.

Therefore, I call this as a bug in the ST70, the old document and the YD log viewer.

speters, 2022-09-10:
Meanwhile YD confirmed the bug in YD CAN log viewer.
So CANboat is right.
Thank you very much for your research and your great software!

closed 2022-09-10 by speters