canboat/canboat #282: pgns-v2.json has lookup table values as string, not integer

meaning-matters, 2022-09-06:
The `"values"` fields in the new `"LookupEnumerations"` are strings, not integers.  This makes parsing the JSON harder than necessary.

For example: `{"name": "Global", "value":"0"},` should be `{"name": "Global", "value":0},`.

keesverruijt, 2022-09-07:
Good point, Cornelis. Will fix.

keesverruijt, 2022-09-07:
If you have _any_ other comments on how the new JSON can be improved, let me know!

keesverruijt, 2022-09-07:
I tested with JS where datatypes don't matter, but not every language is as lenient.

meaning-matters, 2022-09-07:
The JSON format requires numbers to be without `"  "`.  Quoted numbers are strings.

JS is just being nice by not adhering to the JSON RFC.

https://www.ietf.org/rfc/rfc4627.txt

tkurki, 2022-09-07:
Parsing JSON in js retains the type information, there's just automatic coercion to Number if you use the value that way. It is not always your friend (read: causes unexpected fails).

Anyway, numbers should be numbers in JSON, unrelated to how you parse & use the format.

commit ff53d1dd (2022-09-07): Fix #282: JSON Numeric lookup values should be numbers, not strings.
closed 2022-09-07 by keesverruijt