canboat/canboat #283: Further improvements to v2 JSON/XML

keesverruijt, 2022-09-07:
### Datatypes

Would it be good to have a list of data types defined in the XML?

Currently there is a `<type>` but this contains abominations like `<Type>String with start/stop byte</Type>`.

Maybe the XML / JSON should be more descriptive?

Something like:

    <FieldType>
       <Name>STRING_VAR</Name>
       <Description>A varying length string containing single byte codepoints.</description>
       <EncodingDescription>The length of the string is determined either with a start (0x02) and stop (0x01) byte, or with a starting length byte (> 0x02), or an indication that the string is empty which is encoded by either 0x01 or 0x00 as the first byte.
       </EncodingDescription>
       <Comment>It is unclear what encodings are allowed/supported. Possibly UTF-8 but it could also be that only ASCII values are supported.
       </Comment>
    </FieldType>

    <FieldType>
       <Name>FIXED_POINT_NUMBER</Name>
       <Description>A numeric value where the Least Significant Bit does not encode the integer value 1.</Description>
       <EncodingDescription>The `Resolution` attribute indicates what the raw value 1 should represent. The `Signed` and `BitLength` attributes are always present. Together, this gives sufficient information to represent a number in a particular range where non-integral values can be encoded without requiring four or eight bytes for a floating point number.
       </EncodingDescription>
    </FieldType>

    <FieldType>
       <Name>LONGITUDE_I32</Name>
       <Description>A geographical longitude.</Description>
       <EncodingDescription>The `Resolution` attribute indicates what the raw value 1 should represent. The `Signed` and `BitLength` attributes are always present. Together, this gives sufficient information to represent a number in a particular range where non-integral values can be encoded without requiring four or eight bytes for a floating point number.
       </EncodingDescription>
       <BaseFieldType>FIXED_POINT_NUMBER</BaseFieldType>
       <BitLength>32</BitLength>
       <Units>deg</Units>
       <Resolution>0.0000001</Resolution>
       <Signed>true</Signed>
    </FieldType>

Should the description contain straight text, Markdown, HTML or text plus markdown/HTML? Not sure of the use cases here.

I think I can fold all numbers to:
 * Unsigned integer (= RES_INTEGER + all current fields with resolution 1)
 * Signed integer (= RES_INTEGER + all current fields with resolution 1)
 * Unsigned scaled fixed point
 * Signed scaled fixed point
 * Floating point
 * Decimal (BCD type, so 0x33 = 33 decimal)

I think it is useful to have as many "derived" field types like LONGITUDE as possible, but should the field definition then skip the attributes that are theoretically superfluous like `BitLength`, `Units`, etc as these are already defined on the `FieldType`?

keesverruijt, 2022-09-07:
Pinging some relevant contributors ... @sbender9 @tkurki @TwoClocks @arks-public @meaning-matters @aldas

keesverruijt, 2022-09-07:
The 3.0.0 version produces new XML and JSON files, which are better but not perfect yet. This is about making them even better!

I now have more time to work on this, so don't worry about suggesting stuff that might be more work. I do want to keep the base in C, so that it can be downloaded from GitHub onto small devices with only a C compiler.

aldas, 2022-09-07:
List of classifier values (does not need to be XML, could be just some block in readme.md) would be useful especially when they come with technical details how that value relates to nmea/business domain. For me your comments in code have been really valuable source of information about NMEA2000 and quirks about devices own datatypes (strings for example) overall. Having classifiers listed in those generated files is good as you can automatically map those values to your app/library domain values and know when something is added/removed.

keesverruijt, 2022-09-07:
Hmm, maybe a domain thing, but what do you mean with "List of classifier values"? Can you give an example?

aldas, 2022-09-07:
Values that `<type>` can have or any "enum" like field. Basically that partial list you pasted `<FieldType>`.   

a little bit off topic: 

If it comes to XML then it has XSD (JSON has jsonschema) which can have comments/descriptions for fields. For example understanding from XML/JSON alone what `RepeatingFields`, `Complete` etc does means currently that I need basically download Canboat source code and do reverse search where and how this field is created and how that field is used in C code.  It would be super nice if something like that exists (even readme.md block would be super fine with description of each field and maybe some known quirks/gotchas/notes related to that field for newbies like me).

inquestruss, 2022-09-07:
I'm working on software that generates golang types and conversions to/from wire formats. The new changes make this far cleaner. 

I don't have a use case for description to push one way or the other, but I'd favor plain text and markdown over mixing in HTML.

We select numeric types consistent with your type descriptions above, so adding field_type would simplify our code. 

A few additional thoughts: 

The fix to #282 (numbers not strings) generates numbers as keys on an object member: {"0":, "Global"}. Golang's json unmarshal cannot easily map this into a golang struct. Instead I unmarshal the values array into a map, then generate go structs from that. The unmarshal function depends on key:value pairs, where the key is the name of an array, struct, or field. Ideally every json member should have key:value definitions, like {"name":"Global", "value":0} or {"bit":, "label":}. (The keys must be strings; the values can be numbers, strings, objects, arrays, true/false/null.) JSON tags make it easy to map keys into structure identifiers as long as they're used consistently.

There remain several pgn members that have type: "Lookup table" but lack a "LookupEnumeration" member:
  1)"ID":"industryCode" members with a "Match": member. The naming conventions make this easy to work around.
  2) bearingReference (in some form) and calculationType defined with the unknown lookup field #define. We could guess these are the same enumerations as other known pgns. For now I have to see if a lookup with a name consistent with the id exists (to support industryCode) and leave it alone otherwise.

Thanks for moving this forward!

keesverruijt, 2022-09-07:
> The fix to #282 (numbers not strings) generates numbers as keys on an object member: {"0":, "Global"}. 

This is a bug, it was intended to be `{"0": "Global"}` -- I trust that is OK?

> 1)"ID":"industryCode" members with a "Match": member.

I dithered with these because I wanted to keep the v1 `pgns.xml` the same. But I now realise I can fix this up for just the v1 style of lookups. So this could be changed. On the other hand, a Match means you shouldn't do a lookup?

> 2) bearingReference (in some form) and calculationType

Yes, we don't know these, for some reason or other. Some more sleuthing may be in order.

keesverruijt, 2022-09-07:
Sorry, my modern language skills are a bit rusty. What I was actually aiming for was something like this:

    "values": { "0": "No", "1": "Yes" }

instead of the current

    "values": [ {"0": "No"}, {"1": "Yes"} ]

so that the languages that I do know somewhat would be able to say something like (expressed in a pseudo language):

    n = getFieldValue();
    lookup = ...;
    print lookup.values[n];

inquestruss, 2022-09-07:
I should have started  by saying we're generating golang declarations and glue code that's compiled into a library to be used by other routines. It's similar to generating much of your analyzer .h and .c files from the json, rather than the reverse.

We're trying to exploit the value of a strongly typed language, for good or ill. We define a type for each lookup, with associated named const values, and functions that return the text given the number. We then generate types for each pgn and stubs that convert to/from each on the wire rather than build them by hand. 

I can do that with your latest json by delaying processing of the Values arrays, unmarshalling them into arrays of maps, then generating the go declarations from that. (As always with code it's just a boolean and X lines of code. :) ) , but it would be much easier to do if the json representation defined objects with named members. 

Here's the relevant information for the golang json package: 

> To unmarshal JSON into a struct, Unmarshal matches incoming object keys to the keys used by Marshal (either the struct field name or its tag), preferring an exact match but also accepting a case-insensitive match. By default, object keys which don't have a corresponding struct field are ignored (see Decoder.DisallowUnknownFields for an alternative).

Your current encoding uses the numeric value as a key, and the name, label, or description of the numeric value as the value. It works well for dynamic languages, but requires a workaround for our use case, since we can't use numeric values as field names.

The other issue is by using numeric values as keys they must be encoded as strings, so consumers in typed languages have to convert each to a number using a library conversion function.  That was the intent (I think) of issue #282.

`

meaning-matters, 2022-09-07:
#282 (numbers not strings):

In many languages an enum is a list of named tags.  Then optionally each tags can be assigned a value.

So you could encode this in JSON as:

```
{
    "No" : 0,
    "Yes" : 1
}
```

However, there is a major issue with this format in JSON: All the hundreds of keys `"No"`, `"Yes"`, ... must be specified upfront in the code that parses this JSON.  This is undoable because it would mean copy-pasting all this CanBoat enum info into the JSON parser.

What we need, is general structure for defining an enum, like we had previously:

```
        "values": [
        {"name" : "Global", "value" : 0},
        {"name" : "Highway", "value" : 1},
        ...
        ]
```
but then just the numbers without quotes.  This allows the parser to build up internal structures that contain the enum info, which can then be further processed (e.g. generating code in a programming language: Golang, Swift, ...).

keesverruijt, 2022-09-07:
OK, I will change accordingly. 

The "Bit" lookup values were always already generated as a key, also in the old `pgns.json`. So the v2.0.0 `PGNs.json` contains, for instance:

    "EnumBitValues":[
      {"0": "Another device controlling thruster"},
      {"1": "Boat speed too fast to safely use thruster"}]},

but the following for normal lookups:

    "EnumValues":[
       {"name":"Off","value":"0"},
       {"name":"Extend","value":"1"},
       {"name":"Retract","value":"2"},
       {"name":"Reserved","value":"3"}]},

According to your suggestions, I should generate (nearly) the same code for LookupBitEnumerations as for LookupEnumerations, with the values/bits as numbers. 

What should I do for the EnumBitValues in pgns.json? I would like to hear from at least @sbender9 on what to do with the old format... I will keep it the same if I don't hear anything but this seems like a good moment to fix this.

keesverruijt, 2022-09-07:
I just pushed a new version that generates:

pgns-v2.json:

     {
        "name": "YES_NO",
        "maxValue": 3,
        "EnumValues": [
        {"Name": "No", "Value": 0},
        {"Name": "Yes", "Value": 1}
        ]
    ...
     "name": "ENGINE_STATUS_1",
        "maxValue": 15,
        "EnumBitValues":[
        {"Name": "Check Engine", "Bit": 0},
        {"Name": "Over Temperature", "Bit": 1},
        {"Name": "Low Oil Pressure", "Bit": 2},

and pgns.json is only changed for stringy numbers from v2, so has numbery numbers and bad bit enumerations:

          "EnumValues":[
              {"name": "No", "value": 0},
              {"name": "Yes", "value": 1}]},
    ...
         "EnumBitValues":[
              {"0": "Check Engine"},
              {"1": "Over Temperature"},
              {"2": "Low Oil Pressure"},

meaning-matters, 2022-09-07:
Some remarks about the `<FieldType>` examples at the top:
- Nesting, like is done with `LONGITUDE_I32` and makes thing too complicated.
- Because there don't seem to be many variations, it might be handy to replace the current`"Type"`, `"Resolution"`, `"Signed"`, and `"Offset"`unique combinations for each of the numerical values with a new type value, plus a new table that fines each type.  But this seems quite some work and I'm wondering/worrying how error prone this data transformation would be.
- A parser could also deduce duplicates in 'types' (i.e. unique combinations of these four values).
- It's nice to keep the bit related values specified for each field.  Having this in a common/shared table at the top of the JSON would make debugging NMEA 2000 messages (while looking at the JSON) a nightmare.
- It would be good if each field gets a `"Type"`.  At the moment for many fields this is JSON key-value pair is missing.
- Typo: the current JSON tag `"Units"` should be changed to `"Unit"`, because that field can only hold one value.

I've not done much with the fields and types yet, so this is all I have for now.  The current JSON seems okay (except for perhaps adding a `"Type"` for all fields, although a default value can be easily filled in by a parser).

meaning-matters, 2022-09-07:
@keesverruijt Thanks for you very quick action on this.  The new `pgns-v2.json`'s enums and bitfields are nice!

keesverruijt, 2022-09-07:
> * Nesting, like is done with `LONGITUDE_I32` and makes thing too complicated.

If you mean the `BaseFieldType`, I intend to use this only for reference. There will be no "inference" so a field-type will be fully descriptive. It will only there to help you relate field types to each other. Of course, any comments applicable to FIXED_POINT should apply to LONGITUDE_I32. If you mean something else, please explain.

> * Because there don't seem to be many variations, it might be handy to replace the current`"Type"`, `"Resolution"`, `"Signed"`, and `"Offset"`unique combinations for each of the numerical values with a new type value, plus a new table that fines each type.  But this seems quite some work and I'm wondering/worrying how error prone this data transformation would be.
> * A parser could also deduce duplicates in 'types' (i.e. unique combinations of these four values).
> * It would be good if each field gets a `"Type"`.  At the moment for many fields this is JSON key-value pair is missing.

My goal is to make every field refer to a type, and this will be enforced in the C code, so I know the JSON will as well. This means that at C level, most fields won't need to set many struct fields and a number of the current members in the `Field` struct will disappear (like `units` and `hasSign`) . Whether this is shown in the XML and JSON is up for discussion.

> * It's nice to keep the bit related values specified for each field.  Having this in a common/shared table at the top of the JSON would make debugging NMEA 2000 messages (while looking at the JSON) a nightmare.
> * Typo: the current JSON tag `"Units"` should be changed to `"Unit"`, because that field can only hold one value.

👍 

> Good idea use the JSON tag "bit".

Thanks.

sbender9, 2022-09-07:
I'm ok with changing that. Easy fix on my side.

meaning-matters, 2022-09-07:
I find `"Unit"` extremely useful, or even necessary,  and don't understand why or how you would remove that.

Having the unit allows showing it in UIs or printing it, together with the value.  If the unit is not supplied, I'd have to deduce it myself.

Could you explain?

keesverruijt, 2022-09-08:
`Unit` won't be removed, it might just be moved elsewhere. What I said is I am moving it, in the C code, to the list of field types.

Currently all attributes are reported on every PGN field, as this is how it is stored in C so it can't be centralised. With the upcoming internal structure it becomes possible to report just a single list of fieldtypes, with the actual PGN fields referring to the field type. This is just like the lookup values. The new JSON only reports the lookup values once, plus a reference every time that particular lookup is used. The question then is, does the XML repeat the redundant information (I think not) and does the JSON repeat the information (up for discussion).

meaning-matters, 2022-09-10:
Some further thoughts on `BaseFieldType` and nesting:

In programming languages a type specifies its format and how this is stored in memory, not how it's used.  You can use a `double` for latitude, voltage, time-duration, ...

I think it would be good to keep this separation here too:
- Field-type defines how many bits, how these bits are laid out (bitLength and bitStart), the format (e.g. [un]signed integer, IEEE  float, ...), resolution, and offset.
- Then the unit defines how it's used.

Looking at the `<Name>LONGITUDE_I32</Name>` example, you would get a `COORDINATE_I32` field-type plus new units `lat` and `long`; no nesting needed.  The reason for nesting was because `LONGITUDE_I32` is combination of a pure field-type plus a unit.  Then, in case this same I32 format is used for other values, so not only coordinates, an `I32` field-type should be created and perhaps more unit introduced besides `lat` and `long`.

You can say that the notion of 'coordinate' in a way defines use, but if a certain field-type is not used for other things, there's no need to to further strip meaning/use from the field-type name; so it would be fine to name it `COORDINATE_I32` and not the more pure `I32`.

To summarise: Separate types from how they are used, in field-types and units respectively.  (This nesting/referring back to a 'base type' is the wrong way around which leads to a more complex JSON format.)

aldas, 2022-09-10:
just pointing out that currently not all PGN fields in json/xml have `type` field. 

--------------

About "types" why can not it be just enum for deciding how bytes in nmea2000 packet should be converted data type/structure.     
as @meaning-matters wrote that "resolutions"/"types" currently contain at least 2 types of information  - business/domain related info and computer variable/data type/structure info. 

So resolution list is currently
```c
#define RES_RADIANS (1e-4)
#define RES_ROTATION (1e-3 / 32.0)
#define RES_HIRES_ROTATION (1e-6 / 32.0)
#define RES_ASCII (-1.0)
#define RES_LATITUDE (-2.0)
#define RES_LONGITUDE (-3.0)
#define RES_DATE (-4.0)
#define RES_TIME (-5.0)
#define RES_TEMPERATURE (-6.0)
#define RES_6BITASCII (-7.0) /* Actually not used in N2K, only in N183 AIS */
#define RES_INTEGER (-8.0)
#define RES_LOOKUP (-9.0)
#define RES_BINARY (-10.0)
#define RES_MANUFACTURER (-11.0)
#define RES_STRING (-12.0)
#define RES_FLOAT (-13.0)
#define RES_PRESSURE (-14.0)
#define RES_STRINGLZ (-15.0)  /* ASCII string starting with length byte and terminated by zero byte */
#define RES_STRINGLAU (-16.0) /* ASCII or UNICODE string starting with length byte and ASCII/Unicode byte */
#define RES_DECIMAL (-17.0)
#define RES_BITFIELD (-18.0)
#define RES_TEMPERATURE_HIGH (-19.0)
#define RES_TEMPERATURE_HIRES (-20.0)
#define RES_PRESSURE_HIRES (-21.0)
#define RES_VARIABLE (-22.0)
```

now lets take `RES_LONGITUDE` just for an example and search for its usages

https://github.com/canboat/canboat/blob/4e2b06a417730dd5e2c9ac8865807542b42299b3/analyzer/analyzer.c#L626
we see that by field having resolution value of `RES_LONGITUDE` we can deduce direction from it. instead of using it as dumb integer/decimal type.  If I am just interested raw number that I can log I only would need instruction how to parse/convert those field bytes to some data type. Why not have just ordinary common data types (integers/different decimals(depending formats)/bools/enums/strings (+their different variations) and leave "how to interpret/print this number" out of "type" field.

-------------

even with list of "types" I think important magic is still inside this block https://github.com/canboat/canboat/blob/c95d422feaccfb5c9464b3a3cb73c0f19cd2c946/analyzer/analyzer.c#L1809

this is where you would search how to interpret resolutions (how it should look in human readable form). All those "print*" functions are something that you need to implement in your language of choosing. You probably do not need SI units conversions but you do need to know if there are hidden meanings in number. ala Negative number means WEST or SOUTH. Or how this `uint32_t` decodes to time seconds/minutes/hours etc - it is seconds since start of day or epoch or what it is?

meaning-matters, 2022-09-11:
@aldas

As I wrote somewhere above, I think that each field should have a type indicator.  It's not a big deal for software that processes the JSON because it can fall back to a default, but that's extra logic I'd rather not see and it makes the JSON a bit harder to read.

I wouldn't call the sign of lat/long a hiding meaning, because it's just how they are defined.  But there could indeed be hidden meanings in other types of values.  @keesverruijt have you seen this?

aldas, 2022-09-11:
after a long sleep - I think I am pointing out that the list of PNGs alone is not enough it to be usable for other people. each field has their meaning that is currently only visible if you reverse engineer Canboat code. Personally having types like "RES_TEMPERATURE_HIGH" is fine (this is artistic freedom of Canboat authors) - but to use it I need to know how to get human readable number,text or boolean (does not matter what data type) out of series of bits/bytes in the N2K packet, to be able to build my own library/application for N2K.

meaning-matters, 2022-09-11:
@aldas Yes, quite some meaning is hidden.

Separating out the enumeration and bitfields was a massive step to make the JSON more usable.  (Although this could also have been deduced from the JSON, which I was actually working on until a week ago.)

By also having field-types that specify bit-format plus basic type indicator (integer, float, ...) and a separate unit (the determines meaning), I think the JSON would be very usable.  I've not gone through everything, but it seems to me that such JSON would be sufficient to build a program that can both parse and compose NMEA 2000 message (or generate code that does this).

keesverruijt, 2022-09-11:
Yes, I think we are all on the same page here.

I am about 70% into redefining the headers, and 5% into refactoring the source code.

My intention is to be able to cover all your suggestions.

I want to generate some form of HTML pages from the XML, with (amongst others) one page per PGN, so that if you are looking at a particular PGN it is easy to find. It should also contain detailed information per field, allowing you to understand all fields without having to refer back to the source code.

This means that just the XML + XSLT will contain all meaning.

It is then up to you the downstream consumers to decide how much information you want in the JSON (which is generated from the XML.)

keesverruijt, 2022-09-13:
To show you where I am headed I have pushed branch `refactor-analyzer`. It is still very much a work in progress. 

The `pgns-v2.<ext>` files have been moved to the 'docs/' subdirectory and are now named `canboat.<ext>` (as it's no longer just PGNs in there).

So far I am focusing on:
- Reducing the code complexity in C
- Generating the proper XML and XSL so that when rendered in a browser, you get a complete set of documentation on N2K messages.
- Keeping the XML validating with the DTD.

I have _NOT_ looked at the generated JSON yet, at all!

You can have a quick peek at the documentation at https://keversoft.com/canboat/canboat.xml which should render nicely on all modern browsers.

Right now the XML has correct field lists, and links to enumerations and field types. Try them!
What is missing is still a lot of (slightly) more secondary info like whether the PGN is complete, variable field lists, etc.

meaning-matters, 2022-09-13:
Impressive results Kees, in such a short time!

Is field bit-size sufficient information (ere subsequent fields always laid out against each other)?

keesverruijt, 2022-09-13:
Some more hours put in, try it again. The XML/XSL doc page is getting there now! Same URL and new commit in same branch as above.

> Is field bit-size sufficient information (ere subsequent fields always laid out against each other)? If so, that would be a great simplification.

Yes, all fields are always adjacent. The "BitOffset" field was added for one of the downstream projects years ago. I think I will leave it out in the new definitions for now, unless somebody has a persuasive reason.

meaning-matters, 2022-09-13:
Had a look and noticed a few things:
- PGN 65240 has a type MANUFACTURER with unit "Manufacturer Code".  Should be a LOOKUP right?  When going to the definition of MANUFACTURER is says, "This is a specific instance of a LOOKUP".
- There are now CURRENT_UFIX16_A etc types.  I think combining basic types plus units to make a massive number of types is very bad idea (see discussion above).  The type should be UFIX16 and the Unit can then indicate what it's used for.

aldas, 2022-09-14:
I agree these special cases make life harder. If I would write code from decoding I would probably just make all those cases with same suffix ala `LENGTH_UFIX32_MM`, `TIME_UFIX32_S` fall into same switch block as when you are doing raw bytes decoding I do not care about  "business domain" -  If some value in packet which data type is Uint64, is related to GEO or Location - this is not important.

p.s. I would not call that field `unit` but `domain`, `category` or something like that. Example from Nmea0183 - often you could have field for speed and with that you have unit and that unit can be (depending on the device state)  knots, meter per second, km per second. Some devices allow you to configure in which units values are emitted. so naming it as `unit` would be maybe too confusing.

---

I suggest adding `scale` as column. For example packet "PGN 65004 - Bus #1 Average Basic AC Quantities" has field "AC Frequency" and scale would be for that field "0.0078125".  I think `scale` is common way to describe these kind of constanst that you would need to apply after converting byte to your language type.

Order | Name | Description | Size (bits) | Type |	Unit |	Lookup 
1 | Line-Line AC RMS Voltage |   | 16 | VOLTAGE_UFIX16 | V |  
2 | Line-Neutral AC RMS Voltage |   | 16 | VOLTAGE_UFIX16 | V |  
3 | AC Frequency |   | 16 | FREQUENCY_UFIX16 | 0.0078125 Hz |

aldas, 2022-09-14:
Maybe it is worth to check how Canbus packets are usually described and especially field (signals in canbus terms) 
https://docs.openvehicles.com/en/latest/components/vehicle_dbc/docs/dbc-primer.html 
There is "offset" - I do not if this is used in N2K.  
Also having length in addition to starting bit is useful because when I need to create my decoding code I need to know how many bits to read to convert to some type. When it is not present I would take the next field and calculate length from that.

meaning-matters, 2022-09-14:
@aldas I meant is that types that combine bit-packing/data-format type with unit information, make things very hard to use.

First you need to unpack NMEA 2000 packets and convert bit fields into basic values of type int, double, ....  For this you need only UFIX16 etc type information.

Then, at a completely different layer of abstraction/code, these converted values are interpreted.  This is when you need Unit and Resolution (or call it Scale like) information.

aldas, 2022-09-14:
Probably this list would be compacted to primitive types + few special encoding (for strings).  I assume that 99% cases would be covered with cases like:  
* bit  (single bit of data)
* int (+additional field for bits (ie. 8/16/24/64 etc) + field for signed/usigned, +is there need for endianess? For example MODBUS you need endianess sometimes).  Or expand this list as int8,uint8, ... but I think most of the languages and their libraries support decoding integers by providing bytes and setting bit length as additional argument.
* float (different bit lengths?)
* double
* string*** (and its variations)
* lookup (including manufacturer etc and use some additional field for defining which built in lookup table to use) 
* bitlookup
* binary (just N bits of raw data)
* reserved (field that is not used)

* variable (maybe fold into binary for simplicity)

tkurki, 2022-09-15:
>You can have a quick peek at the documentation at https://keversoft.com/canboat/canboat.xml which should render nicely on all modern browsers.

Cool! I have a few improvement ideas and it could published in Github pages, automatically via Github Actions, but those are probably better handled as separate issues/PRs. Can help with that.

Otherwise I think having a rich enough XML source of truth is a good way to go. Other downstream artifacts can be produced off that, even if it means wrecking your brains with XSL 😁 .

tkurki, 2022-09-15:
> I'm working on software that generates golang types and conversions to/from wire formats.

@inquestruss is this open source & available someplace? If so maybe worth linking from the README (#301)?

closed 2022-10-05 by keesverruijt

meaning-matters, 2022-10-05:
In an earlier version `pgns-v2.json` was present.  Does it have to generate `pgns-v2.json` by hand or will you add it later?

keesverruijt, 2022-10-05:
See the changelog, these are now in `docs/canboat.xml` and `docs/canboat.json`. There are a validator file (xsd) and a stylesheet (xml) to generate a human-readable version as well.

keesverruijt, 2022-10-05:
It was renamed because it no longer only contained PGNs.

keesverruijt, 2022-10-05:
And @tkurki you may have seen the Github Pages doc generated by canboat.xml + canboat.xsl ...

If you guys have further improvement suggestions, keep them coming ...

keesverruijt, 2022-10-05:
https://canboat.github.io/canboat/canboat.xml

meaning-matters, 2022-10-05:
What is the purpose of having `BaseFieldType` and the resulting very large number of types that differ in resolution only?

Looking at `NUMBER`, `INTEGER`, and `UNSIGNED_INTEGER` that don't contain details that define field size or format, while all other types do contain this, what is the generic meaning of a type?

Types now mix what the field is used for with memory format details (apart from the three above).  Why is this considered to be useful?  For example: `DILUTION_OF_PRECISION_FIX16`, what does this add to having a `FIX16` type that defines it's 16 bit and range -32767 ... 32765 and then just have a `resolution` value per field?

Something else, field `"Id":"azimuthOffset"` is of type `ANGLE_FIX16`, but both the type and the field define `resolution`: `0.0001` and `0.001`.  This shows that something is wrong.  What's the purpose of `resolution` in the type if it is overridden in a field?

I suggest:

    "Formats":[
    {
        "Name":"FIXED_POINT_NUMBER"
        "Description":"A numeric value where the Least Significant Bit does not encode the integer value 1",
        "EncodingDescription":"The ....",
        "URL":"https://en.wikipedia.org/wiki/Fixed-point_arithmetic",
    },
    ....
    ],
    "FieldTypes":[
    {
        "Name":"FIX16",
        "Format":"FIXED_POINT_NUMBER",
        "Bits":16,
        "Signed":true,
        "RangeMin":-32767,
        "RangeMax":32765
    },
    ...
    
       {
            "Order":5,
            "Id":"azimuthOffset",
            "Name":"Azimuth offset",
            "Description":"Positive: sensor rotated to port, negative: sensor rotated to starboard",
            "BitOffset":24,
            "BitStart":0,
            "Unit":"rad",
            "Resolution":0.0001,
            "FieldType":"FIX16"
          },

keesverruijt, 2022-10-05:
> Something else, field `"Id":"azimuthOffset"` is of type `ANGLE_FIX16`, but both the type and the field define `resolution`: `0.0001` and `0.001`. This shows that something is wrong. What's the purpose of `resolution` in the type if it is overridden in a field?

This is exactly why I am doing it the current way. All ANGLE_FIX16 fields should be the same in everything except the order in the PGN and possibly their description. You have just caught an issue, the initialiser should find this and complain. The reason that there is still a `Field.resolution` field is that I am lazy and some PGN fields are totally unique so defining a shared `FieldType` doesn't offer any advantages.

> suggestion on reducing # of field types

You have repeatedly stated your position. I don't really get why you couldn't write the parser as I have just proven I can do this in C. In fact, it is dead easy now.

One of the best examples why I like the current format is:

> SIGNED_ALMANAC_PARAMETER - Almanac parameter, signed
> 
> Encoding: These encode various almanac parameters consisting of differing sizes and sign. They are all using an interesting resolution/scale, which is always a number of bits that the value is shifted left or right. This is reflected by resolution field containing some factor of 2^n or 2^-n.
> 
> For more information see: https://www.gps.gov/technical/icwg/IS-GPS-200N.pdf
> 

So having SIGNED_ALMANAC_PARAMETER adds nothing besides this piece of documentation! Indeed, but still useful!
Currently the resolutions for SIGNED_ALMANAC_PARAMETER are still printed as floating point values, but there is no reason the XSL couldn't transform this back to `2^n` with n usually a large negative value.

This is NOT useful for a parser, but it IS useful from a documentation point of view.

I actually use the current definitions internally so that the C code for the analyzer is shorter and simpler. Everything that derives from NUMBER is printed by `fieldPrintNumber()`, unless a derived field type overrides this. Having these extra field types also eases the addition of new PGNs as only a few attributes need to be specified. And if a new field type is needed, you know that you are likely wrong or that the NMEA has messed up and defined something wanky again.

Would it help if the JSON contained an attribute which interpreter (parser) the C code uses for interpreting the field? E.g. show "NUMBER" for all fields that are printed by `printFieldNumber` etc.? This would be your `Format`? You can then ignore the FieldTypes if you want.

keesverruijt, 2022-10-05:
Maybe I start seeing your point now. I think the problem is that the current JSON (and XML) reflects the internal representation a bit too much. I will try to fix this.

aldas, 2022-10-05:
maybe @ttlappalainen would like to contribute his thoughts and experience.  His library has abstracted type listand API to quite small number of variants https://github.com/ttlappalainen/NMEA2000/blob/master/src/N2kMsg.h#L125

keesverruijt, 2022-10-05:
Dear downstream users, could you add yourself to #312 ?

meaning-matters, 2022-10-05:
> This is exactly why I am doing it the current way. All ANGLE_FIX16 fields should be the same in everything except the order in the PGN and possibly their description. You have just caught an issue, [...]

My point is dat `FIX16` is all the information you need to extract field from the PGN bit-stream.  The fact that it's an angle is already clear from the field's unit & id and often from the PGN.  You just need the basic type, like `FIX16` plus the resolution.  All these sub-types are redundant and result in a noise & unclarity.

> 
> > suggestion on reducing # of field types
> 
> You have repeatedly stated your position. I don't really get why you couldn't write the parser as I have just proven I can do this in C. In fact, it is dead easy now.

One can write a parser for any valid JSON.  I just wish the JSON presents the NMEA 2000 information in a compact and orthogonal/simple/consistent way, because doing so make the JSON easier to understand and use.

> One of the best examples why I like the current format is:
>
>
> So having SIGNED_ALMANAC_PARAMETER adds nothing besides this piece of documentation! Indeed, but still useful! Currently the resolutions [...]

It's very useful to have the format info and references.  That's why I suggested to list them all in a new array `"Formats":[...]` and to add a "Format" field to each `FieldType`.

> I actually use the current definitions internally so that the C code for the analyzer is shorter and simpler. Everything that derives from NUMBER is printed by `fieldPrintNumber()`, unless a derived [...]

`NUMBER` does not contain any format info, so that would not be useful.  As said before, each field should simply have a base-type the only specifies how the field bits can be turned into an (unsigned) integer or floating point value.  Each field then specifies where the bits are to found in the PGN stream (`BitLength`, `BitOffset`, and `BitStart`) and how the base value is to be interpreted by applying `Resolution`, `RangeMin`, `RangeMax`, and `Unit`.

keesverruijt, 2022-10-05:
OK, please examine the latest push. There are still a few corner cases I see now, like the ranges for bitfields are incorrect (they should probably be hidden) but it is way past bedtime here!

meaning-matters, 2022-10-06:
@keesverruijt The JSON now looks very good.  I'm amazed you made this massive change in such a short time.

I have one remark: Would be good to have `Resolution` for every `NUMBER` because it's needed by a decoding function, also when it's `1`.  Things get a bit more simple/orthogonal if there are no exceptions/defaults like this.

`PhysicalQuantities` is really nice!

aldas, 2022-10-06:
my two cents. Smaller types list is very nice change - it makes writing parsing/decoding parts easier.  

----

I can not seem to find place where json or xml structure is described. For example what `Match` means and when to use it or what is the difference between `Type` and `FieldType`

----

About enum/lookuptables. In json I find these fields

First field example:
```json
          {
            "Order":9,
            "Id":"industryCode",
            "Name":"Industry Code",
            "BitLength":3,
            "BitOffset":60,
            "BitStart":4,
            "Type":"Lookup table",
            "Signed":false,
            "RangeMin":0,
            "RangeMax":6,
            "FieldType":"NUMBER",
            "EnumValues":[
              {"name": "Global", "value": 0},
              {"name": "Highway", "value": 1},
              {"name": "Agriculture", "value": 2},
              {"name": "Construction", "value": 3},
              {"name": "Marine", "value": 4},
              {"name": "Industrial", "value": 5}]},
```

Second field example:
```json
          {
            "Order":3,
            "Id":"industryCode",
            "Name":"Industry Code",
            "Description":"Marine Industry",
            "BitLength":3,
            "BitOffset":13,
            "BitStart":5,
            "Match":4,
            "Type":"Lookup table",
            "Signed":false,
            "RangeMin":0,
            "RangeMax":6,
            "FieldType":"NUMBER"},
```

So these 2 fields are enums (or "lookup table" in Canboat terminology). I assume that first field could have different values and second one is constant. But for second field it is not super obvious for `Industry Code` field to know which enum/looktable to use if would like to map that "4" value to text.

I think that using both ` "FieldType":"NUMBER"` and ` "Type":"Lookup table",` to describe enum/lookup is redundant because enum value is always integer. Looking at https://canboat.github.io/canboat/canboat.xml#lookup-enumerations I do not see any exceptions to that rule.

Maybe for these cases to use `"FieldType":"ENUM"` + `"EnumType":"INDUSTRY_CODE"` and drop `"Type"` altogether 

 so second field would be
```json
          {
            "Order":3,
            "Id":"industryCode",
            "Name":"Industry Code",
            "Description":"Marine Industry",
            "BitLength":3,
            "BitOffset":13,
            "BitStart":5,
            "Match":4,
            "FieldType":"ENUM",
            "EnumType":"INDUSTRY_CODE"
},
```

keesverruijt, 2022-10-06:
Please make sure you are looking at `canboat.xml` and `canboat.json` in the `docs/` directory. It is accompanied by a `canboat.xsd` file that explains the _structure_ of the XML (but not the _purpose_ of each field yet.) Adding an explanation on field level is next on my list, but you can read a lot from the `canboat.xsl` file that does the magic of translating the .xml file to a .html file on the fly when your browser opens the .xml file.

`pgns.xml` and `pgns.json` (containing the repeated lookup field definitions) are obsolete and do not contain all new fields. I see now that `PhysicalQuantity` leaked in there which is a bug, the idea is we keep those file as they were (as much as possible). There is no point in having it there as the list of PhysicalQuantities is not in those files.

I will add the name of the lookup table to the fields that contain a `Match`. And add a 1 for empty number resolutions.

aldas, 2022-10-06:
oh sorry, I was looking from analyzer folder from old habits.

keesverruijt, 2022-10-06:
From `CHANGELOG.md`:

- XML v2 (`canboat.xml` + `canboat.json`) changes:
  - Added LOOKUP and BITLOOKUP as base field types in the FieldList. This means they
    are easily recognised (they are no longer a NUMBER).
  - Emit `Resolution` even when it is 1. Only stringy and bitfield types now do not have a resolution.
  - Emit `LookupEnumeration` even for fields that have a `Match`.
- XML v2 `canboat.xsl` changes:
  - Show offset for number fields and add textual explanation.

keesverruijt, 2022-10-06:
> @keesverruijt The JSON now looks very good. I'm amazed you made this massive change in such a short time.

That's because I basically just had to hide emitting most fieldtypes and walk up the chain to print the base type, redefining what is a base type for a _few_ fieldtypes, and add a list of physical quantities and refer to those in the existing derived fieldtypes. This means I only had to add 76 references to a physical quantity instead of thousands -- there are 2522 fields defined!

So I am still glad I have the derived fields internally, it makes the work a lot less as a _definer_. I just had to learn that you as _consumers_ are not interested in how I come up with the huge list!

aldas, 2022-10-16:
Maybe it should go to discussions but any - I will post it here

Latest changes intruduced new field type `INDIRECT_LOOKUP` and there are now extra fields `LookupIndirectEnumeration` and `LookupIndirectEnumerationField`.

It seems that `LookupIndirectEnumerationField` contains  `field.name` value but in schema we have also `Id` field which is just camel case of `Name`. 

Some thoughts:
* is `Id` meant to be unique? If it is not - why have that field all?
* are IDs guaranteed to be unique?
* if `Id` is unique maybe `LookupIndirectEnumerationField` should reference to `Id` value instead of `Name`

keesverruijt, 2022-10-17:
> is Id meant to be unique in that PGN? If it is not - why have that field all?

The `id` was added on the request of an old downstream project -- which I think is now long defunct...

> are IDs guaranteed to be unique within PGN?

Yes. But so is `Name` from which it is automatically derived.

> if Id is unique maybe LookupIndirectEnumerationField should reference to Id value instead of Name in XML/JSON

I will add an Order, which is even easier for automated code and leave the current field in for the benefit of the humans reading the documentation.

aldas, 2022-10-18:
@keesverruijt there are fields names/ids that are not unique. Mostly it is `reserved` that is offending name but there are other names also. For example PGN `129556` has 2 fields with name  `tnA`  order 13 and order 9

I generated list

```csv
PGN, Field name, Conflicting field order,  first order of conflicting name
65240, Reserved, 10, 6
65280, Reserved, 5, 2
65284, Reserved, 7, 2
65287, Reserved, 6, 2
65309, Reserved, 7, 2
65312, Reserved, 6, 2
65345, Reserved, 6, 2
65359, Reserved, 7, 2
65360, Reserved, 7, 2
65361, Reserved, 6, 2
65371, Reserved, 9, 2
65379, Reserved, 7, 2
65409, Reserved, 7, 2
65410, Reserved, 7, 2
126720, Reserved, 6, 2
126720, Reserved, 6, 2
126720, Reserved, 6, 2
126720, Reserved, 6, 2
126720, Reserved, 6, 2
126720, Reserved, 6, 2
126720, Reserved, 6, 2
126720, Reserved, 6, 2
126720, Reserved, 6, 2
127233, Reserved, 7, 4
127233, Reserved, 13, 7
127233, Reserved, 18, 13
127245, Reserved, 6, 3
127493, Reserved, 7, 3
128776, Reserved, 14, 6
129026, Reserved, 6, 3
129038, Reserved, 19, 17
129039, Regional Application, 15, 14
129040, Regional Application, 12, 11
129040, Reserved, 16, 13
129040, Reserved, 25, 16
129041, Reserved, 22, 19
129283, Reserved, 6, 3
129285, Reserved, 9, 7
129540, Reserved, 11, 3
129556, tnA, 13, 9
129792, Reserved, 9, 4
129792, Spare, 10, 6
129793, Reserved, 13, 8
129795, Reserved, 8, 4
129795, Reserved, 10, 8
129796, Reserved, 6, 4
129796, Reserved, 9, 6
129797, Reserved, 6, 4
129801, Reserved, 9, 6
129803, Reserved, 8, 4
129806, Reserved, 8, 5
129806, Reserved, 15, 8
129806, Reserved, 19, 15
129807, Reserved, 12, 6
129807, Spare, 14, 4
129807, Reserved, 15, 12
129810, Reserved, 15, 12
130323, Reserved, 10, 2
130324, Reserved, 10, 2
130570, Station, 17, 7
130816, Reserved, 4, 2
130816, Reserved, 4, 2
130816, Reserved, 11, 4
130816, Reserved, 4, 2
130816, Reserved, 4, 2
130816, Reserved, 4, 2
130816, Reserved, 4, 2
130816, Reserved, 4, 2
130816, Reserved, 4, 2
130816, Reserved, 11, 4
130816, Reserved, 4, 2
130816, Reserved, 4, 2
130816, Reserved, 4, 2
130816, Reserved, 4, 2
130816, Reserved, 4, 2
130816, Reserved, 4, 2
130816, Reserved, 4, 2
130816, Reserved, 4, 2
130816, Reserved, 4, 2
130816, Reserved, 4, 2
130816, Reserved, 4, 2
130816, Reserved, 4, 2
130820, Reserved, 4, 2
130820, Reserved, 6, 4
130836, Reserved, 12, 2
130837, Reserved, 12, 2
130842, Reserved, 18, 2
130944, Reserved, 5, 2
```

keesverruijt, 2022-10-18:
Well it was a bit late when I wrote this, should have been in bed ages ago. But it does lead to your interesting results that I did not expect!

I have created issue #321 to fix this.

commit d4434046 (2022-10-05): Fix #283: Simplify fieldtypes
Fix #311. Fix #310.
commit 875f6d0d (2022-10-06): Fix #283 (again): more comments incorporated.