canboat/canboat #289: JSON tags "name" and "maxValue" not capitalized

meaning-matters, 2022-09-09:
All JSON tags start with a capital letter, except "name" and "maxValue" that are part of the new enumeration and bitfield.

Clearly a detail that was missed in the very large 3.0.0 update.

keesverruijt, 2022-10-05:
Please verify as fixed on master.

arks-public, 2022-10-13:
Looks like "Name" tags were fixed in master JSON. I still see some "name" keys, but all of them are inside "EnumValues" arrays.

And I don't see any "maxValue" or "MaxValue" tags in the master JSON. So I think that both are fixed.

P.S. It doesn't look like analyzer/pgns2json.xslt is still being used, but it still has the uncapitalized "name" and "maxValue" in case that is a problem.

keesverruijt, 2022-10-17:
I don't want to change pgns.xml for reasons of compatibility with old existing downstream projects.

closed 2022-10-17 by keesverruijt