canboat/canboat #293: Two ways of writing "AIS RAIM flag"

meaning-matters, 2022-09-09:
The JSON contains both `"AIS RAIM flag"` and `"AIS RAIM Flag"`.  Should use only one spelling.

keesverruijt, 2022-10-05:
Please verify as fixed on master.

arks-public, 2022-10-13:
It looks like the goal was to change all occurrences of "RAIM", "AIS RAIM Flag", and "AIS RAIM flag" to just be "RAIM"?

I can verify this was accomplished in all the main files. However, I did find three occurrences of "AIS RAIM flag" in n2kd/gps_ais.c

I don't know if this file was supposed to be updated as part of the change. If not, then this issue is fixed on master.

meaning-matters, 2022-10-13:
Just checked the JSON and it's nicely `RAIM` in every occurrence of this field.

keesverruijt, 2022-10-17:
The AIS flags in gps_ais.c are fixed in #318

closed 2022-10-17 by keesverruijt