canboat/canboat #300: Explain units

Selebrator, 2022-09-13:
Maybe this sounds silly, but I think having documentation on Units would be good.
Units that caused me trubble are
- vol (I still don't know what this is, maybe volume in %)
- Cos Phi, var and VAR (I figured them out, but having documentation here would have make it easier)
- m/m (I think this is for specifying how much wider the earth is, then it is tall)
- s/s (I think this is for comparing the speed of two clocks)

I explained what I think they mean, to either help with the documentation, or to show why documenting them is important :)

Selebrator, 2022-09-13:
I don't find vol being used. I must have made it up

arks-public, 2022-10-04:
I introduced the units m/m and s/s, although currently they are only used for the GPS Almanac Data (pgn 129541). I considered marking them as unitless, but I decided against it since they represent a ratio of two related but not equivalent quantities.

For example, s/s represents a clock drift over time, similar to how acceleration represents change in velocity over time (m/s /s). Even though the numerator is also seconds in this case, it didn't seem appropriate to me to reduce the expression to be unitless.

keesverruijt, 2022-10-05:
Please take a look at https://canboat.github.io/canboat/canboat.xml and if you think more can be done to explain units, please suggest something.

Note how nice the description for the PGN 129541 GPS Almanac Data fields is now!

The best way would be to indicate which fields you think need which extra explanation and/or an external URL.