canboat/canboat #303: Lookups with name for largest values still interpreted as error and unknown

Selebrator, 2022-09-16:
Some Lookups have a mapping for the largest value. In those cases I would expect that mapping to be used.

For Lookup corresponding to large enough fields, the interpretation for the largest value is overwritten with `Unkown`, and that of the second largest value with `ERROR`.

In the case of `NAV_STATUS` used by `AIS Class A Position Report  (129038)` the value 14 `AIS-SART` is replaced by `ERROR` because of this issue. That sound dangerous.

The issue is illustrated well by the Lookup `ATON_TYPE` used in `AIS Aids to Navigation (AtoN) Report (129041)`, as it has an interpretation for all 32 values. So it should be impossible to have a case that is without an interpretation.

Value of 'AtoN Type' field | Expected interpretation | Actual interpretation
|-|-|-|
00 | Default: Type of AtoN not specified | Default: Type of AtoN not specified
... | | All good
1d | Floating AtoN: safe water | Floating AtoN: safe water
1e | Floating AtoN: special mark | **ERROR**
1f | Floating AtoN: light vessel/LANBY/rigs | **Unknown**

## Command to reproduce
```
$ echo "2022-06-15T11:18:17.536Z,4,129041,5,255,48,d5,e6,6c,22,3b,60,6c,de,04,93,b8,14,20,45,78,00,78,00,3c,00,3c,00,1f,ee,00,e0,16,01,4e,47,31,37,20,57,49,4e,44,46,41,52,4d,40,40,40,40,40,40,40" | analyzer
2022-06-15T11:18:17.536Z 4   5 255 129041 AIS Aids to Navigation (AtoN) Report:  Message ID = 21; Repeat Indicator = Unknown; User ID = 992111846; Longitude =  8.1685600; Latitude = 53.8228883; Position Accuracy = High; AIS RAIM Flag = not in use; Time Stamp = 17; Length/Diameter = 12.0 m; Beam/Diameter = 12.0 m; Position Reference from Starboard Edge = 6.0 m; Position Reference from True North Facing Edge = 6.0 m; AtoN Type = Unknown; Off Position Indicator = No; Virtual AtoN Flag = No; Assigned Mode Flag = Autonomous and continuous; AIS Spare = 0x0; Position Fixing Device Type = Surveyed; AtoN Status = 0x0; AIS Transceiver information = Channel A VDL reception; AtoN Name = NG17 WINDFARM

$ echo "2022-06-02T15:06:48.074Z,4,129038,5,255,28,01,a0,d9,97,0c,e6,af,dd,04,b8,48,e9,1f,bc,9c,3f,00,00,53,c0,01,ce,8d,00,00,ce,f8,ff" | analyzer
2022-06-02T15:06:48.074Z 4   5 255 129038 AIS Class A Position Report:  Message ID = 1; Repeat Indicator = Initial; User ID = 211278240; Longitude =  8.1637350; Latitude = 53.5382200; Position Accuracy = Low; RAIM = not in use; Time Stamp = 47; COG = 93.3 deg; SOG = 0.00 m/s; Communication State = 0x1c053; AIS Transceiver information = Channel A VDL reception; Heading = 208.0 deg; Rate of Turn = 0.00 deg/s; Nav Status = ERROR; Special Maneuver Indicator = Not available; AIS Spare = 0x0; Sequence ID = Unknown
```

Tested Versions: 
- 2.0.0
- Latest dev version, commit 7ef83a4c8f91a444bc1b6c5debec0cb0096b7162 on refactor-analyzer

# Suggestion for a fix
I think it would be best to not apply the usual rule for determining if a field is empty on Lookups.
Instead I would like the values with special meaning to be explicitly listed in the Lookup declaration.

Here are my supporting arguments

- Occasionally the value meaning "no value present" is a value different from the largest value (e.g. SHIP_TYPE: 0 = Unavailable). With the proposed solution we have a consistent way to handle this.
- This solution would be great for code using enums generated from the Lookup declaration. I would no longer have to check for the 2 special cases, before I can enter a switch/match expression over the enum.
- Should reduce confusion among N2k Novices.

In my opinion, the name for "no data available" should be the same among all Lookups. Currently we have
| Lookup name | Value | Name for Value |
|-|-|-|
| `SHIP_TYPE` | 0 | `Unavailable`
| `POSITION_FIX_DEVICE` | 0 | `Default: undefined`
| `TIME_STAMP` | 0 | `Not available`
| `DSC_FIRST_TELECOMMAND` and `DSC_SECOND_TELECOMMAND` | 126 | `No information`
| ...

Selebrator, 2022-09-22:
I implemented my suggested Fix in my Java code generator and am happy with the result.
For example my enum derived from `NAV_STATUS` looks like this now.
```java
public enum NavStatus implements N2kEnum {
	UNDER_WAY_USING_ENGINE(0, "Under way using engine"),
	AT_ANCHOR(1, "At anchor"),
	NOT_UNDER_COMMAND(2, "Not under command"),
	RESTRICTED_MANEUVERABILITY(3, "Restricted maneuverability"),
	CONSTRAINED_BY_HER_DRAUGHT(4, "Constrained by her draught"),
	MOORED(5, "Moored"),
	AGROUND(6, "Aground"),
	ENGAGED_IN_FISHING(7, "Engaged in Fishing"),
	UNDER_WAY_SAILING(8, "Under way sailing"),
	HAZARDOUS_MATERIAL___HIGH_SPEED(9, "Hazardous material - High Speed"),
	HAZARDOUS_MATERIAL___WING_IN_GROUND(10, "Hazardous material - Wing in Ground"),
	AIS_SART(14, "AIS-SART"),
	IMPLICIT_UNKNOWN(15, "N2k Unknown Sentinel");

	// ...
}
```

I think these can be derived automatically in canboat's definitions, but I'm not a C macro wizard. My approach would be to just write them all out manually.

keesverruijt, 2022-10-16:
Has been on master since v4.0.0

closed 2022-10-16 by keesverruijt