canboat/canboat #304: Dont render AIS Spare

Selebrator, 2022-09-16:
AIS Spare has the same semantics as NMEA 2000 Reserved. I don't think it makes sense to output one but not the other.

Removing AIS Spare from the output would avoid confusion among those who don't know that spare is just another word for reserved (that's me a couple of weeks ago).

keesverruijt, 2022-09-22:
Actually, they are different as Spare fields should be all zeroes instead of all ones. The big refactor coming will e plain this better!

Selebrator, 2022-09-22:
Sure, they are not all the same. But still a Reserved field and a Spare field carry the same meaning. With both types the creators intend to reserve a field for future use. With both we don't know what a value different from the default value should mean.

Selebrator, 2022-10-05:
Confirming this as fixed on `refactor-analyzer` branch.

I think it's a bit weird, that in json format, the value is a String, if different from 0.
```
$ echo "2022-06-02T15:06:48.074Z,4,129038,5,255,28,01,a0,d9,97,0c,e6,af,dd,04,b8,48,e9,1f,bc,9c,3f,00,00,53,c0,01,ce,8d,00,00,cf,fa,fb" | ./analyzer -json | jq
INFO 2022-10-05T09:10:39.062Z [analyzer] Detected normal format with all frames on one line
{
  "version": "4.0.0-alpha",
  "units": "std"
}
{
  "timestamp": "2022-06-02T15:06:48.074Z",
  "prio": 4,
  "src": 5,
  "dst": 255,
  "pgn": 129038,
  "description": "AIS Class A Position Report",
  "fields": {
    "Message ID": 1,
    "Repeat Indicator": "Initial",
    "User ID": "211278240",
    "Longitude": 8.163735,
    "Latitude": 53.53822,
    "Position Accuracy": "Low",
    "RAIM": "not in use",
    "Time Stamp": "47",
    "COG": 93.3,
    "SOG": 0,
    "Communication State": "53 C0 01",
    "AIS Transceiver information": "Channel A VDL reception",
    "Heading": 208,
    "Rate of Turn": 0,
    "Special Maneuver Indicator": "Not available",
    "Spare": "02",
    "Sequence ID": 251
  }
}

```

keesverruijt, 2022-10-05:
It’s a binary field so it is printed as a set of hex digits, repeating for every byte. Since this is not a number, it needs quotes. JSON only has straight decimal numbers or strings.

Selebrator, 2022-10-05:
Ok, that makes sense. I was not aware, because the Spare Fields I knew of are too small to require hex digits.

closed 2022-10-05 by keesverruijt