canboat/canboat #305: Print warning when reserved field is not empty

Selebrator, 2022-09-16:
I see only 2 ways for a reserved field to not be empty
- The data is faulty
- The field is not reserved anymore

If we had a warning for non-empty reserved fields, users could easily check for this condition. That would either help them with finding errors in their setup; or it would make it easier for canboat contributors to discover new fields.

Same applies to AIS spare fields or any other field that should be a constant.

We could add an option to disable this to reduce io load on weak systems.

Selebrator, 2022-09-16:
For example #100 could have been found by this

Selebrator, 2022-10-05:
Confirming this as fixed on `refactor-analyzer` branch.

```
$ echo "2022-09-20T09:55:27.188Z,2,128259,105,255,8,00,5a,00,ff,ff,00,ff,fe" | ./analyzer -json
{"version":"4.0.0-alpha","units":"std"}
INFO 2022-10-05T09:27:25.474Z [analyzer] Assuming normal format with one line per frame
{"timestamp":"2022-09-20T09:55:27.188Z","prio":2,"src":105,"dst":255,"pgn":128259,"description":"Speed","fields":{"SID":0,"Speed Water Referenced":0.90,"Speed Water Referenced Type":"Paddle wheel","Reserved":"F0 FE"}}
```

keesverruijt, 2022-10-05:
Cool. Do you have a source that actually produces this interesting PGN 128259 or are you just generating test code on purpose?

Selebrator, 2022-10-05:
I have a source for this PGN, but the example is altered by hand in order to test the behavior of analyzer.

closed 2022-10-05 by keesverruijt