canboat/canboat #307: Analyzer not printing PRN, SID, Sats in View for 129540 (Satellites in View)

arks-public, 2022-10-04:
I noticed that the analyzer is not printing certain fields of PGN 129540 (GNSS Sats in View), even though they are present in the raw data. This is probably related to the recent refactoring, perhaps temporary, but I still wanted to point out the missing fields.

For example, the "Sats in View" field and the "PRN" repeating fields are not printed:

```
2022-10-04T21:27:08.327Z 6  35 255 129540 GNSS Sats in View:  SID = Unknown; Mode = Unknown; Elevation 1 = 0.0 deg; Azimuth 1 = 0.0 deg; SNR 1 = 0.00 dB; Range residuals 1 = Unknown; Status 1 = Unknown; Reserved 1 = Unknown
2022-10-04T21:27:08.327Z 6 035 255 129540 : ff ff 01 11 00 00 00 00 00 00 ff ff ff 7f ff
```

I also noticed that the "SID" field is only printed when it is Unknown (0xFF); it disappears if it is provided in the normal range 0-252. As far as I can tell, these are the only 3 fields that go missing.

keesverruijt, 2022-10-05:
This was a bug introduced by the initial refactor (v3.0.0). Now that v4.0.0 is in master (2nd refactor) it should be fixed (I even test for this now in `make tests` !)

Please close this issue if you concur.

arks-public, 2022-10-05:
Yes, I just tried v4.0.0 and it is working, thank you!

closed 2022-10-05 by arks-public