canboat/canboat #309: Reverse engineering of Navico autopilot controls

keesverruijt, 2022-10-05:
(A lot of) work has already been done on this by @htool, see https://github.com/htool/RaymarineAPtoFakeNavicoAutoPilot 

Hans' focus was to be able to use a Raymarine AP with a Navico control head, for this to work it needs to know that the AP is fully commissioned. 

I intend to go the other way:

* Add PGNs for the most common reports like mode (Standby/Auto/Wind/FollowUp).
* Add PGNs for the most common commands like mode / -10 / -1 / +1 / +10.
* Add PGNs that make the control head believe the autopilot is commissioned (if this exists, I think this may not be possible for all control heads.)

TwoCanPlugIn, 2022-10-25:
Was following up a few issues I had raised on SignalK and saw a link to this. Canboat has been an inspiration for the OpenCPN TwoCan plugin, so happy to give something back.

I haven't been able to get the NAC-3 to go into Navigation or No-Drift Mode because I have yet to figure out what other PGN's are required. For example, in Navigation Mode, even though I generate and transmit valid PGN 129283 (XTE), PGN 129284 (Navigation) and PGN 129285 (Route/Waypoint List) messages, they are not sufficient to enable the NAC-3 to engage in Navigation Mode. There are a couple of PGN's sent by a B&G Zeus MFD that appear to be necessary to support Navigation Mode, but as yet, I have not been able to decipher them

Note the NAC-3 behaves as four separate NMEA 2000 devices, Autopilot, NMEA 183 Input, Rudder Feedback and Virtual Rudder Feedback. Similarly the Zeus-2 MFD behaves as several NMEA 2000 devices, MFD, Navigator, Autopilot Controller, GPS.

All of these are Manufacturer Proprietary PGN's, with the first two bytes encoded with the Manufacturer and Industry codes as per usual.

PGN 65305, 
Broadcast by NAC-3 Autopilot. 
Reports the Status and Mode of the Autopilot
These correspond to what is displayed on the Zeus-2 MFD

NAC-3 Autopilot Status 

Sample Raw Data
0x41 0x9f 0x00 0x02 0x10 0x00 0x00 0x00
0x41 0x9f 0x00 0x02 0x02 0x00 0x00 0x00
0x41 0x9F 0x64 0x02 0x10 0x00 0x00 0x00

Byte 2 indicates the Autopilot Model. 0x64 - NAC-3, 0x00 - AC-12
Byte 3 (0x02) indicates Status Report
Byte 4 indicates the Autopilot Status, 0x10 - Enagaged, 0x02- Standby
	
NAC-3 Autopilot Mode

Sample Raw Data
0x41 0x9F 0x64 0x0A 0x40 0x00 0x00 0x40
0x41 0x9F 0x64 0x0A 0x00 0x04 0x00 0x04
0x41 0x9F 0x64 0x0A 0x10 0x00 0x00 0x00
0x41 0x9F 0x64 0x0A 0x00 0x01 0x00 0x00
0x41 0x9F 0x64 0x0A 0x08 0x00 0x00 0x00

Byte 2 indicates the Autopilot Model. 0x64 - NAC-3, 0x00 - AC-12
Byte 3 (0x0A) indicates Mode Report
Bytes 4,5 indicate the Autopilot Mode 
GPS = 0x0040
Wind = 0x0400
Heading = 0x0010
No Drift = 0x0100
Standby = 0x0008

PGN 65431 
Broadcast by NAC-3 Autopilot
Reports the heading or apparent wind angle dependent on the autopilot mode

Sample Raw Data
0x41 0x9F 0xFF 0xFF 0x03 0xFF 0x7F 0x3D 
0x41 0x9F 0xFF 0xFF 0x02 0xFF 0xFF 0xFF
0x41 0x9F 0xFF 0xFF 0x0B 0xFF 0xD1 0x24

Byte 4 reflects the Autopilot Mode (or more correctly), the type of angle
Wind - 0x03
GPS - 0x0A
Heading - 0x02
No Drift - 0x0B

Byte 5 and Byte 6 encode the angle expressed as radians with a resolution of 0.0001

PGN 130856
Broadcast by NAC-3 as some sort of alarm message. 
Noticed they were transmitted by the NAC-3 during Heading Hold and Nav Mode whilst the vessel was stationary

Sample Raw Data
0x41 0x9F 0x10 0x00 0x10 0x01 0x4C 0x6F 0x77 0x20 0x62 0x6F 0x61 0x74 0x20 0x73 0x70 0x65 0x65 0x64 

Decoded Message
0x41 0x9F Manufacturer & Industry Code
0x10 0x00 Some sort of command or message ID
0x10 0x01 Length 20, String Encoded as ASCII
[... 81 lines cut ...]

The PGN broadcast from the MFD which seems to be the key to engaging the NAC-3 in GPS mode is PGN 139820

Sample Raw Data
0x7D 0x99 0x02 0x41 0x00 0x00 0x00 0x00
0x03 0x41 0x36 0x87 0x07 0x00 0x00 0x41
0x00 0x00 0x00 0x00 0x01 0x41 0xC0 0xE1
0x6C 0x30 0x06 0x41 0x40 0x62 0xC0 0x2E
0x04 0x41 0x40 0x44 0x78 0x32 0x05 0x41
0x80 0xE8 0x02 0x00

It doesn't match any of the formats in Canboat, but does seem to be a series of key/value pairs.

0x7D 0x99 
0x02 0x41 0x00 0x00 0x00 0x00
0x03 0x41 0x36 0x87 0x07 0x00 
0x00 0x41 0x00 0x00 0x00 0x00 
0x01 0x41 0xC0 0xE1 0x6C 0x30
0x06 0x41 0x40 0x62 0xC0 0x2E
0x04 0x41 0x40 0x44 0x78 0x32
0x05 0x41 0x80 0xE8 0x02 0x00

Decoding each of the values as int, unsigned int, short, unsigned short and converting to latitude, longitude, speed, distance, angle time, doesn't correspond to anything that makes sense.

Hope this is of some help.

keesverruijt, 2022-10-25:
Thanks. I will be out sailing for a few months, but aim to pick this up after that.

TwoCanPlugIn, 2022-10-25:
No probs,
I'm the reverse, packing up the boat for winter before heading home.

BTW, Apologies for posting in the wrong spot, but posted an analysis of PGN 130850 & 130851 from the data that Teppo provided in https://github.com/canboat/canboat/issues/272

Enjoy your sailing.

TwoCanPlugIn, 2022-10-26:
BTW, Dyslexic fingers caused typo.
"The PGN broadcast from the MFD which seems to be the key to engaging the NAC-3 in GPS mode is PGN 139820 " should read
"The PGN broadcast from the MFD which seems to be the key to engaging the NAC-3 in GPS mode is PGN **130824**

A little bit of further analysis, I think the value of key 5, corresponds to distance to waypoint. Really needs to be captured while underway as any values such as sog,, time to waypoint etc. are meaningless when stationary.

TwoCanPlugIn, 2023-09-09:
Revisited this.
All of the above regarding engaging the autopilot, including Navigation Mode, seem to be correct, at least on a NAC-3.

All that the NAC-3 requires is valid PGN 129283 (XTE) and PGN 129284 (Navigation Data) messages with the following weird observation, in that Navigation mode could only be reliably selected when the Device Class and Function class were set to 50 & 130 respectively. This made little sense to me.

Even if I set the other fields in PGN 60928 (address claim), such as the manufacturer code and the fields in PGN 126996 (Product Information) to match those from a B&G Zeus MFD, it seemed to give random results. Perhaps I was holding my tongue wrong.

keesverruijt, 2023-09-09:
Note that I have just merged a number of fixes to the "Simrad: Key Value" PGN 130845 which is now mostly readable. Not really AP control yet, but most of the data generated by changing settings on my AP48 are now supported.

sbender9, 2024-06-15:
Does anyone have a log of n2k data when using the autopilot?

ersingencturk, 2024-06-15:
> Does anyone have a log of n2k data when using the autopilot?

i capture these recently. when i playback it successfully controls the boat:

1 starboard 

2024-06-03-17:57:54.174,2,130850,0,255,8,20,0c,41,9f,01,ff,ff,0a
2024-06-03-17:57:54.175,2,130850,0,255,8,21,1a,00,03,af,00,00,ff

10 starboard
2024-06-03-13:55:23.757,2,130850,0,255,8,00,0c,41,9f,01,ff,ff,0a
2024-06-03-13:55:23.757,2,130850,0,255,8,01,1a,00,03,d1,06,00,ff

1 port

2024-06-03-17:57:43.925,2,130850,0,255,8,00,0c,41,9f,01,ff,ff,0a
2024-06-03-17:57:43.926,2,130850,0,255,8,01,1a,00,02,af,00,00,ff

10 port

2024-06-03-13:55:25.156,2,130850,0,255,8,20,0c,41,9f,01,ff,ff,0a
2024-06-03-13:55:25.157,2,130850,0,255,8,21,1a,00,02,d1,06,00,ff

sbender9, 2024-06-15:
That helps, but I need a complete session that includes changing navigation modes and includes all the messages coming from the AP.

tkurki, 2024-06-17:
@sbender9 dropped you a url for a capture file. If anybody else is interested ping me.

TwoCanPlugIn, 2024-06-18:
I would be interested as well, if only to verify what I am already parsing.

Haven't been able to retest this year, but at the end of last year, pretty confident that I could control a NAC-3 (change mode, heading, wind angle) and generate the correct PGN 129283 & 129284 & Keep Alive messages.

tkurki, 2024-06-18:
@TwoCanPlugIn drop me an email at teppo.kurki@iki.fi and I'll reply with link to the capture file.

TwoCanPlugIn, 2024-07-02:
OK, Here goes. I was able to "borrow" neighbour's boat on weekend to capture some more data. It agrees what I had written earlier and at least for scenario 1. Controlling a NAC-3 with another device, it confirms what I was able to do at the end of last season. (My problem is how to package it for OpenCPN, either include it in TwoCan, develop a separate plugin that uses TwoCan, or a separate plugin that uses OpenCPN's native NMEA 2000 support. However that's my problem, not yours).

Regarding the other scenarios, 2. Commissiong, is going to require a lot of configuration & analysis, and for 3. Using something like a MFD or Triton 2 to control a PyPilot I think would also be difficult. I also doubt whether someone who invests in a PyPilot (hobbyist, tinkerer) is going to spend a large sum of money on a Triton2?

The setup, or at least the captures, which I've filtered to exclude the noise.
3. NAC-3 Autopilot. (not to be confused with NAC-3 Pilot Controller, NMEA 0183 Gateway, Rudder Feedback)
5. Zeus Pilot Controller
12. Zeus Navigator
18. Zeus Display

For Autopilot functions, the capture is between 3 & 5. and the order is the following:
Standby->Heading Hold->Port 10->Starboard 10-> Port 1-> Port 1-> Starboard 1-> Starboard 1->Standby
Standby->Wind->Port 10->Starboard 10-> Port 1-> Port 1-> Starboard 1-> Starboard 1->Standby
Standby->No Drift->Port 10->Starboard 10-> Port 1-> Port 1-> Starboard 1-> Starboard 1->Standby
Standby->Nav->Standby Note there are two Nav messages, the Zeus needs a confirmation to start in Nav mode.

Important PGNS are 130850 (Event Command), 130851 (Event Reply), 65305 (SImnet Autopilot Status), 65341 (Simnet Angle). Also important is he use of 65305 as a Keep lAlive from the Pilot Controller to the Autopilot.

As moored alongside, a few "Low Boat Speed" alarms.

BTW, A couple of additional Alram messages. (PGN 130856 contains the code & text, 130850 contains the error code).
0x15 - No Nav Data
0x16 - Compass Data missing.
0x22 - No autopilot computer

For Nav mode, the capture is between 12 and 3.
It was simply a matter of selecting a route (in this case named RRR,, the first waypoint named AAA) and starting it. Evident in PGN 129284.

I was only able to perform the Dockside commissioning, which looks to only configure the NAC-3 Virtual Rudder Feedback (eg. turning wheel to full lock and centering. I wasn't going to play with the other settings (eg. boat displacement, boat type, tacking angle etc.), so perhaps not much use.

The attached text files are just my parsing of some of the PGN's. They kind of make sense to me and at least for PGN's 130850, 130851, 65341 and 65305 it matches what I have been able to use to control the NAC-3.

I guess the one important thing to note, and perhaps one should feel some sympathy towards the developers of the MFD's at Navico. I guess through all of the acquisitions/mergers (Simrad/Robertson, B&G) each autopilot (TP32, AC-12, AC-42 and NAC-2/3), each autopilot has an inconsistent set of commands, so the decoding for the NAC-3 will differ to that of say for example the Simrad TP-32 Tillerpilot.

It would be great to get another capture of Teppo's TP-32, albeit with a description of the modes & commands that were issued.

Regards...
[nac3-zeus-autopilot.log](https://github.com/user-attachments/files/16072432/nac3-zeus-autopilot.log)
[nac3-zeus-dockside.log](https://github.com/user-attachments/files/16072433/nac3-zeus-dockside.log)
[nac3-zeus-nav.log](https://github.com/user-attachments/files/16072434/nac3-zeus-nav.log)
[65305.txt](https://github.com/user-attachments/files/16072940/65305.txt)
[65341.txt](https://github.com/user-attachments/files/16072941/65341.txt)
[130850.txt](https://github.com/user-attachments/files/16072942/130850.txt)
[130851.txt](https://github.com/user-attachments/files/16072943/130851.txt)

TwoCanPlugIn, 2024-07-07:
Oops, forgot to add Non Follow Up mode (NFU).

When autopilot is in Standby.

Again PGN 130850 sent from MFD to Controller. The usual palaver, Byte 2 is the autopilot controller's CAN bus address, Byte 5 is the Command (0x02) and Byte 8 controls whether to move rudder to port or starboard. For the port & starboard commands these are sent continuously while the port or starboard button is pressed.

0x41 0x9F 0x03 0xFF 0xFF 0x02 0x0D 0x00 0xFF 0xFF 0xFF 0xFF - NFU Mode
0x41 0x9F 0x03 0xFF 0xFF 0x02 0x0D 0x00 0x04 0xFF 0xFF 0xFF - NFU Mode Move rudder to Port
0x41 0x9F 0x03 0xFF 0xFF 0x02 0x0D 0x00 0x05 0xFF 0xFF 0xFF - NFU Mode, Move rudder to Starboard

Didn't get the opportunity to test "dodging" when in Nav mode nor auto tacking when in Wind mode.

fatyogi, 2025-01-26:
> > Does anyone have a log of n2k data when using the autopilot?
> 
> i captured these recently. when i playback it successfully controls the boat:
> 
> 1 starboard
> 
> 2024-06-03-17:57:54.174,2,130850,0,255,8,20,0c,41,9f,01,ff,ff,0a 2024-06-03-17:57:54.175,2,130850,0,255,8,21,1a,00,03,af,00,00,ff
> 
> 10 starboard 2024-06-03-13:55:23.757,2,130850,0,255,8,00,0c,41,9f,01,ff,ff,0a 2024-06-03-13:55:23.757,2,130850,0,255,8,01,1a,00,03,d1,06,00,ff
> 
> 1 port
> 
> 2024-06-03-17:57:43.925,2,130850,0,255,8,00,0c,41,9f,01,ff,ff,0a 2024-06-03-17:57:43.926,2,130850,0,255,8,01,1a,00,02,af,00,00,ff
> 
> 10 port
> 
> 2024-06-03-13:55:25.156,2,130850,0,255,8,20,0c,41,9f,01,ff,ff,0a 2024-06-03-13:55:25.157,2,130850,0,255,8,21,1a,00,02,d1,06,00,ff

Hi All,

Just looking to revive this thread. Did someone managed to write a signalk Autopilot provider plugin to control a NAC-3 or other Navico autopilot remotely?

I could only find this Garmin provider plugin - https://github.com/jorgen-k/signalk-autopilot-provider-garmin

torevalley, 2025-09-03:
I'm also interested in support for Simrad pilots.
I have a NAC-3 autopilot computer, with Triton2 MFD and Pilot controller.
The Pilot is as far as I understand using a separate NMEA2000 GPS/compass.

The Triton2 MFD reports these devices:
Name                                             Brand     SW version                         ID
HS60 GPS Compass                       Simrad   12308-001   0.1.00.00         16
NAC-3-NMEA0183                        B&G       1.1.05.00                               0
NAC-3_Autopilot                           B&G       1.1.05.00                               3
NAC-3_Pilot controller                  B&G       1.1.05.00                                1
NAC-3_Rudder   feedback             B&G       1.1.05.00                               4
NAC-3_Virtual rudder feedback    B&G        1.1.05.00                               5
RF25   Rudder feedback                Simrad     1.2.01.02                              6
Triton2 Instrument                        B&G         01000_E   2.2.57.3                8
Triton2   Pilot keypad                    B&G         1100 141730.2                     2

The boat is too big for me to handle single handed (Beneteau 50) but I can try capturing logs on the dock if that helps.
Or anything else that can be useful for bringing this forward.

Our boat has a two helm setup but only controller on port side, we tend to use an iPad and SignalK etc on the starboard helm.
Getting the basic buttons to work from th iPad would be great!!

Tritons manual screenshot as reference:

[image]

torevalley, 2025-09-13:
Some screenshots that might be helpful, or not....
I'm not sure where the blank spots are in interpreting the commands...

Pushing Auto generates this:

[image]

[image]

Between auto and standby there is a flow of these two:

[image]

[image]

+10 degrees generates this

[image]

and -10 

[image]

Let me know how I can be os further assistance!!

sbender9, 2026-05-24:
My signalk autopilot plugin is working great with the NAC-3. 

Also have confirmation from others users with older autopilots. 

One issue is that we have to spoof an existing device to get it to work with some devices. There is some authentication or handshake going on that we have not found...

closed 2026-06-27 by keesverruijt