canboat/canboat #312: Add references to downstream projects

keesverruijt, 2022-10-05:
Can those downstream users that are able to please add themselves with a short description and link to your project, so I can add these to the documentation?

keesverruijt, 2022-10-05:
@aldas, @meaning-matters, @sbender9, @inquestruss, and others ...

meaning-matters, 2022-10-05:
I have no link to my project (yet), but am interested to see that of others.

aldas, 2022-10-05:
About 1 year ago I started working on Go lang library https://github.com/aldas/go-nmea-client that would be able to read in Canboat PGN database (runtime). Work in that repo is/was in its very early stage and was in "abandoned" state for some time as the need for N2K was not that important (that time seemed that we can get away with NMEA0183). I am planning to finish that work as the need for N2K has risen again (we are planning to use Actisense N2K-1 devices for monitoring N2K and J1939 busses).

It is not worth adding to link to it yet.

mbj4668, 2023-03-10:
I have a library for decoding NMEA 2000 in Erlang, https://github.com/mbj4668/n2k.   I use it on my boat for various custom projects, and also for off-line (sailing performance) analysis.

It is now updated to use the new `canboat.xml` - thank you for this new format, it is an improvement over the previous format!   I use the XML file to generate Erlang code, which is then compiled.

negrusti, 2025-01-08:
https://github.com/negrusti/NMEA2000-Analyzer
NMEA2000 Analyzer for Windows, in early stages of development but somewhat functional.
Utilizes your JSON definitions but no code.

keesverruijt, 2025-01-08:
Thanks I added them to the Wiki.

keesverruijt, 2026-11-12:
All three projects posted here (go-nmea-client, mbj4668/n2k, NMEA2000-Analyzer) are listed in the downstream survey, docs/downstream-survey-2026-10.html, which is linked from the docs site. Closing; thanks for the pointers.

closed 2026-11-12 by keesverruijt