canboat/canboat #313: Electrical Power fields with offset -2000000000 or very large range are correct?

meaning-matters, 2022-10-07:
This is a question.

While scrolling through the current 4.1 JSON, I noticed electrical power fields of 32 bits with an offset of -2000000000 (without `Resolution` or range values).  Others had `RangeMax` of 4294967293 with `Resolution` of 1.

It seems odd to me that 32-bit field is used with a resolution of 1 (either implicitly of explicitly) with an offset of -2 tera Watt or a range up to 4 tera Watt.

Is this correct?

keesverruijt, 2022-10-07:
The offset is correct, but they should have a Range (-20000000000 .. +2294967295) and Resolution 1. Thanks for catching this. This is different from a normal 2’s complement 32 bit integer with range -2147483648 .. +2147483647.

These PGNs originate from ISO, so they are _weird_ (only place that Excess K notation is used, range is a little exorbitant for a yacht).

keesverruijt, 2022-10-07:
Nononono the offset presence indicates the field is using excess k instead of 2’s complement. So the presence has meaning.

meaning-matters, 2022-10-07:
Thanks.  Assuming this, like many many other things, was documented, where could I have read this?

keesverruijt, 2022-10-07:
Getting passive aggressive? Remember it is not my fault that N2K is so complicated.

As it happens, I am explaining exactly this in the canboat.xml + xsl. Go to these pgns with an offset and click on the word ‘offset’. 

The presence of the offset field makes it show this link.

I am not the worlds best xml tooling expert, I think there is a tool (file type) that allows me to explain xml files more directly. Any help with this would be very much appreciated.

meaning-matters, 2022-10-08:
No not passive aggressive at all.  I am pretty new to CanBoat and N2K.

I felt stupid being surprised to find out about Excess-K.  My first sentence was an attempt to kindly ask if there was a quick way to avoid more silliness from my side.

I truly appreciate the large amounts of references, documentation, and descriptions.

keesverruijt, 2022-10-08:
See `canboat.xsd`.

closed 2022-10-15 by keesverruijt