canboat/canboat #315: pgn.h contains 2 identical fallback pgn 130816

aldas, 2022-10-14:
I'm writing n2k decoder that uses Canboat PGN database. For that I am investigating how PGN ranges work in Canboat. I see that `Fallback` field is used for pgn database to be able to print PGNs with some meaningful name for range of PGNS and from `searchForUnknownPgn` I see how field it is used.

https://github.com/canboat/canboat/blob/02e730f2ce5651a83782a9a9e5b976c0caeb2451/analyzer/pgn.c#L69-L73

now in `pgn.h` there are two fallback for same pgn 130816

https://github.com/canboat/canboat/blob/02e730f2ce5651a83782a9a9e5b976c0caeb2451/analyzer/pgn.h#L4917

and little later
https://github.com/canboat/canboat/blob/02e730f2ce5651a83782a9a9e5b976c0caeb2451/analyzer/pgn.h#L5204

Is it for some specific reason or one of them was added by accident?

keesverruijt, 2022-10-14:
> Is it for some specific reason or one of them was added by accident?

It is a bug, the 2nd one will be removed.

> p.s. docs/canboat.json does not contain Fallback PGNs but XML does - is it meant to be like that?

They are present as an artifact so I can do a fallback during analysing, I don't think they are really useful but I do also use them in the XML to show section headings in the XSL processed HTML documentation version. Do you want them in the JSON, then I can add them?

aldas, 2022-10-14:
Thanks for the answer.  

It seems that for my needs "fallback" pgns are not needed as I decided just to error out when raw message with unknown PGN is encountered and let decoder return only "known" stuff. that way I do not have to deal with complexity of "what is real and what is pseudo pgn"

commit b79892b4 (2022-10-16): Fix #315: Remove superfluous fallback PGN
closed 2022-10-16 by keesverruijt