canboat/canboat #316: PGN 127513 missing "Temperature Coefficient" field

arks-public, 2022-10-14:
The definition for PGN 127513 no longer includes the "Temperature Coefficient" field, I'm guessing this was an unintended regression?

It seems like this commit introduced the change to the XML files and I guess it flowed out from there: https://github.com/canboat/canboat/commit/9e7643d010bc080019f3632eb6f368e379beca80

Example output in v4.2.2 (no "Temperature Coefficient" field):

```
2022-10-14T16:12:30.855Z 6 120 255 127513 Battery Configuration Status:  Instance = 0; Battery Type = Unknown; Supports Equalization = No; Nominal Voltage = Unknown; Chemistry = Li; Capacity = 20 Ah; Peukert Exponent = Unknown; Charge Efficiency Factor = 0 %
2022-10-14T16:12:30.855Z 6 120 255 127513 : 00 cf 1f 14 00 ff 00 62
```

keesverruijt, 2022-10-15:
Thanks Tim, fix coming up.

keesverruijt, 2022-10-15:
@arks-public have you seen the long comment in `pgn.h`? Do you have access to the individual frames on your N2K bus? In that case we can determine whether this is a "fast packet" PGN or not...

arks-public, 2022-10-15:
Hi Kees, yes I noticed that comment in `pgn.h`. This PGN is also tricky for me because unfortunately, I do not have any official devices that output this PGN, only my own device.

But thankfully, I have the N2KAnalyzer program which is based on official N2K definitions. And what I found by varying the data that I sent, is that the packet is exactly 8 bytes long and it must be sent as a fast packet. When I sent an identical single frame, it would not appear in N2KAnalyzer at all. The details are in https://github.com/canboat/canboat/issues/143#issuecomment-1150390465 but I copied below:

"I used N2KAnalyzer v3.2.0.3 to analyze Battery Configuration Status (127513) given different data lengths and data contents.

Here is what I found:

- The order and lengths of the data fields in pgn.h are correct (as of last commit October 2021)
- Seems like this must be a fast packet, even though there are only 8 bytes of data. If I send more than 8 bytes, the extra data is shown to be unused. But if I send the 8 bytes of data as a single frame, the message doesn't show up in N2KAnalyzer.
- The "Supports Equalization" field corresponds with LOOKUP_YES_NO, so it doesn't need a custom lookup table
- For the "Chemistry" field I observed a slightly different lookup table: "0=Lead Acid,1=LiIon,2=NiCad,3=ZnO,4=NiMH"
- The other lookup fields (Battery Type and Nominal Voltage) are correct
- Capacity and Peukert Exponent are unsigned fields
- Temperature Coefficient and Charge Efficiency Factor are signed fields in increments of 1% (so these fields range from -128% to 124% since the 3 missing values having special meanings)"

keesverruijt, 2022-10-15:
👍

commit a67c48d9 (2022-10-16): Fix #316: PGN 127513 field Temperature Coefficient

Start with testing stable JSON output of analyzer.
closed 2022-10-16 by keesverruijt