canboat/canboat #325: Automatic changelog creation

tkurki, 2022-10-18:
I've been pretty happy with PR-based changelog creation. In many Signal K repos it works so that
Prerequisites:
- Github repo's issue/pr labels are modified to include the values one wants to use
- (needed GH actions are added to repo)

Then:
- PRs are labeled (feature, fix, chore, other, ignore - how you want the changelog to be grouped)
- GH action checks that PRs have the label
- maintainer creates a GH release (can be automated off a pushed git tag with GH action also)
- GH action creates the changelog from PRs merged since the previous release

The last two release of [sk server] (https://github.com/SignalK/signalk-server/releases) use this method, earlier ones different tooling but essentially the same logic: changelog is created from PRs.

I can add the actions if this is something @keesverruijt you'd be interested in.

keesverruijt, 2022-10-18:
It would need careful implementation and operation, there are three things interacting here:

1. Generated XML and JSON files, especially when there are interacting PRs.
2. Creating a release.
3. Changelog.

If we go for GH actions, I would prefer it if the workflow refused straight changes to master other than CI changes. CI changes would be a pain to do via PRs as sometimes I need a lot of attempts at "will this work then to get the builder working?")

Also would be good if the version number of master would always be "unreleased"/"wip" and the versions live on a branch -- either "released" or a branch per version, although I don't see work on multiple branches being needed here.

If you are willing to do this for me, and explain in detail how this then works, great.

keesverruijt, 2022-10-18:
I guess I just proved why this is necessary, I accidentally referred to the wrong issue in a commit message.

tkurki, 2022-10-18:
The definite version information is in version.h, right? Is everything else derived stuff from that? So in master it could be `unreleased`. To create a release you'd decide the next version, update version.h and commit that locally, create a tag and push it. (or even that can be automated, the release GH action would update version.h off the tag).

Re XML and JSON files: I would let the dog wag the tail, have the release process generate them. So they would not be in git, but available as part of the release artifacts. (the default artifact is just the repo contents in a zip file, but can include other files also).

There's also the option to have not only a single, latest version at https://canboat.github.io/canboat/canboat.xml, but like SK spec: https://signalk.org/specification/latest/ redirects to the latest version 1.7.0, but all the published versions are still available and addressable, for example https://signalk.org/specification/1.5.0/doc/. The release process could automate all this.

aldas, 2022-10-20:
> Re XML and JSON files: I would let the dog wag the tail, have the release process generate them

This would have poor "user experience" as you would loose git history (blame) for these files. For example IDEA products like Clion you can select some lines and see all changes for these lines in history. I guess VScode etc have similar features.  This is useful for code and "configuration" for looking where things went wrong in history. Seeing uncommitted changed and stepping is also useful to notice broken things.

We have been doing this 5+ years for over 40+ vessels at work and it is useful to see where some parameter value has changed in our configurations.

As long these changes are most of the time minimal having XML/JSON "binaries" should not be a problem.

tkurki, 2022-10-20:
Git diffs and history is a good point, skip that idea.

keesverruijt, 2022-10-21:
> The definite version information is in version.h, right? Is everything else derived stuff from that? 

Correct!

> So in master it could be `unreleased`. To create a release you'd decide the next version, update `version.h` and commit that locally, create a tag and push it. 

Maybe I'm stupid, but on what branch?  I need to push the version update commit -- a tag without a known commit id has no purpose. So if I do this on master, we're back at the current situation. 

> (or even that can be automated, the release GH action would update version.h off the tag).

Super cool.

> XML and JSON files: I would let the dog wag the tail, have the release process generate them

I'm with Matti here, can we have the actions make sure they are up-to-date? Even on PR requests, then it becomes easy to verify the impact of a change.

Should we start with a copy of the current repository so we can test these actions?

tkurki, 2022-10-21:
Probably the easiest if i iterate first in a fork by myself.

keesverruijt, 2022-10-21:
> Probably the easiest if i iterate first in a fork by myself.

Good idea.

aldas, 2022-12-23:
Hi @keesverruijt could you add release assets on release page. You are doing excellent work but these changes to schema are hard to work when there are no reliable way to download some fixed version with  `curl`.   

I have a application that  requires `canboat.json` at start and so far I have gotten away with makefile and `@curl https://raw.githubusercontent.com/canboat/canboat/master/docs/canboat.json -o canboat/testdata/canboat.json`  but this is not very reliable when there are schema changes which my app does not recognize yet.

I tried out github action with canboat fork that upload assests when new release tag is pushed.

For it to work: similar workflow needs to be created. It uses https://github.com/softprops/action-gh-release as action

Example file: `.github/workflows/release.yaml`

```yaml
name: Upload release assets

# this flow will be run only when new tags are pushed that match our pattern
on:
  push:
    tags:
      - "v*"

jobs:
  build:
    runs-on: ubuntu-latest
    steps:
      - name: Checkout
        uses: actions/checkout@v3

      - name: Add release assets
        uses: softprops/action-gh-release@v1
        with:
          files: |
            docs/canboat.json
            docs/canboat.xml
            docs/canboat.xsd
```

And your work flow would be

1. you: tag a new version. `vX.X.X`. Basically anything that starts with `v*` is ok.
2. you: push tag to github
3. github: starts this workflow and uploads these 3 files to release page

Example on my fork: https://github.com/aldas/canboat/releases

maybe @tkurki  has recommendations but at the moment I left changelog out of it and this workflow is not resposible for generating/updating these files. It only uploads them so release would have history for these assets.

tkurki, 2022-12-26:
Instead of blindly pushing whatever files happen to be in git as a release I think there should be a safeguard that checks that the generated files match the source of truth, so PR #362 adds some minimal checks.

And btw in the current master at 61293d7823e60aff934cdb4e79ff4981c6e14b47 canboat.json|xml do NOT match 😄 

One next step would be to have the action check that the version.h and the tag match. Or do it the other way around: any push in master that changes the version in version.h to a value that has no corresponding tag should run tests and if they succeed create the tag & release artifacts.

aldas, 2022-12-26:
Having CI checks for json/xml is a good safeguard but  automatically releasing things is going to be quite punishing -  there are cases when you sometimes want to wait before releasing new version. You might want to wait for X,Y and Z changes to  be ready, or wait for user feedback, wait for end of the month and release everything then etc. 

Looking from naive consumer standpoint - basic necessity is to able to link (download) to something that does not change in time. So maybe release workflow does not need to be that complex and most of these safeguards/checks should go to  continuous integration flow like https://github.com/canboat/canboat/pull/362 does it.  So the problem scope can be divided - release  and CI.

tkurki, 2022-12-28:
>automatically releasing things is going to be quite punishing

I think you misunderstood me, I think we are in agreement. The idea is that the maintainer does just one thing, basically says "publish version so and so". In your steps above it is "tag and push", in my it is "push to master version.h with a new version".

commit b4a9d03f (2022-10-18): Fix #325: pgns: some fields are missing resolution value