canboat/canboat #334: Parsing of packet that are missing byte for some of fields

aldas, 2022-10-28:
PGN 127506 was changed in  #244 and field `ELECTRIC_CHARGE_UFIX16_AH("Amp Hours"),` was added thus meaning this packet should be 1 Field longer (bit 72 to bit 88). https://github.com/canboat/canboat/blob/fc66882597bad9066395fb30ae1f09e3dc09a76a/analyzer/pgn.h#L2785

as json
```json
    {
      "Order": 8,
      "Id": "remainingCapacity",
      "Name": "Remaining capacity",
      "BitLength": 16,
      "BitOffset": 72,
      "BitStart": 0,
      "Unit": "Ah",
      "Resolution": 1,
      "Signed": false,
      "RangeMin": 0,
      "RangeMax": 65533,
      "FieldType": "NUMBER",
      "PhysicalQuantity": "ELECTRICAL_CHARGE"
    }
```

Now there are a lot of samples that have this packet only 9 bytes. `grep "127506" samples/*`

For example: `dirona-actisense-serial.raw` has lines like `2016-02-28T19:57:06.432Z,6,127506,176,255,9,eb,01,00,64,ff,83,3f,a5,00`

```bash
echo "2016-02-28T19:57:06.432Z,6,127506,176,255,9,eb,01,00,64,ff,83,3f,a5,00" | ./rel/linux-x86_64/analyzer -json -debug -raw -si
```

Output:
```
{"version":"4.6.1","units":"si","showLookupValues":false}
INFO 2022-10-28T06:34:04.573Z [analyzer] Detected normal format with all frames on one line
{"timestamp":"2016-02-28T19:57:06.432Z","prio":6,"src":176,"dst":255,"pgn":127506,"description":"DC Detailed Status","fields":{{"SID":235,"bytes"="EB"},{"Instance":1,"bytes"="01"},{"DC Type":"Battery","bytes"="00"},{"State of Charge":100,"bytes"="64"},{"State of Health":null,"bytes"="FF"},{ "Time Remaining": "270:59:00","bytes"="83 3F"},{"Ripple Voltage":1.65,"bytes"="A5 00"}}}
2016-02-28T19:57:06.432Z 6 176 255 127506 : eb 01 00 64 ff 83 3f a5 00
```

There are samples of 2 raw can/nmea frames that make of fast packet (somehow have 9 bytes)
https://github.com/canboat/canboat/blob/fc66882597bad9066395fb30ae1f09e3dc09a76a/samples/merrimac-ydwg-2020.txt#L1049-L1052

Question/Problem:

Although there should be 1 field more in packet the analyzer does not signal that this field parsing/decoding was not successful as this packet has bits 72 to 80  but last byte (bits 80-88) are missing.

* Should there be warning at least?
* Is this field actually 16 bits or 8 bits?

keesverruijt, 2022-10-31:
Yes, this is the case for N2K. I am 100% sure that the original definition for PGN 127506 did not have the `Remaining Capacity` field, *and* that it does have it now. But should the analyzer complain for every field that is "not there" at the end of the field list? I don't think so. Just like it just doesn't print fields that have _all ones_.

So I think the current behaviour is OK: it generates a message that contains all the fields that it can find. Missing ones are not printed.

keesverruijt, 2022-10-31:
Maybe we should have an attribute that says something like "NMEA Version Dependent"?

aldas, 2022-11-01:
I see that you have added `pgn.MinLength` field to the schema. Maybe `pgb.MinFieldCount` would be a way to indicate that some fields at the end may be missing.  Adding attribute to `field` seems little bit to excessive as most of the fields do not need it. 

p.s. I do not know if you receive messages from canboat discussions. If you do not - maybe you want to consider clicking 
[image]

p.s.s. maybe it would be worth creating separate discussion  "All related to the PGN schema" where thought/questions about schema could be gathered. 

p.s.s.s. at the landing page of repo, at the right corner there is "About sections" You can add tags there like "n2k", "nmea2000" etc so people searching stuff from Github would find Canboat by these keywords.

commit d0bf6139 (2022-11-14): Fix #334: print error on missing fields in repeating set
closed 2022-11-14 by keesverruijt