canboat/canboat #338: Printing PGN 126208 (Request/Reply) gives errors but seemed to work with older versions

aldas, 2022-11-04:
I am trying to understand how "VARIABLE" fields work. These field types are found with PGN 126208. It seems that with current version analyzer does not print most of the samples I provide to it and I do not understand why. So I checked out some older version to see if it could be some kind of regressions. This is what I see:

Analyzing PGN 126208 with v2.0.0 (commit bb41e9c3941e89840f6ffeee746861ef0fc9ae76)
```bash
echo "2020-04-19T00:35:55.571Z,2,126208,0,67,21,01,16,f0,01,ff,01,02,0e,01,59,44,3a,56,4f,4c,55,4d,45,20,36,30" | ./rel/linux-x86_64/analyzer -json -debug -raw -si
```

Output
```
{"version":"2.0.0","units":"si"}
INFO 2022-11-04T08:10:38.095Z [analyzer] Detected normal format with all data on one line
INFO 2022-11-04T08:10:38.095Z [analyzer] New PGN 126208 for device 0 (heap 11157 bytes)
{"timestamp":"2020-04-19T00:35:55.571Z","prio":2,"src":0,"dst":67,"pgn":126208,"description":"NMEA - Command group function","fields":{"Function Code":"Command","PGN":126998,"# of Parameters":1,"list":[{"Parameter":259 44 3a 56 4f 4c 55 4d 45 20 36 30 ,"Value":"YD:VOLUME 60"}]}}
2020-04-19T00:35:55.571Z 2 000 067 126208 : 01 16 f0 01 ff 01 02 0e 01 59 44 3a 56 4f 4c 55 4d 45 20 36 30
```

Same packet with current version
```
{"version":"4.7.0","units":"si","showLookupValues":false}
INFO 2022-11-04T08:13:15.629Z [analyzer] Detected normal format with all frames on one line
ERROR 2022-11-04T08:13:15.629Z [analyzer] Field Value: cannot derive variable length for PGN 15734273 field # 21
ERROR 2022-11-04T08:13:15.629Z [analyzer] PGN 126208 analysis error
2020-04-19T00:35:55.571Z 2 000 067 126208 : 01 16 f0 01 ff 01 02 0e 01 59 44 3a 56 4f 4c 55 4d 45 20 36 30
```

Was older output wrong and now it is fixed and that error is valid?  or older output is correct?

--------
Another example for differences.  In current version we have file `samples/126208.raw`  I'll copy first 4 lines from it for example

```
2020-04-19T00:35:55.571Z,2,126208,0,67,21,01,16,f0,01,ff,01,02,0e,01,59,44,3a,56,4f,4c,55,4d,45,20,36,30
2016-02-28T19:57:41.000Z,3,126208,40,72,08,01,0d,f2,01,f8,01,03,01
2016-04-09T16:41:09.092Z,6,126208,25,32,6,02,01,ff,01,01,00
2016-02-28T19:57:42.145Z,3,126208,24,1,9,02,08,ff,01,03,05,00,03,f0
```

output for it with current version is:
```
cat samples/126208_new.raw | ./rel/linux-x86_64/analyzer -json -debug -raw -si
{"version":"4.7.0","units":"si","showLookupValues":false}
INFO 2022-11-04T08:30:27.136Z [analyzer] Detected normal format with all frames on one line
ERROR 2022-11-04T08:30:27.136Z [analyzer] Field Value: cannot derive variable length for PGN 15734273 field # 21
ERROR 2022-11-04T08:30:27.136Z [analyzer] PGN 126208 analysis error
2020-04-19T00:35:55.571Z 2 000 067 126208 : 01 16 f0 01 ff 01 02 0e 01 59 44 3a 56 4f 4c 55 4d 45 20 36 30
ERROR 2022-11-04T08:30:27.136Z [analyzer] Field Value: cannot derive variable length for PGN 15863041 field # 8
ERROR 2022-11-04T08:30:27.136Z [analyzer] PGN 126208 analysis error
2016-02-28T19:57:41.000Z 3 040 072 126208 : 01 0d f2 01 f8 01 03 01
{"timestamp":"2016-04-09T16:41:09.092Z","prio":6,"src":25,"dst":32,"pgn":126208,"description":"NMEA - Acknowledge group function","fields":{"Function Code":{"2","bytes":"02"},"PGN":{"value":130817,"bytes":"01 FF 01"},"PGN error code":{"value":"PGN not supported","bytes":"01","bits":"0001"},"Transmission interval/Priority error code":{"value":"Acknowledge","bytes":"00","bits":"0000"},"Number of Parameters":{"value":0,"bytes":"00"}}}
2016-04-09T16:41:09.092Z 6 025 032 126208 : 02 01 ff 01 01 00
{"timestamp":"2016-02-28T19:57:42.145Z","prio":3,"src":24,"dst":1,"pgn":126208,"description":"NMEA - Acknowledge group function","fields":{"Function Code":{"2","bytes":"02"},"PGN":{"value":130824,"bytes":"08 FF 01"},"PGN error code":{"value":"Access denied","bytes":"03","bits":"0011"},"Transmission interval/Priority error code":{"value":"Acknowledge","bytes":"00","bits":"0000"},"Number of Parameters":{"value":5,"bytes":"05"},"list":[{"Parameter":{"value":"Acknowledge","bytes":"00","bits":"0000"}},{"Parameter":{"value":"Acknowledge","bytes":"00","bits":"0000"}},{"Parameter":{"value":"Parameter out of range","bytes":"03","bits":"0011"}},{"Parameter":{"value":"Acknowledge","bytes":"00","bits":"0000"}}]}}
2016-02-28T19:57:42.145Z 3 024 001 126208 : 02 08 ff 01 03 05 00 03 f0
```

Output with commit v2.0.0 (commit bb41e9c3941e89840f6ffeee746861ef0fc9ae76) 
```
cat samples/126208_new.raw | ./rel/linux-x86_64/analyzer -json -debug -raw -si
{"version":"2.0.0","units":"si"}
INFO 2022-11-04T08:29:45.396Z [analyzer] Detected normal format with all data on one line
INFO 2022-11-04T08:29:45.397Z [analyzer] New PGN 126208 for device 0 (heap 11157 bytes)
{"timestamp":"2020-04-19T00:35:55.571Z","prio":2,"src":0,"dst":67,"pgn":126208,"description":"NMEA - Command group function","fields":{"Function Code":"Command","PGN":126998,"# of Parameters":1,"list":[{"Parameter":259 44 3a 56 4f 4c 55 4d 45 20 36 30 ,"Value":"YD:VOLUME 60"}]}}
2020-04-19T00:35:55.571Z 2 000 067 126208 : 01 16 f0 01 ff 01 02 0e 01 59 44 3a 56 4f 4c 55 4d 45 20 36 30
INFO 2022-11-04T08:29:45.397Z [analyzer] New PGN 126208 for device 40 (heap 22312 bytes)
{"timestamp":"2016-02-28T19:57:41.000Z","prio":3,"src":40,"dst":72,"pgn":126208,"description":"NMEA - Command group function","fields":{"Function Code":"Command","PGN":127501,"Priority":"Leave priority unchanged","# of Parameters":1,"list":[{"Parameter":3,"Value":"On"}]}}
2016-02-28T19:57:41.000Z 3 040 072 126208 : 01 0d f2 01 f8 01 03 01
INFO 2022-11-04T08:29:45.397Z [analyzer] New PGN 126208 for device 25 (heap 33467 bytes)
{"timestamp":"2016-04-09T16:41:09.092Z","prio":6,"src":25,"dst":32,"pgn":126208,"description":"NMEA - Acknowledge group function","fields":{"Function Code":"Acknowledge","PGN":130817,"PGN error code":"PGN not supported","Transmission interval/Priority error code":"Acknowledge","# of Parameters":0}}
2016-04-09T16:41:09.092Z 6 025 032 126208 : 02 01 ff 01 01 00
INFO 2022-11-04T08:29:45.397Z [analyzer] New PGN 126208 for device 24 (heap 44622 bytes)
{"timestamp":"2016-02-28T19:57:42.145Z","prio":3,"src":24,"dst":1,"pgn":126208,"description":"NMEA - Acknowledge group function","fields":{"Function Code":"Acknowledge","PGN":130824,"PGN error code":"Access denied","Transmission interval/Priority error code":"Acknowledge","# of Parameters":5,"list":[{"Parameter":"Acknowledge"},{"Parameter":"Acknowledge"},{"Parameter":"Parameter out of range"},{"Parameter":"Acknowledge"},{"Parameter":"Acknowledge"}]}}
2016-02-28T19:57:42.145Z 3 024 001 126208 : 02 08 ff 01 03 05 00 03 f0
```

aldas, 2022-11-04:
Another example - second line can only be printed when first line exists. When I try second line alone I see this:

Current version:
```
echo "2016-02-28T19:57:41.000Z,3,126208,40,72,08,01,0d,f2,01,f8,01,03,01" | ./rel/linux-x86_64/analyzer -json -debug -raw -si{"version":"4.7.0","units":"si","showLookupValues":false}
INFO 2022-11-04T08:36:40.251Z [analyzer] Assuming normal format with one line per frame
2016-02-28T19:57:41.000Z 3 040 072 126208 : 01 0d f2 01 f8 01 03 01
```

With v2.0.0 (commit bb41e9c3941e89840f6ffeee746861ef0fc9ae76)
```
echo "2016-02-28T19:57:41.000Z,3,126208,40,72,08,01,0d,f2,01,f8,01,03,01" | ./rel/linux-x86_64/analyzer -json -debug -raw -si
{"version":"2.0.0","units":"si"}
INFO 2022-11-04T08:37:45.495Z [analyzer] Assuming normal format with one line per packet
INFO 2022-11-04T08:37:45.495Z [analyzer] New PGN 126208 for device 40 (heap 11155 bytes)
{"timestamp":"2016-02-28T19:57:41.000Z","prio":3,"src":40,"dst":72,"pgn":126208,"description":"Unknown single-frame addressed"}
2016-02-28T19:57:41.000Z 3 040 072 126208 : 01 0d f2 01 f8 01 03 01
```

keesverruijt, 2022-11-04:
It’s highly likely this is a regression. I am prepping a boat to sail from Panama to Galapagos so it will be a while before I can delve into this.

commit 3bac58f1 (2022-11-04): Fix #338: pgn 129608 indirect refpgn
closed 2022-11-04 by keesverruijt