canboat/canboat #341: Sample lowrance-65285.raw structure inconsistent with definitions

inquestruss, 2022-11-06:
65285 is of type single. 2 matching records are defined in canboat.json, one for Lowrance temperature, the other Airmar boot state acknowledgement. The Lowrance variant has a length of 5 bytes. The sample file has 9 data bytes per line and a length byte of 9, which implies Fast.

 Each line contains the correct data in the first 5 bytes, 3 bytes of FF padding, and a 9th byte with various values.

keesverruijt, 2023-02-17:
See #381.

inquestruss, 2023-02-27:
Got it, and agree with your proposal.

 I primarily use log files to validate the implementation we're building. We don't correctly recognize the sample file "lowrance-65285.raw". Looking at the data it appears to be a Lowrance temperature PGN, but it should have a length of 5 (2 bytes for manId|reserved|industry, 1 byte for source, 2 bytes for temp), followed by 3 bytes of FF to fill out the 8 byte data. The contents match that, but the data length is 9 and the last byte isn't always ff. I'd guess it's the interface that garbled it, but who knows. 

I'm closing the issue.