canboat/canboat #342: Definition of PGN 130323 inconsistent with sample samples.raw

inquestruss, 2022-11-07:
Running analyzer produces this:
ERROR 2022-11-06T15:08:55.701Z [analyzer] Unhandled string type 72 in PGN
ERROR 2022-11-06T15:08:55.701Z [analyzer] PGN 130323 analysis error
2011-04-25-10:16:42.485 6 049 255 130323 : f0 f1 3a 40 95 0f 16 d2 03 74 1f 11 ab 16 03 25 01 7f 0a fa ff ff fe 03 ed 71 02 48 65 6c 6c 6f 01 02 57 6f 72 6c 64 01
These sequences have start and stop bytes
 02 48 65 6c 6c 6f 01 "Hello"
02 57 6f 72 6c 64 01 "World"
The definition types them with "STRING_LAU"
           "Order":14,
            "Id":"stationId",
            "Name":"Station ID",
            "BitLengthVariable":true,
            "FieldType":"STRING_LAU"
          },
          {
            "Order":15,
            "Id":"stationName",
            "Name":"Station Name",
            "BitLengthVariable":true,
            "FieldType":"STRING_LAU"
          }
It appears they should be "STRING_VAR"

TwoCanPlugIn, 2022-11-08:
I can't comment on the origins of the samples, however a more recent sample in my possession. the Station ID and Station Name are indeed STRING_LAU.

This particular sample is in Yacht Devices format and parsed with my own routines, however, clearly the Station ID and Station Name are encoded with a length and encoding byte (even though in this case both are null strings).

Fast Message Raw Data

0x60 0x1E 0xF0 0x8B 0x4A 0x78 0xEA 0x98
0x61 0x2C 0xB2 0x59 0x5E 0x14 0x58 0x55
0x62 0xEF 0xB8 0x2A 0x01 0x87 0xB4 0xFA
0x63 0xFF 0xFF 0xF4 0x03 0xDB 0x6F 0x02
0x64 0x01 0x02 0x01 0xFF 0xFF 0xFF 0xFF

Data

F0 8B 4A 78 EA 98 2C B2
59 5E 14 58 55 EF B8 2A
01 87 B4 FA FF FF F4 03
DB 6F 02 01 02 01 FF FF
FF FF

]

Decoded Message

Mode: 0
Time (UTC):   01/04/2022 20:47:01
Latitude: 34 10.37
Longitude: -119 13.65
Wind Speed: 5.79
Wind Angle: 264.79
Wind Reference: 2
Wind Gusts: 0.00
Atmospheric Pressure: 1012
Ambient Temperature: 13.20
Station Id:
Station Name:

keesverruijt, 2022-11-08:
Sorry I am about to go sailing for a week and I don’t have access to the source other than a webbrowser 😁 and when we arrive in the Galapagos I don’t know whether we will have internet. 

I think there are at least a few pgns samples are of devices that would probably not produce the same data with recent firmware!

commit 2a57c0c8 (2022-11-14): Fix #342: Remove STRING_VAR

I thik this entire type was 'found' through incorrect samples. There is no proof anymore
that this type of string exists.
closed 2022-11-14 by keesverruijt