canboat/canboat #350: Missing Ranges

Selebrator, 2022-11-19:
I encountered a bug caused by missing `RangeMin` and `RangeMax` on MMSI fields.
So I checked what field types don't have range properties.

I think these *must* have range properties:
- MMSI
- TIME

I think these *should* have range properties:
- SPARE
- RESERVED
- BINARY

For these range properties don't make sense:
- STRING_LZ
- STRING_FIX
- STRING_LAU
- VARIABLE

All other field types have range properties set always.

Selebrator, 2022-11-19:
I want these range properties, because without them I need to have a function checking if a value is ERROR or UNKNOWN. With the recent [discussion](https://github.com/canboat/canboat/discussions/330) on the existence of ERROR, I would rather not have a custom function for that. I would like to just use a range check and take Canboats position on the issue.

Range properties are also valuable for writing N2k payloads. If I have range properties on a field, I can easily check if a value fits in the field. That is why I would welcome range properties even on fields where the semantics are uncertain. Even If I don't know what a number means, I want to know if it fits.

aldas, 2022-11-19:
You probably should not check SPARE/RESERVED and BINARY fields.   BINARY field could be longer than any number could hold and I think it is mostly used for fields which content is not known. RESERVED/SPARE are padding field, that act as "no value" and should have always same value (all bits 1 or 0). 

There are comments about meaning/usage of these fields:

```json
    {
        "Name":"RESERVED",
      "Description":"Reserved field",
      "EncodingDescription":"All reserved bits shall be 1",
      "Comment":"NMEA reserved for future expansion and/or to align next data on byte boundary"
      },
```
```json
    {
        "Name":"SPARE",
      "Description":"Spare field",
      "EncodingDescription":"All spare bits shall be 0",
      "Comment":"This is like a reserved field but originates from other sources where unused fields shall be 0, like the AIS ITU-1371 standard."
      },
```

```json
    {
        "Name":"BINARY",
      "Description":"Binary field",
      "EncodingDescription":"Unspecified content consisting of any number of bits."
      },
```

keesverruijt, 2022-11-19:
This makes total sense. I just need to special case these as they have non obvious ranges.

keesverruijt, 2022-11-19:
And yes, spare/reserved fields will not get a range, imo this is undefined.

commit f6f801c4 (2022-11-19): Fix #350: Add ranges

Including verification so that this cannot happen again.
Excluded: RESERVED, SPARE, BINARY, VARIABLE and STRING*.
closed 2022-11-19 by keesverruijt