canboat/canboat #359: Add User 1-32 and Remote 0-9

htool, 2022-12-14:
echo "2022-12-12-16:21:10.648,7,130833,0,255,31,7d,99,38,00,ff,fe,31,32,33,34,35,36,37,38,31,32,33,34,35,36,37,38,39,61,62,63,64,65,66,67,00" |socketcan-writer can0
will set short name to '12345678' and long name to '12456789abcdefg'
short is fixed size (8 bytes)
Long is 00 terminated.

'38' is the 'User 1' key id.

htool, 2022-12-14:
```
    {"B&G: User and Remote rename",
     130833,
     PACKET_INCOMPLETE | PACKET_FAST,
     {COMPANY(381),
      LOOKUP_FIELDTYPE_FIELD("Key", 12, BANDG_KEY_VALUE),
      SIMPLE_DESC_FIELD("Length", 4, "Length of field 8"),
      MATCH_FIELD("Command unknown", BYTES(3), 65534, "0x00fffe"),
      STRING_FIX_FIELD("Short name", BYTES(8)),
      STRINGVAR_FIELD("Long name"), END_OF_FIELDS}}
```

I can't get it to work yet...
Note: Long name is encoded without a length, just closing 0x0.

htool, 2022-12-16:
Another detail: the 'fe' field can set the number of decimals to display from 0x0 0 to 0x4 to 4.
I played around with the nearby 00 and ff to see if the unit could be set (that would be very nice), but so far no success.

FarFarSouth, 2022-12-18:
Hi Hans,

I have been using pgn 130824 & 130833 combination with success. Below is the chunk of C# code used to build 130833. I only use it for the 10 remotes.

        public void SendRemoteCaption(int remote, string caption, byte decimals) {
            if (remote >= NavicoN2KExporter.MinRemote && remote <= NavicoN2KExporter.MaxRemote) {
                PGN130824Builder.Code navicoCode = (PGN130824Builder.Code)((ushort)PGN130824Builder.Code.Remote0 + remote);
                byte[] data = new byte[] {
                        
                        0x00, 0x00,                                         //   0 - Reserved manufacturer code (Navico).
                        0x00, 0x00,                                         //  16 - data code (key).
                        0xFF,                                               //  32 - Reserved
                        0xFE,                                               //  40 - Decimals 0x00-0x04
                        0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,     //  48 - Caption on display.
                        0x52, 0x65, 0x6D, 0x6F, 0x74, 0x65, 0x20, 0x00,     // 112 - Name in selection 16 ASCII. Keep "Remote #" inserting # in bit 168
                        0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
                    };

                ProprietaryPGN pgn = new ProprietaryPGN(381, 4, 130833, 3, 0, 255, data);

                pgn.BitString.InsertUnsignedInteger((ulong)navicoCode, 16, 16);
                pgn.BitString.InsertUnsignedInteger(decimals, 40, 8);
                pgn.BitString.InsertASCII(caption, 48, 64);
                pgn.BitString.InsertASCII(remote.ToString(), 168,  8);

                this.SendPGN(pgn);
            }
        }

htool, 2022-12-18:
@FarFarSouth  wrote:
> Hi Hans,
> 
> I have been using pgn 130824 & 130833 combination with success. Below is the chunk of the C# code used to build 130833. I only use it for the ten remotes. It sets the data caption but keeps "Remote #" for the display configuration list.

For me the Remote0-9 do work well. Both short and long name and visible in the selection screen.

Do you know if the unit for these custom fields can be defined as well? I've played around with the 'ff' field, but so far the only thing I noticed was a reboot at some point.

FarFarSouth, 2022-12-19:
No. I haven't found a way to display the units.
BTW... which are the key codes for the User 1-32 custom fields?
Have you tried to handle them like with the remotes 0-9?

htool, 2022-12-19:
> No. I haven't found a way to display the units. BTW... which are the key codes for the User 1-32 custom fields? Have you tried to handle them like with the remotes 0-9?

Yes, works the same way. Codes are [here](https://github.com/canboat/canboat/blob/bg/analyzer/lookup.h#L1990).