canboat/canboat #361: Base64 decoding not working as expected for iKonvert

bikerflyer, 2022-12-20:
Hi,

I have a new SignalK installation installed on Odroid C4 with (Armbian 22.11 Jammy),  similar to RaspberryPi. It is working well except all of the data shown from the Ikonvert USB connection to SignalK is not showing the correct values.

If I use 'ikonvert-serial /dev/ttyUSB0 | analyzer' from the cli the data is consistent with the databrowser in SignalK and shows incorrectly....

This is also consistent with the nodejs version's output from /usr/lib/node_modules/signalk-server/node_modules/Atcanboat/canboatjs/bin
 node ./ikonvert-serial /dev/ttyUSB0 | ./analyzerjs

For example:
(Incorrect data output)
./ikonvert-serial /dev/ttyUSB0 | /usr/lib/node_modules/signalk-server/node_modules/Atcanboat/canboatjs/bin/analyzerjs | grep Depth

INFO 2022-10-15T09:24:52.258Z [ikonvert-serial] Connected to SELF_TEST_PASSED
{"prio":3,"pgn":128267,"dst":255,"src":2,"timestamp":"2022-10-15T09:24:53.362Z","input":["2022-10-15T09:24:53.362Z,3,128267,2,255,12,14,5f,76,d3,5d,34,d3,40,bc,d3,41,45"],"fields":{"SID":20,"Depth":15741394.87,"Offset":-11.468,"Range":640},"description":"Water Depth"}

However in testing if I use 'ikonvert-serial **-x** /dev/ttyUSB0 | analyzer' from the cli the output data shows correctly  ....

(Correct data output)
./ikonvert-serial -x /dev/ttyUSB0 | /usr/lib/node_modules/signalk-server/node_modules/Atcanboat/canboatjs/bin/analyzerjs | grep Depth

INFO 2022-10-15T09:25:18.061Z [ikonvert-serial] Connected to SELF_TEST_PASSED
{"prio":3,"pgn":128267,"dst":255,"src":2,"timestamp":"2022-10-15T09:25:19.138Z","input":["2022-10-15T09:25:19.138Z,3,128267,2,255,8,ff,93,01,00,00,c8,00,ff"],"fields":{"Depth":4.03,"Offset":0.2},"description":"Water Depth"}

I'm thinking it has something to do with the serial output encoding (Hex vs Base64) from iKonvert but am unsure about how to best deal with correctly configuring the SignalK connection.

Hoping someone may have advise to getting the data into SignalK correctly.

Many Thanks
Steve

bikerflyer, 2022-12-20:
So after a bit of digging into the node modules I have managed to fix the issue....well someone with more knowledge might have a proper fix , I've managed a cludge that seems to work for me and appears now to report the correct data into SignalK.

 Appears to me like the PGN binary data from my serial connection to Ikonvert is as suspected hex encoded not base64. Odd because all the documentation says it should be base64. Anyways this code change worked for me. Maybe helpful for others.

From the file /usr/lib/node_modules/signalk-server/node_modules/@canboat/canboatjs/lib/stringMsg.js

I simply replaced 'base64' for 'hex' within the iKonvert section and bingo now works as expected.. 

I'm not sure if this is happening with just my iKonvert or all of them. Seems strange not more people are reporting this problem.? Over to you guys to hopefully make more sense of this.

// iKonvert
// !PDGY,126992,3,2,255,0.563,d2009e45b3b8821d
exports.isPDGY = startsWith('!PDGY,')
exports.parsePDGY = (input) => {
  const parts = input.split(',')
  if ( parts.length === 7 ) {
    const [ prefix, pgn, prio, src, dst, timer, data ] = parts
    return buildMsg(
      buildCanId(prio, pgn, dst, src), 'PDGY', Buffer.from(data, 'hex'),
      { timer: Number(timer), prefix, coalesced: true },
    )
  } else if ( parts.length === 4 ) {
    const [ prefix, pgn, dst, data ] = parts
    return buildMsg(
      buildCanId(0, pgn, dst, 0), 'PDGY', Buffer.from(data, 'hex'),
      { coalesced: true }
    )
  } else {
    return buildErr('iKonvert', 'Invalid parts.', input)
  }
}
exports.encodePDGY = ({ prefix = '!PDGY', pgn, data, dst = 255}) => (
  [ prefix, pgn, dst, data.toString('hex')].join(',')
)

sbender9, 2022-12-20:
This should be in the canboatjs repository. 

I think there is a setting for the  iKonvert to use hex instead of base64. It defaults to base64, not sure how yours got into hex mode.

keesverruijt, 2022-12-20:
I suggest sending the ikonvert a factory reset message, see https://github.com/digitalyacht/iKonvert/blob/master/iKonvert_Gateway_Developers_Guide_V1-04.pdf

bikerflyer, 2022-12-20:
Thanks guys.  The Ikonvert is a new one recently shipped. I'll reset and advise.

tkurki, 2023-01-14:
I believe this was covered elsewhere (the encoding was wrong on that particular device) and the issue can be closed.

closed 2023-01-14 by keesverruijt