canboat/canboat #365: Search engine optimisation, bookmarkability/addressability of the documentation

tkurki, 2023-01-14:
While https://canboat.github.io/canboat/canboat.xml works pretty nicely for humans I don't think the content ends up in search engines because of the xml + xsl rendering method.

I think the ability to bookmark / share a url that takes you to a specific PGN would also be useful. Not sure if that can be achieved with xml + xsl, as even if we add anchors to the pgns they are not present initially when the browser loads the page.

tkurki, 2023-01-14:
There are plenty of different kits for generating documentation as static html, just pick one that is stable and won't be gone in 12 months. GH Actions can render the static html on master commits and version releases.

keesverruijt, 2023-01-14:
It is trivial to generate a HTML version with xslt.

keesverruijt, 2023-02-12:
I just did some google searches, and you are right, no links to canboat.xml seem to come up.

closed 2023-02-12 by keesverruijt