canboat/canboat #372: `analyzer`: add support for Actisense W2K format

keesverruijt, 2023-02-09:
See discussion #371 

@aldas has obtained the following info from Actisense:

	// Ahhmmss.ddd <SS><DD><P> <PPPPP> b0b1b2b3b4b5b6b7.....bn<CR><LF>
	// A = message is N2K or J1939 message
	// 173321.107 - time 17:33:21.107
	// <SS> - source address
	// <DD> - destination address
	// <P> - priority
	// <PPPPP> - PGN number
	// b0b1b2b3b4b5b6b7.....bn - data payload in hex. NB: ISO TP payload could be up to 1786 bytes
	//
	// Example: `A173321.107 23FF7 1F513 012F3070002F30709F\n`

aldas, 2023-02-09:
some remarks. Actisense calls this format "N2K ASCII". This one of the possible outputs of https://actisense.com/products/w2k-1-nmea-2000-wifi-gateway/  This is actually pretty cool device (run on ESP32).

`hhmmss.ddd` they say that `.ddd` is optional. I have not seen yet line where it missing.
Also there is not that "Fields may be added after the data to expand the standard. Decoders should only parse the data up to the <CR> or the last whitespace character (last non-hex character)."

closed 2023-02-10 by keesverruijt

vidax, 2023-02-10:
Great! Thank you very much.

commit 3c3a2ce1 (2023-02-10): Fix #372: Add support for Actisense N2K Ascii protocol