canboat/canboat #389: Latest pgns.xml missing 'Type' for some fields

sbender9, 2023-03-24:
Sorry, I tried to figure out why, but no luck...

DEBUG 2023-03-24T21:10:04.713Z [analyzer-explain] PGN 129033 len=8
      <Length>8</Length>
      <Fields>
        <Field>
          <Order>1</Order>
          <Id>date</Id>
          <Name>Date</Name>
          <BitLength>16</BitLength>
          <BitOffset>0</BitOffset>
          <BitStart>0</BitStart>
          <Units>d</Units>
          <Resolution>1</Resolution>
          <Signed>false</Signed>
          <RangeMin>0</RangeMin>
          <RangeMax>65533</RangeMax>
        </Field>
        <Field>
          <Order>2</Order>
          <Id>time</Id>
          <Name>Time</Name>
          <Description>Seconds since midnight</Description>
          <BitLength>32</BitLength>
          <BitOffset>16</BitOffset>
          <BitStart>0</BitStart>
          <Units>s</Units>
          <Resolution>0.0001</Resolution>
          <Signed>false</Signed>
          <RangeMin>0</RangeMin>
          <RangeMax>86402</RangeMax>
        </Field>

keesverruijt, 2023-03-24:
Let’s add a test so this can’t happen again.

sbender9, 2023-03-24:
Also "Discrete Status 1" missing type for pgn 127489

sbender9, 2023-03-24:
type missing for "Parameter value" , pgn 126208

PR #392 (2023-03-30): fix: missing field types in v1 pgns.xml
Fixes #389