canboat/canboat #399: Support for 'standard' frame length CAN messages

grazzam, 2023-04-18:
I have a REC ABMS that outputs a standard CAN frame (11-bit identifier) instead of the "extended" CAN frame (29-bit total identifiers) required by NMEA 2000. Is it possible to parse these sentences?

[image]

Here is an output of "candump can0 | candump2analyzer | analyzer" with these shorter frames:

[image]

keesverruijt, 2023-04-18:
Analyzing random CAN messages is not exactly within the purview of `canboat` which only aim is to analyse NMEA2000. 

Feel free to fork and improve, but I don't see myself spending much time on this.

I will create a bug that `candump2analyzer` should filter these messages out.

keesverruijt, 2023-04-18:
See issue #400.

grazzam, 2023-04-18:
Understood, thanks for taking a look.