canboat/canboat #402: Connect to Garmin marine equipment

abhayastudios, 2023-04-29:
Hi all,

I apologize in advance as there is a fairly high chance that my question is not related to this project. In that case I would greatly appreciate if anyone could point me to a relevant community. I'm hoping though that what I am trying to achieve is possible using this project (preferably without any additional hardware like the NMEA to SignalK converters that are mentioned in this project's wiki).

I have a friend with a boat equipped with Garmin navigation hardware including a blackbox with an AIS module, a chart plotter and some addition displays and sensors including wind speed/direction sensors like the [Garmin GNX Wired Sail Pack](https://www.garmin.com/en-US/p/516433/pn/010-01248-70).

My goal is to connect to this onboard equipment in some way (e.g. Wifi, Bluetooth) and extract the wind direction and speed somehow from a laptop/mobile phone.

There is a Wifi network on board that Garmin creates (I guess the blackbox does that) and when I connect my laptop to that Wifi network, I see that several services are published through mDNS:

```
$ dns-sd -B _services._dns-sd._udp
Browsing for _services._dns-sd._udp
17:29:17.011  ...STARTING...
Timestamp     A/R    Flags  if Domain               Service Type         Instance Name
[...]
17:29:17.218  Add        3   6 .                    _tcp.local.          _garmin-ota-updt
17:29:17.218  Add        3   6 .                    _tcp.local.          _garmin-mrn-stng
17:29:17.218  Add        3   6 .                    _udp.local.          _garmin-ds-sink
17:29:17.218  Add        3   6 .                    _tcp.local.          _garmin-swifi
17:29:17.218  Add        3   6 .                    _tcp.local.          _garmin-mrn-card
17:29:17.218  Add        3   6 .                    _tcp.local.          _garmin-gvs
17:29:17.218  Add        3   6 .                    _tcp.local.          _garmin-helm
17:29:17.218  Add        3   6 .                    _tcp.local.          _garmin-vssl-inv
17:29:17.218  Add        3   6 .                    _tcp.local.          _garmin-bl-id
17:29:17.218  Add        3   6 .                    _tcp.local.          _garmin-sn-sink
17:29:17.218  Add        3   6 .                    _tcp.local.          _garmin-updt-tst
17:29:17.218  Add        3   6 .                    _tcp.local.          _garmin-wx-rx
17:29:17.218  Add        3   6 .                    _tcp.local.          _garmin-bcm
```

I've looked but I cant find any documentation on these services and what they provide. I understand that in the past Garmin published a Marine SDK but it seems that it is no longer available. I've reached out to Garmin support but they were not able help me either. Can anyone point me in the direction of what service/port I should connect to and which protocol top use? Does Garmin expose SignalK in some way through any of these published services?

Alternatively, I see that the Garmin equipment also supports bluetooth connections. Does anyone know if it is possible to obtain that data through bluetooth? If so, I think I would prefer that even over Wifi.

If anyone has some documentation available or even tips that could help me in any way that would be really great!

Thanks!

acolomb, 2023-04-30:
IIRC, the chart plotter opens a Wi-Fi, where you can connect to it either through a web page or a smartphone app. It's been quite some time since I've used it. But there were at least some navigation features exposed for usage via that Wi-Fi, so in principle the information should be there. If it serves a web page, you could extract the request for individual data items from the running code and use that to replicate in your laptop. Might have been only a screen sharing feature though, and they probably added much more stuff in recent years.

I don't think they are involved with SignalK, though you might easily build something involving that code with just a cheap CANbus (which NMEA2000 is based on) to USB converter, wired to your laptop.

abhayastudios, 2023-04-30:
@acolomb Thanks André for your response! Indeed it seems that using a CANbus converter would allow this, but I was hoping to avoid having to use one (so that what I will create can be taken to other Garmin systems as well).

Indeed I was able to connect to the Garmin Wifi system and connect to some of the ports published. For example, on the Garmin network I discovered a service of type `_garmin-helm` that was published on one of the devices on port `51200`. I could establish a TCP connection, but it does not seem to be a HTTP(S) service. Since their Marine SDK no longer seems to be available I was hoping someone could point me to some documentation or other resources on which protocol that is and how to extract sensor data from it.

Cheers

nohal, 2023-04-30:
I had a very brief look when a boat with a Garmin plotter passed through  here. None of the exposed ports seems to be providing anything directly useful, so the only option probably will be to do the good old reengineering of the traffic between the app and the plotter. Honestly not sure it is worth any effort when you can connect to the n2k bus...

keesverruijt, 2023-04-30:
Have you tried a simple ‘nc host port ‘ ? Some protocols start sending automatically and are not http compatible.

keesverruijt, 2023-04-30:
> O
> 
If Pavel says no data is immediately available I strongly suggest you believe him. 

How about one of the wireless N2K adapters for a quick and painless connection if you don’t want to tinker with hardware much?

nohal, 2023-04-30:
> Have you tried a simple ‘nc host port ‘ ? Some protocols start sending automatically and are not http compatible.

At the time I tried to connect to all the TCP endpoints `nmap` found on the plotter and none of them provided anything useful. But I really invested pretty much no time to it and then lended them a NMEA2000 capable Miniplex I have at home...

abhayastudios, 2023-05-02:
Thanks all for sharing your thoughts.

closed 2023-05-02 by abhayastudios