canboat/canboat #403: Is Puerket decoding right ?

ieb, 2023-05-07:
https://github.com/canboat/canboat/blob/357f0c9a161719272853dedcdf635386abcd8560/analyzer/fieldtype.h#L836

Take Puerket 1.25 (typical lead acid)

As per https://canboat.github.io/canboat/canboat.html
Range is 1 - 1.506
1 bit = 0.002
Offset = 500

1.25/0.002 = 625
625-500 = 125 = 0x7D

analyse produces 

2023-05-07-09:12:54.231 6  26 255 127513 Battery Configuration Status:  Instance = 1; Battery Type = AGM; Supports Equalization = No; Reserved = 00; Nominal Voltage = 12V; Chemistry = Pb (Lead); Capacity = 0 Ah; Temperature Coefficient = 5 %; Peukert Exponent = 0.250; Charge Efficiency Factor = 95 %

I dont think its a signed int vs unsigned int thing.

1.45/0.002 = 725
725-500 = 225 = 0xE1

2023-05-07-09:17:57.662 6  26 255 127513 Battery Configuration Status:  Instance = 1; Battery Type = AGM; Supports Equalization = No; Reserved = 00; Nominal Voltage = 12V; Chemistry = Pb (Lead); Capacity = 0 Ah; Temperature Coefficient = 5 %; Peukert Exponent = 0.450; Charge Efficiency Factor = 95 %

What's the right way to encode Puerket ? Or is this a bug in the decoding ?
I tried to find out where it might be in decoding, but could not see, the offset seems to be mentioned, and I assume it is applied, but perhaps not ?

0xE1 = 225
225*0.002 = 0.45 which matches the output seen.

There must be an offset because Puerket only goes from 1 - 1.506, and 1/0.002 = 500 which is > uint8_t ?

keesverruijt, 2023-06-10:
Hi, I haven't analysed this fully, but I have changed your 'Puekret'  to 'Peutert' (the poor fellow deserves his name written right, at least I think?) (See https://en.wikipedia.org/wiki/Peukert%27s_law )

keesverruijt, 2023-06-10:
Yup, it was a mistake in the decoder. Fix on master and in upcoming 4.13 release.

commit e6154aeb (2023-06-10): Fix #403: Peukert exponent incorrect in PGN 127513.

Note that this was a generic bug in unsigned fields with an offset-K notation.
However, the Peuker exponent was the only such field so it was not seen before.

Silly as I use this PGN in the tests, and I should have seen that the value was incorrect.
closed 2023-06-10 by keesverruijt