canboat/canboat #405: Json Fast packet syntax documentation

fulup-bzh, 2023-05-15:
I'm in the process of implementing a Rust version of canboat. While most information seems obvious. Before losing time in debugging, I would be please to get some precision.

Using "PGN":129285 as sample because it has a candump log
-  "MinLength":10 => 10*8= 80bits 
   * order:1 bitoffset:0
   * order:7 bit offset: 69, len:5 => 71bits
   * order:8 "BitLengthVariable":true, len within the string
   * order:9 "BitLength":8 (no bit start) does it start after string from order-8 ?

Thank you

keesverruijt, 2023-05-15:
I've been thinking about a Rust implementation as well. I assume that your version will be fully open source?

All fields are always immediately next to each other. Any "spacing" bits are explicit in the definition. For instance, field 7 in 129285 is such a "spacing" field, in this case to get the next field on a byte boundary.

So: Yes, field 9 starts after field 8. And field 10 after that. Etc. Note that 129285 is one of a few where there are repeating fields. There are a few ways this is done, this is a "simple" case with a repeating set that repeats _n_ times where _n_ is defined by another field, in this case field 2.

The full explanation can be found on https://canboat.github.io/canboat/canboat.html#pgn-129285 where you will find an English HTML version of `canboat.xml` (translated by `canboat.xsl`)

keesverruijt, 2023-05-15:
BTW I don't think you need the BitStart and BitOffset fields really, in a full blown parser. My C parser doesn't use this, I think.

fulup-bzh, 2023-05-15:
It will be opensource. I already push the initial DBC version on https://github.com/redpesk-labs/canbus-rs last week I added J1939 and I wait to have an initial working version of N2K before pushing it. One difference with canboat is that we use embedded devices with can devices and use socket-can to read can packet. 
Thank you for your explanation, I will start from this and will keep you aware of  my progress.

fulup-bzh, 2023-05-16:
My initial version of fastpacket for Linux sockcan is avaliable from https://github.com/redpesk-labs/canbus-rs/blob/master/examples/basic/src/read-j1939.rs

closed 2023-05-16 by fulup-bzh