canboat/canboat #406: Json PGNs schema

fulup-bzh, 2023-05-16:
I just push an initial Rust json parser for Canboat.json PGNs spec. My testing code is available at https://github.com/redpesk-labs/canbus-rs/blob/master/n2kparser/src/json-parser.rs Now that I known how to parse the spec, I need to understand it.

Here after few field that I do not understand. If documentation exist please let me known. Otherwise some explanation would avoid me to reverse engineering C code. 
```
- "Type":"Single"        ?? is this only used for enum ??
- "BitOffset":136        ??  "PGN":61184 we have "BitLength":2,"BitOffset":11,"BitStart":3. How to process this
- "Complete": false    ?? should we ignore the full message or we may nevertheless decode a part of it ??
- "Resolution":1e-07 ??  this can be any float value, what are we suppose to do with it ?
- RepeatingField        ?? "PGN":126208 RepeatingFieldSet1Size":2,"RepeatingFieldSet1StartField":6,"RepeatingFieldSet1CountField":5
- Type AsciiText what the difference with AsciiString ?
```
note: some type enum are very long as "ISO Transport Protocol, Data Transfer" something shorter as 'ISO-Transport' would be simpler to process. Until Json is generated automatically the risk of error is small, but if someone edit spec by hand, then such long label are very risky.

I

fulup-bzh, 2023-05-16:
Hoop yet an other field that I do not understand
```
PGN":60416
- "Match":16, ?? what is this ??
```

aldas, 2023-05-16:
@fulup-bzh canboat.xsd has description for fields. For example `match`

https://github.com/canboat/canboat/blob/357f0c9a161719272853dedcdf635386abcd8560/docs/canboat.xsd#L367-L375

sbender9, 2023-05-16:
FYI, you may find the test data useful here https://github.com/canboat/canboatjs/tree/canboat-v4-compat/test/pgns

aldas, 2023-05-16:
I doubt you can get away from reading CanBoat as there are small nuances hidden everywhere.  I also recommend checking https://github.com/ttlappalainen/NMEA2000 parser implementation - reading both repos code and comments helped me a lot.  And you most likely need to decide which Canboat concepts you implement and which not - for example if intend to use library mostly for storing decoded/parsed packets to database you can get away with numeric values for lookups.  Also PGN 126208 has different variants + field type `VARIABLE` which itself is quite a thing.

Maybe checking my own take of [decoder](https://github.com/aldas/go-nmea-client/blob/de2f4ab88488ee621ccdfac0835290083e24eac0/canboat/decoder.go#L76) helps you as it has little bit different "flow" but produces "same" result.

fulup-bzh, 2023-05-17:
Thank you for all those explanations. My initial effort is to generate the RUST encoder/decoder corresponding to N2K as I do for DBC. I have no doubt that I will have to deep dive into some existing code, nevertheless the mode good doc you find, the more time you save. I was not aware of ttlappalainen project but I will check it in deep.

keesverruijt, 2023-05-17:
Again I suggest you take a good look at the HTML link I referred to earlier, it is generated 100% from canboat.xml and canboat.xsl and contains 100% of all knowledge needed to generate a parser. 

The xml and json are fully equivalent. They contain some extra fields desired by smaller parsers.

the PGN descriptions and thus JSON names may be long and winded, but we cannot change these as downstream parsers may depend on them. This is historical ballast that we cannot avoid. I learned this the hard way when I started “optimising” names.

fulup-bzh, 2023-05-17:
I already look at https://canboat.github.io/canboat/canboat.html it is very useful not not friendly for automatic parsing. For names, I understand that it is too late for change, until they are generated automatically it should not be an issue.

keesverruijt, 2023-06-09:
> I already look at https://canboat.github.io/canboat/canboat.html it is very useful not not friendly for automatic parsing. 

Yes, the HTML is for humans and the XML / JSON is for code generators. But as I have stated already: `canboat.html` is the outcome of applying `canboat.xsl` to `canboat.xml`, and thus if you need to know how a particular field works, read the HTML file. If you understand the HTML but not the XML, look at the XSL to see how the HTML is generated.

keesverruijt, 2023-06-09:
> Hoop yet an other field that I do not understand
> 
> ```
> PGN":60416
> - "Match":16, ?? what is this ??
> ```

This version of the PGN only applies when that field contains the value 16. Many PGN numbers have multiple explanations, that depend on the value of one or more of the fields. For instance, all company specific PGNs only apply to a particular company by means of the first few fields.