canboat/canboat #423: Garmin GMI20 DIsplay: Backlight Level, Day Night Mode and Colours

TwoCanPlugIn, 2023-09-20:
FYI.

PGN 126720

Change Backlight Level and Day/Night Mode

byte[0] = 0xE5; 
byte[1] = 0x98;
byte[2] = 0xDE;
byte[3] = 0x05;
byte[4] = 0x05;
byte[5] = 0x05;
byte[6] = 0x00;
byte[7] = 0x00;
byte[8] = mode; // 0-Daylight Mode, 1-Night Mode
byte[9] = 0x00;
byte[10] = level; // in 5% increments

Change Colours
byte[0] = 0xE5;
byte[1] = 0x98;
byte[2] = 0xDE;
byte[3] = 0x05;
byte[4] = 0x05;
byte[5] = 0x05;
byte[6] = 0x00;
byte[7] = 0x00;
byte[8] = 0x0D; // Colour Mode command
byte[9] = 0x00; 
byte[10] = mode; // 0-Day Full Colour, 1, Day High Contrast, 2-Night Full Colour, 3- Night Red/Black, 4-Night Green/Black

closed 2023-12-11 by keesverruijt

stevehayles, 2025-04-28:
Is this expected to work with devices other than the GMI20 displays ?

keesverruijt, 2025-04-28:
Probably all Garmin.

stevehayles, 2025-04-29:
> Probably all Garmin.

Thanks for this,  I had hoped so but just tried on a set of GNX120 displays and it doesn't appear to work.  Will try and dig deeper...

TwoCanPlugIn, 2025-05-02:
It would be great if can capture a log when you are changing brightness levels, day/night colour mode and if the display is standalone or part of a group.

stevehayles, 2025-05-02:
I will try and dig deeper @TwoCanPlugIn and grab a log of data when the brightness is being changed from one of the displays.  The GNX displays are part of a group of 6 displays set to 'Network Sharing' in the display menu.

commit afc47428 (2023-12-11): Fix #423: Generated files