canboat/canboat #425: add pgn 65293 definition

cedschmidt-lab, 2023-10-13:
Hi,
I want to add a pgn 65293 new definition to canboat/JS but don't really know how to proceed. It contains load cell values.

An Actisense-serialjs output is:
2023-10-11T18:40:13.553Z,2,65293,85,255,8,81,9a,0d,ff,4c,06,00,00
2023-10-11T18:40:13.755Z,2,65293,85,255,8,81,9a,0d,ff,3f,06,00,00
2023-10-11T18:40:13.953Z,2,65293,85,255,8,81,9a,0d,ff,91,06,00,00
2023-10-11T18:40:14.156Z,2,65293,85,255,8,81,9a,0d,ff,5c,06,00,00
2023-10-11T18:40:14.356Z,2,65293,85,255,8,81,9a,0d,ff,7d,06,00,00
2023-10-11T18:40:14.557Z,2,65293,85,255,8,81,9a,0d,ff,5e,06,00,00
2023-10-11T18:40:14.756Z,2,65293,85,255,8,81,9a,0d,ff,96,06,00,00
2023-10-11T18:40:14.962Z,2,65293,85,255,8,81,9a,0d,ff,3d,06,00,00

I found the following structure by reverse engineering:
3rd byte in the 64 bit data field is an instance ID.  The numerical value is the last 4 bytes blocks of data as UintLE. 

Can someone help me?

keesverruijt, 2023-10-15:
`canboat-js` depends on `canboat`, so if you want to develop your own workflow it's easier to use `canboat` (which works in Signal K just as well).

See the commits in branch `add-pgn-65293-definition` on how to do this. The last commit contains manual updates to `CHANGELOG.md`, `common/version.h` and the output changed after running `make generated`. As a downstream developer we expect you to only add your changes to the _unreleased_ section in `CHANGELOG.md` and possibly the output of `make generated`.

@sbender9 do you want to add anything on the 2nd part of the process (updating `canboat-js`?)

closed 2023-10-15 by keesverruijt

sbender9, 2023-10-15:
Currently updating canboatjs pgn definitions is a manual process. So reach out to me or open an issue in canboatjs. I would like to automate this and will be looking at that in the off season. 

Today I released new versions of the pgn definitions and a new version of canboatjs, this change is included.