canboat/canboat #428: PGN 127509 Issues

patg, 2023-10-31:
Mastervolt Equipment (Combimaster, Masterbus-Bridge, all on newest firmware) seem to send PGN 127509 as 'Fast' packets and not single (which is strange, since the data payload would easily fit in a single packet)

But indeed the NMEA Technical Corrigendum TC# 2000 201401032  (link to document seems no longer
to be correct) seems to confirm this:

Single Frame: No
Destination: Global
Priority Default: 6
Query Support: required
Default Update Rate:   1500 milliseconds
Command Support: Optional
Frequency: .7 cycles per second
ACK Rqmnts: None

If I change it to 'fast', the packet is decoded correctly.

diff --git a/analyzer/pgn.h b/analyzer/pgn.h
index c01de50..7077c5a 100644
--- a/analyzer/pgn.h
+++ b/analyzer/pgn.h
@@ -3254,7 +3254,7 @@ Pgn pgnList[] = {
     {"Inverter Status",
      127509,
      PACKET_COMPLETE,
-     PACKET_SINGLE,
+     PACKET_FAST,
      {INSTANCE_FIELD,
       UINT8_FIELD("AC Instance"),
       UINT8_FIELD("DC Instance"),

keesverruijt, 2023-10-31:
On re-reading the technical note I think I agree with you, and would have changed this if I had some more proof. Can you upload a bit of your raw(ish) data that shows this? So far I only have "single line logs" which don't show whether a PGN is received in Fast or Single mode as the existing log files show the data after it is converted to single-line-per-PGN format.

patg, 2023-10-31:
can0  1DF2150B   [8]  A0 08 00 01 00 C4 AC 3A
  can0  1DF2150B   [8]  A1 00 00 FF FF FF FF FF
  can0  1DF2150B   [8]  C0 08 00 01 00 C4 AC 3A
  can0  1DF2150B   [8]  C1 00 00 FF FF FF FF FF
  can0  1DF2150B   [8]  E0 08 00 01 00 C4 AC 3A
  can0  1DF2150B   [8]  E1 00 00 FF FF FF FF FF
  can0  1DF2150B   [8]  00 08 00 01 00 C4 AC 3A
  can0  1DF2150B   [8]  01 00 00 FF FF FF FF FF

This is sent by a Mastervolt Bridge (BEP Marine MBI) with the newest CZONE-Release (Rel 21)
Don't have any other devices sending this PGN, would be interesting to see what Victron sends.

keesverruijt, 2023-10-31:
Interestingly this had 8 actual data bytes, not four. Are you able to derive what the last four bytes mean? The technical note has some info, can be found via archive.org. This says it’s not about the inverter. It could be battery related?

patg, 2023-11-02:
Yes, BEP Marine is still using the original coding of the 127509 PGN with field 7 and 8, but these fields are always zero as far as I can see. Unfortunately I couldn't find any documentation on the encoding of these fields. BEP just lists the current 6 fields (found the user manual, but not on the mastervolt page)

https://www.tjustel.se/wp-content/uploads/2018/08/10000012741_01MasterBus-NMEA2000_InterfaceEN.pdf

So it looks like the NMEA technical note just removed field 7 and 8, but left the packet mode as 'fast' to keep compatiblity with devices sending the message in original form.

I was just wondering if there is maybe equipment around sending it (wrongly) as single frame since this was unnoticed for a long time.

keesverruijt, 2023-11-02:
Side note here, I never could get my Mastervolt NMEA bridge to be reliable; it needed a power cycle every month or so.

patg, 2023-11-03:
Did a quick check on another project, https://github.com/ttlappalainen/NMEA2000, arduino NMEA library.

https://github.com/ttlappalainen/NMEA2000/blob/master/src/NMEA2000.cpp
127509 is listed as 'default fast packet' message there, which means it is treated as fast packet message
by default,but somtimes also as single message, but haven't yet analysed the fuzzy logic used there to detect if
it is a single frame.

commit f61941dc (2023-12-11): Fix #428
closed 2023-12-11 by keesverruijt