canboat/canboat #431: Add Support for Yacht Device N2K Format.

cas206, 2023-12-19:
According to the YDNU-02 manual, the N2K format "is based on Data Link Escape encoding partially compatible with ActiSense NGT format".  The actisense-serial utility partially works to format a N2K dump file for use by analyze.

Issue 1:  A 0x67 character in the middle of a message ("state" flag set to MSG_MESSAGE) is identified as ESC which causes the "state" flag to be erroneously set to MSG_ESCAPE.  That valid message is lost, but the program recovers and continues.

cas206, 2023-12-19:
Issue 2.  When a message terminates at the end of a 500 byte read block (0x10 0x03),  the start of the next 500 byte block is somehow mangled.  The first 10 bytes of the new 500 byte block appears to be overwritten by "10 02 a1 03 11 02 00 49 10 03" which breaks the original message that was there.  Still debugging what the cause it.

cas206, 2023-12-21:
DEBUG 2023-12-21T19:21:53.881Z [actisense-serial] NGT data: 0a 00 08 fc f8 2a 0a ff 7f ff ff b9 10 03 10
02 93 13 07 63 ff 00 ff cc b9 66 0a 00 08 3b 9f 40 00 00 00 02 ff da 10 03 10 02 93 13 07 4f ff 00 ff cc
ba 66 0a 00 08 3b 9f ff ff ff 34 ed ff 11 10 03 10 02 93 13 07 50 ff 00 ff cc ba 66 0a 00 08 3b 9f ff ff
ff 30 ee ff 13 10 03 10 02 93 13 02 0d f1 01 ff ac 02 67 0a 00 08 fc f8 2a 0a 01 0a ff ff 02 10 03 10 02
93 13 02 12 f1 01 ff cc 19 67 0a 00 08 ff 34 ed ff 7f ff 7f fd de 10 03 10 02 93 14 07 00 ef 01 ff cc 1b
67 0a 00 09 3b 9f f0 81 ae 02 01 0c a5 55 10 03 10 02 93 13 02 13 f1 01 ff cc 1b 67 0a 00 08 ff 1b 3e 00
00 ff ff ff 9f 10 03 10 02 93 13 02 19 f1 01 ff cc 1c 67 0a 00 08 ff c4 f7 d2 00 e0 ff ff 83 10 03 10 02
93 13 02 0d f1 01 ff cc 1c 67 0a 00 08 00 ff 2a 0a 01 0a ff ff bd 10 03 10 02 93 13 02 0d f1 01 ff cc 1d
67 0a 00 08 fc f8 2a 0a ff 7f ff ff 54 10 03 10 02 93 13 07 4f ff 00 ff cc 1d 67 0a 00 08 3b 9f ff ff ff
34 ed ff ad 10 03 10 02 93 13 07 64 ff 01 ff cc 1e 67 0a 00 08 a0 07 3b 9f ff ff ff 7f 90 10 03 10 02 93
13 07 64 ff 01 ff cc 1f 67 0a 00 08 a1 ff ff ff ff ff ff ff f2 10 03 10 02 93 13 07 50 ff 00 ff cc 1f 67
0a 00 08 3b 9f ff ff ff 30 ee ff ad 10 03 10 02 93 20 02 05 f1 01 ff ac 68 67 0a 00 15 3c c2 1f 2a 0a ff
ff ff ff ff ff ff ff ff 7f ff 7f ff 7f ff ff fa 10 03 10 02 93 13 02 0d f1 01 ff ac 6f 67 0a 00 08 fc f8
2a 0a f4 09 ff ff a3 10 03 10 02 93 13 02 12 f1 01 ff cc 7e 67 0a 00 08 ff 33 ed ff 7f ff 7f fd 7a 10 03
10 02 93 13 02 13 f1 01 ff cc 7f 67 0a 00 08 ff a2 28 00 00 ff ff ff ca 10 03 10 02 93 13
DEBUG 2023-12-21T19:21:53.881Z [actisense-serial] message command = 93 len = 19
2023-12-21T19:21:53.881Z,2,127245,204,255,8,fc,f8,2a,0a,ff,7f,ff,ff
DEBUG 2023-12-21T19:21:53.881Z [actisense-serial] message command = 93 len = 19
2023-12-21T19:21:53.881Z,7,65379,204,255,8,3b,9f,40,00,00,00,02,ff
DEBUG 2023-12-21T19:21:53.881Z [actisense-serial] message command = 93 len = 19
2023-12-21T19:21:53.881Z,7,65359,204,255,8,3b,9f,ff,ff,ff,34,ed,ff
DEBUG 2023-12-21T19:21:53.881Z [actisense-serial] message command = 93 len = 19
2023-12-21T19:21:53.881Z,7,65360,204,255,8,3b,9f,ff,ff,ff,30,ee,ff
DEBUG 2023-12-21T19:21:53.881Z [actisense-serial] message command = 93 len = 19
2023-12-21T19:21:53.881Z,2,127245,172,255,8,fc,f8,2a,0a,01,0a,ff,ff
DEBUG 2023-12-21T19:21:53.881Z [actisense-serial] message command = 93 len = 19
2023-12-21T19:21:53.881Z,2,127250,204,255,8,ff,34,ed,ff,7f,ff,7f,fd
ERROR 2023-12-21T19:21:53.881Z [actisense-serial] DLE followed by unexpected char 67, ignore message
ERROR 2023-12-21T19:21:53.881Z [actisense-serial] Ignoring message with invalid checksum
ERROR 2023-12-21T19:21:53.881Z [actisense-serial] DLE followed by unexpected char 67, ignore message
ERROR 2023-12-21T19:21:53.881Z [actisense-serial] Ignoring message with invalid checksum
DEBUG 2023-12-21T19:21:53.881Z [actisense-serial] message command = 93 len = 19
2023-12-21T19:21:53.881Z,2,127257,204,255,8,ff,c4,f7,d2,00,e0,ff,ff
DEBUG 2023-12-21T19:21:53.881Z [actisense-serial] message command = 93 len = 19
2023-12-21T19:21:53.881Z,2,127245,204,255,8,00,ff,2a,0a,01,0a,ff,ff

actisense-serial -r test.bin mostly works except it gets tripped up by a 0x1b character in the message.

10 02 
&nbsp;&nbsp;&nbsp;&nbsp;93 13 02 13 f1 01 ff cc 1b 67 0a 00 // The ESC character shows up here in the time field causing an error
&nbsp;&nbsp;&nbsp;&nbsp;08 ff 1b 3e 00 00 ff ff ff 
&nbsp;&nbsp;&nbsp;&nbsp;9f 
10 03

cas206, 2023-12-22:
> ~Issue 2. When a message terminates at the end of a 500 byte read block (0x10 0x03), the start of the next 500 byte block is somehow mangled. The first 10 bytes of the new 500 byte block appears to [...]
Issue 2.
Without the -r option of "analyzer", parts of the input file gets overwritten with "10 02 a1 03 11 02 00 49 10 03".  This leads to an incomplete message in the input file that triggers an error message in "analyzer".

keesverruijt, 2024-08-02:
Can you upload / send me your test.bin file?

cas206, 2024-08-02:
It's what you renamed as "ydnu-02-raymarine-ev1.dle" in the samples folder.