canboat/canboat #435: analyzer unit conversions?

freol35241, 2024-01-16:
Hi and thanks for all the work put into this set of tools!

I think there are som inconsistencies with regards to unit conversion in the analyzer command-line tool, this was noticed when used to parse PGN 129026 (SOG and COG rapid update).

When running `analyzer --json` (which would implicitly set the `--no-si` flag I believe?) we see that:
* COG gets converted from radians to degrees
* SOG is not converted and remains in m/s (I would have expected this to be converted to knots)

https://github.com/canboat/canboat/blob/master/analyzer/fieldtype.c#L138-L183

keesverruijt, 2024-04-15:
Sorry, this will not be merged. Changing the output in non-SI mode would need a major version change, as this impacts the compatibility of the analyzer program with other projects.

For historical reasons the 'non SI' units need to remain the same to remain compatible with historical downstream packages. The recommended way to use `analyzer` with other downstream packages is to always use the `-si` flag.

closed 2024-04-15 by keesverruijt

freol35241, 2024-04-16:
@keesverruijt fair enough, thanks for getting back on this.

If not already, I would suggest to document the recommended way of using analyzer (i.e. with the `--si` flag) to avoid future confusion. And also perhaps  clarify what one can expect in terms of output when using the `--no-si` flag.