canboat/canboat #438: Hi, also i have same question,

keesverruijt, 2024-02-16:
Hi, also i have same question,
[image]
picture is show PNG 127489 messages. How can i read Engine Load Percent from here ? @keesverruijt

_Originally posted by @snnszgzl in https://github.com/canboat/canboat/issues/176#issuecomment-1947774082_

keesverruijt, 2024-02-16:
Not sure what you want to know.

The image indicates that you are using PCAN-VIEW from Peak Systems with their PEAK-USB CAN adapter, on Microsoft Windows. Nothing of that is immediately compatible with CANboat.

The canboat tools can read various export formats, you will have to experiment to see of PCAN-VIEW exports to a compatible format. 

Direct hardware access on Microsoft Windows is not something that the usual CANboat user is familiar with. It might work, it might not. I can only cross compile to Windows and hope it works; I do not have Windows systems with hardware access.

snnszgzl, 2024-02-16:
I try to read Engine Percent Load from Volvo Main Engine. Protocol is
nmea2000.
so PGN number is 127489. I filtered only this message from PCAN-VIEW for
decode the message but i couldnt. I just want to read Percent Engine Load
data. How can i decode the message and how can i read that ?

Kees Verruijt ***@***.***>, 16 Şub 2024 Cum, 11:28 tarihinde
şunu yazdı:

keesverruijt, 2024-02-16:
Do you understand what I wrote in #176 ? (I just updated it to be a little more explanatory, expanding on how to detect the first packet and to see what packets form a complete message.)

snnszgzl, 2024-02-16:
of course i ınderstood. I just ask different questions :)

Kees Verruijt ***@***.***>, 16 Şub 2024 Cum, 11:45 tarihinde
şunu yazdı:

TwoCanPlugIn, 2024-02-18:
The Peak Systems trace formats are well documented [here](https://www.peak-system.com/produktcd/Pdf/English/PEAK_CAN_TRC_File_Format.pdf).

Here is a regular expression I have used to parse one of the versions of their trace files for my own set of NMEA 2000 tools. It wouldn't be rocket science to convert a trace file to, for example, the candump format, that canboat analyzer can then parse and decode.

`std::regex picanMRegex("^\\s{1,5}[0-9]{1,5}\\)\\s+[0-9]+.[0-9]{1}\\s{2}Rx\\s{5}([0-9A-Fa-f]{8})\\s{2}8\\s{2}([0-9A-Fa-f]{2})\\s([0-9A-Fa-f]{2})\\s([0-9A-Fa-f]{2})\\s([0-9A-Fa-f]{2})\\s([0-9A-Fa-f]{2})\\s([0-9A-Fa-f]{2})\\s([0-9A-Fa-f]{2})\\s([0-9A-Fa-f]{2})\\s$");`

cas206, 2024-02-23:
This question should probably be listed under "Discussions" as it's not really an "Issue" with any of the code.  I beg forgiveness to the curator for the following add-on discussion.

What exactly is your desired outcome?  Do you want to use "analyzer" from this project to decode your engine data?  Are you wanting the information to decode the bytes yourself?  Do you just need a "one off" answer using your posted data?  Or something else?  If you are asking for this project to directly support your tool file format, then that would an "issue" with a statement asking for such support.

Here is one way of manually getting a decode of your data which answers one of these questions.  I typed the first four lines of data from your picture which represents a single fast pack message into a file.  This looks like:
00 1A 00 00 0A FF 0D AC
01 8A F0 0A 1F 00 E8 44
02 00 00 7C 01 50 0F 7F
03 00 00 01 00 02 7F FF

I replaced all the spaces with comma's:
00,1A,00,00,0A,FF,0D,AC
01,8A,F0,0A,1F,00,E8,44
02,00,00,7C,01,50,0F,7F
03,00,00,01,00,02,7F,FF

Then I prepended a fake header, "2024-02-17T05:15:49.768Z,7,127489,204,255,8," to get this:
2024-02-17T05:15:49.768Z,7,127489,204,255,8,00,1A,00,00,0A,FF,0D,AC
2024-02-17T05:15:49.768Z,7,127489,204,255,8,01,8A,F0,0A,1F,00,E8,44
2024-02-17T05:15:49.768Z,7,127489,204,255,8,02,00,00,7C,01,50,0F,7F
2024-02-17T05:15:49.768Z,7,127489,204,255,8,03,00,00,01,00,02,7F,FF

Run this through "analyzer" to get: 
2024-02-17T05:15:49.768Z 7 204 255 127489 Engine Parameters, Dynamic:  Instance = Single Engine or Dual Engine Port; Oil pressure = 2.560 bar; Oil temperature = 85.15 C; Temperature = 81.85 C; Alternator Potential = 28.00 V; Fuel Rate = 3.1 L/h; Total Engine hours = 04:54:00; Coolant Pressure = 0.380 bar; Fuel Pressure = 39.200 bar; Reserved = 7F; Discrete Status 1 = None; Discrete Status 2 = Warning Level 1; Engine Load = 2 %; Engine Torque = Unknown

For getting a quick answer, this suffices.  But you will want to automate if you need continuous data using something like TwoCanPlugion proposed.

closed 2024-08-02 by keesverruijt

snnszgzl, 2024-08-02:
Hi Kees Verruijt,
I am so appreciate to answer my e-mail.
Actually i fixed my problem and also i solved all nmea2000 pgn message. I
understood how it is decode :)  I will share with project page. it is
conning display for ship.
[image: COONING.png]

Kees Verruijt ***@***.***>, 2 Ağu 2024 Cum, 11:29 tarihinde
şunu yazdı:

keesverruijt, 2024-08-02:
Hi snnszgzl,

If you reply to a mail from github, do not include the original message or any footer. This ends up in the github issue! I have removed it for you.