canboat/canboat #442: PGN129799 : Channel Number incorrect

BaktusBror, 2024-04-15:
Hi, I have hooked a Lowrance Link9 VHF to SignalK, and found that SignalK only understands the Rx and Tx Frequency fields.
My Actisense NGT-1 hooked up to Actisense Reader makes a correct interpretation.

I enclose a screen dump from Actisense EBL reader showing both the Hex code and the Actisense interpretation - I hope this is sufficient to get the PGN definition updated.
[image]

BaktusBror, 2024-04-15:
Addendum: The first two characters of the Channel Number stays at "90" regardless of what channel I select at the VHF. The last two are the Channel Number, but the two middle ones I have not decoded quite yet. 

I enclose two exports of the same EBL, one "raw" and one "decoded". The .csv files contains a sweep through all the channels (Norway channel list).
[COM5 Actisense NGT_20240414_095350.rx.n2k.csv](https://github.com/canboat/canboat/files/14983725/COM5.Actisense.NGT_20240414_095350.rx.n2k.csv)

[COM5 Actisense NGT_20240414_095350.rx.n2kdecoded.csv](https://github.com/canboat/canboat/files/14983726/COM5.Actisense.NGT_20240414_095350.rx.n2kdecoded.csv)

closed 2024-04-15 by keesverruijt

keesverruijt, 2024-04-15:
Thanks, your data was (just) enough to come up with a better definition.

Note that the Frequency field is indeed not numeric, I think "90" stands for VHF, the following "00"/"10","20" value is for "normal", "low" and "high". The latter have been created as part of the changes to VHF channels required for the introduction of VDES. Not sure about Norway but here in NL all VHFs have to be reprogrammed or replaced to change channel allocations.

BaktusBror, 2024-04-15:
Thanks!
That was quick.

TwoCanPlugIn, 2024-05-25:
I think you will find that the coding of the channel number is defined in Table A1-5 in the ITU standard for Digital Selective-Calling (DSC) ,  ITU Recommendation ITU-R M.493-15(01/2019).

Or if you want something easier to digest, refer to the NMEA 0183 standard for the DSC Sentence

Nothing to do with VDES as it very much predates VDES

Essentially
90 - Means VHF
0 - Means duplex operation
016 Channel 16 (with a leading zero)

HTH.
--

TwoCanPlugIn, 2024-05-26:
Further to above, the VHF channel must be alphanumeric. In the UK there are VHF Channels M1 & M2, and a bit of research on my part discovered (as you no doubt already know) there are channels L1, L2, L3 & F1, F2, F3 that are Leisure & Fishing Channels in Nordic waters.

For VHF radios, transmitter power will be either 1 watt or 25 watts. As you know you for VHF you can select low or high power and that some channels are only low power.

From the DSC document,, for radio channel the six digits are termed HM (Hundreds of thousands), TM (Tens of thousands), M (Thousands), H (Hundreds) T (Tens) and U (Units).

if the HM is 0,1,2 it is a HF radio with the TM, M, H, T & U digits indicating the frequency in multiples of 100Hz. Eg. 130770 indicates 13077 Khz
If the HM is 3, the other digits indicate a HF channel. Eg. 300401 indicates HF channel 401
If the HM is 4 it indicates a HF radio but a frequency using 7 digits in multiples of 10Hz. I presume this must mean that the message must be 21 bytes, instead of the 19 bytes generated by VHF radios. Or perhaps my assumption about the HF/MF encoding is totally wrong. It would be great to get some captures from NMEA 2000 equipped HF/MF radios.
If the HM is 9, it is a VHF radio. TM should be zero, if M = 1 the ships transmit frequency is used as the simplex channel, if M =2 the shore stations transmit frequency is used as the simplex channel and if M = 0 it uses the duplex frequencies described in ITU Radio Regulations Appendix 18. (Eg. Channel 16 uses156.8 MHz) . H, T, U indicate the VHF channel number, with leading zeroes. 

I've always assumed the mode maps to the following which is described for the NMEA 0183 FSI sentence.
F3E/G3E simplex, telephone 
F3E/G3E duplex, telephone 
J3E, telephone 
H3E, telephone 
F1B/J2B FEC NBDP, Telex/teleprinter
F1B/J2B ARQ NBDP, Telex/teleprinter
F1B/J2B receive only, teleprinter/DSC | 
A1A Morse, tape recorder 
A1A Morse, Morse key/head set
F1C/F2C/F3C, FAX-machine

The three alphanumeric characters (eg. F3E) are defined in ITU Radio Regulations Appendix 1.

commit 3434201e (2024-04-15): Fix #442
commit 34a5e3ff (2024-04-15): Fix #442