canboat/canboat #444: Enhancment,  Output in Comma Separated Variable (CSV) format

TwoCanPlugIn, 2024-07-02:
I use my own tool set for capturing & analysing traffic in a number of formats,

One of the methods I use is to output my data in a csv format which I can then feed into an Excel spreadsheet or a SQL database.

Example csv header could be:
timestamp, source, destination, prioroty, pgn, raw data bytes, text, where text is human readable, could even just be canboat's json output) 

Using either, it makes it easy to filter out the noise, In excel just use the standard filters available for columns or in SQL using a query such as select pgn, text from table where pgn = 130850 or pgn = 130851 and source = 3 or source = 4 order by timestamp

If I use canboat, It means I have to process the json output into a format suitable for importing into excel or sql.

keesverruijt, 2024-08-02:
I'm not in favour of adding extra output formats to analyzer, it is complicated enough. What would be simple is to add a JSON output format that has all the data you need, e.g. always generate output even if the analyzer doesn't have data, and add a "raw bytes" field. Then a simple perl / php / whatever script will do what you want.

keesverruijt, 2024-08-02:
@TwoCanPlugIn , you can use the filter and new option added (`-debugdata`) like this:

    analyzer < samples/merrimac-actisense-serial-2011.raw -json -debugdata | ./analyzer2csv/analyzer2csv

or if you want to override the columns:

    analyzer < samples/merrimac-actisense-serial-2011.raw -json -debugdata | ./analyzer2csv/analyzer2csv timestamp src pgn data

closed 2024-12-21 by keesverruijt