canboat/canboat #445: Add PGN priorities

TwoCanPlugIn, 2024-07-02:
There are now enough samples to derive the priority values for most of the PGN's listed in canboat.

Given that most of the data is captured from equipment manufactured by commercial vendors, it should be safe to assume that they adhere to the standards and use the correct priorities when transmitting messages onto the network.

arks-public, 2024-07-31:
I agree that canboat should ideally include the standard priority values for the PGNs where priority is known.

Here is the code snippet that I've been using in the meantime, perhaps this could be used as a starting point. We only deal with like 20 PGNs

```
switch(pgn) {
//case 59904                    : priority = 6; break; // per Maretron USB100
case ISOAddressClaim            : priority = 6; break; // per NMEA - https://www.nmea.org/Assets/20140710%20nmea-2000-060928%20iso%20address%20claim%20pgn%20corrigendum.pdf
case PGNList                    : priority = 6; break; // per https://github.com/ttlappalainen/NMEA2000/blob/master/src/NMEA2000.cpp
case SystemTime                 : priority = 3; break; // per Airmar GPS
case Heartbeat                  : priority = 7; break; // per NMEA - https://www.nmea.org/Assets/20140102%20nmea-2000-126993%20heartbeat%20pgn%20corrigendum.pdf
case ProductInformation         : priority = 6; break; // per Airmar GPS and Maretron USB100
case ConfigurationInformation   : priority = 6; break; // per Airmar - https://www.manualslib.com/manual/629148/Airmar-Pb200.html?page=44
case Attitude                   : priority = 3; break; // per Airmar GPS
case MagneticVariation          : priority = 7; break; // per Airmar GPS
case BatteryStatus              : priority = 6; break; // per https://github.com/ttlappalainen/NMEA2000/blob/master/src/N2kMessages.cpp
case BatteryConfigurationStatus : priority = 6; break; // per https://github.com/ttlappalainen/NMEA2000/blob/master/src/N2kMessages.cpp
//case 128275                   : priority = 6; break; // per NMEA - https://www.nmea.org/Assets/nmea-2000-digital-interface-white-paper.pdf
case PositionRapidUpdate        : priority = 2; break; // per Airmar GPS
case COGSOGRapidUpdate          : priority = 2; break; // per Airmar GPS
case GNSSPositionData           : priority = 3; break; // per Airmar GPS
case TimeDate                   : priority = 3; break; // per Airmar GPS
case Datum                      : priority = 6; break; // per Airmar GPS
case CrossTrackError            : priority = 3; break; // per https://github.com/ttlappalainen/NMEA2000/blob/master/src/NMEA2000.cpp
case GNSSDOPs                   : priority = 6; break; // per Airmar GPS
case GNSSSatellitesInView       : priority = 6; break; // per Airmar GPS
case GPSAlmanacData             : priority = 6; break; // per https://github.com/ttlappalainen/NMEA2000/blob/master/src/NMEA2000.cpp
case TemperatureExtendedRange   : priority = 5; break; // per NMEA - https://www.nmea.org/Assets/nmea%202000%20pgn%20130316%20corrigenda%20nmd%20version%202.100%20feb%202015.pdf
case SupportedZoneData          : priority = 6; break; // per NMEA - https://www.nmea.org/Assets/20160715%20corrigenda%20entertainment%20pgns%20.pdf
}
```

keesverruijt, 2024-08-02:
Yes, I agree, we should add this. "Observed in the field" is as good as anything I guess.

keesverruijt, 2024-08-02:
Many priorities have been added, generated by just taking the lowest priority seen in the samples files. Fixed in v5.1.0.

closed 2024-08-02 by keesverruijt