canboat/canboat #446: PGN 130824 is not decoded properly in a presence of other PGNs

sergei, 2024-07-30:
I'm trying to decode the PGN 130824, but for some reasons the analyzer only understands the very first message in the log file. 
However if I leave PGN 130824 only in the input stream then all messages are processed properly. I attached the log in question.

Here is bad output:
```text
$ cat all-pgns.log | rel/darwin-arm64/analyzer | grep 130824 | head

INFO 2024-07-30T18:56:34.520Z [analyzer] Assuming normal format with one line per frame
2024-07-19-23:24:46.829 3  60 255 130824 B&G: key-value data:  Manufacturer Code = B & G; Industry Code = Marine Industry; Key 1 = Roll Rate; Length 1 = 4; Value 1 = -846.5 deg; Key 2 = Pitch Angle; Length 2 = 4; Value 2 = -903.9 deg; Key 3 = 68; Length 3 = 4; Value 3 = 86 44 F3 FF
INFO 2024-07-30T18:56:34.520Z [analyzer] Detected normal format with all frames on one line
2024-07-19-23:24:46.913 3  27 255 130824 0x1FF00-0x1FFFF: Manufacturer Specific fast-packet non-addressed:  Data = 20 2A 7D 99 7E 20 CB 00
2024-07-19-23:24:46.914 3  27 255 130824 0x1FF00-0x1FFFF: Manufacturer Specific fast-packet non-addressed:  Data = 21 7C 20 00 05 32 21 A8
2024-07-19-23:24:46.914 3  27 255 130824 0x1FF00-0x1FFFF: Manufacturer Specific fast-packet non-addressed:  Data = 22 46 33 21 B5 4D 1D 21
```

And much better one if only 130824 is left in inout stream:
```text
$ cat all-pgns.log | grep 130824 | rel/darwin-arm64/analyzer | grep 130824 | head

INFO 2024-07-30T18:56:58.263Z [analyzer] Assuming normal format with one line per frame
ERROR 2024-07-30T18:56:58.263Z [analyzer] Received incomplete fast packet PGN 130824 from source 27
2024-07-19-23:24:46.829 3  60 255 130824 B&G: key-value data:  Manufacturer Code = B & G; Industry Code = Marine Industry; Key 1 = Roll Rate; Length 1 = 4; Value 1 = -846.5 deg; Key 2 = Pitch Angle; Length 2 = 4; Value 2 = -903.9 deg; Key 3 = 68; Length 3 = 4; Value 3 = 86 44 F3 FF
2024-07-19-23:24:46.931 3  60 255 130824 B&G: key-value data:  Manufacturer Code = B & G; Industry Code = Marine Industry; Key 1 = Roll Rate; Length 1 = 4; Value 1 = -873.2 deg; Key 2 = Pitch Angle; Length 2 = 4; Value 2 = 314.9 deg; Key 3 = 68; Length 3 = 4; Value 3 = 42 7F F3 FF
2024-07-19-23:24:47.029 3  60 255 130824 B&G: key-value data:  Manufacturer Code = B & G; Industry Code = Marine Industry; Key 1 = Roll Rate; Length 1 = 4; Value 1 = -909.1 deg; Key 2 = Pitch Angle; Length 2 = 4; Value 2 = 1538.6 deg; Key 3 = 68; Length 3 = 4; Value 3 = D0 29 F5 FF
2024-07-19-23:24:47.036 7  15 255 130824 B&G: key-value data:  Manufacturer Code = B & G; Industry Code = Marine Industry; Key 1 = Active Perf Mode; Length 1 = 1; Value 1 = 3; Key 2 = Gust Bear Away; Length 2 = 2; Value 2 = Unknown; Key 3 = TWS Bear Away; Length 3 = 2; Value 3 = Unknown; Key 4 = Heel Compensation; Length 4 = 2; Value 4 = Unknown; Key 5 = Pilot Net Course; Length 5 = 2; Value 5 = -0.0 deg; Key 6 = Pilot Target Wind Angle; Length 6 = 2; Value 6 = Unknown; Key 7 = Pilot Weather Helm; Length 7 = 2; Value 7 = -5.3 deg; Key 8 = Pilot Mean Heel; Length 8 = 2; Value 8 = 6.7 deg
```

[all-pgns.log](https://github.com/user-attachments/files/16432248/all-pgns.log)

sergei, 2024-07-30:
It seems that `enum MultiPackets multiPackets` of analyzer.c can get  stuck in   `multiPackets == MULTIPACKETS_COALESCED` state if I don't specify explicitly `-format PLAIN_OR_FAST`. In this case for every message it initially set to MULTIPACKETS_SEPARATE and then might be set to MULTIPACKETS_COALESCED. 

Don't have enough knowledge to suggest the proper fix. 
This change worked for me:
``` 
 while (fgets(msg, sizeof(msg) - 1, file))
  {
    multiPackets = MULTIPACKETS_SEPARATE;
```

keesverruijt, 2024-08-02:
The problem is that your log file contains both coalesced and separated instances of fast PGNs. This is a bug in your converter: it doesn't realise that _all_ PGNs in the proprietary range 0x1FF00 to 0x1FFFF are always FAST pgns. 

I've added a `-format PLAIN_MIX_FAST` option which will switch back and forth as necessary, but this option will fail to analyse some existing PGNs that are FAST but not longer than 8 bytes, for instance 127509.

Fixed in #448 released in v5.1.0.

sergei, 2024-08-03:
Thanks, I sent the e-mail to YDVR support bringing their attention to this issue.

closed 2024-08-03 by sergei