canboat/canboat #451: PGN 127489 Fast Packet Decode Doesn't Work

ba05, 2024-08-06:
When I run:
`candump2analyzer.exe "127489.log" | analyzer`
on the below file, the fast packet decodign doesnt seem to work. I only get the first couple fields. These fields are then repeated and have the wrong values.:

[https://github.com/canboat/canboat/blob/ad8f22e75204d79f0c14411e27fa730811f11539/analyzer/pgns.json#L13889](https://github.com/canboat/canboat/blob/ad8f22e75204d79f0c14411e27fa730811f11539/analyzer/pgns.json#L13889)

```
1970-01-01-00:00:00.352 2   1 255 127489 Engine Parameters, Dynamic:  Instance = 96; Oil pressure = 0.026 bar; Oil temperature = -101.25 C; Temperature = -238.84 C; Alternator Potential = 28.27 V
1970-01-01-00:00:00.353 2   1 255 127489 Engine Parameters, Dynamic:  Instance = 97; Oil pressure = 47.750 bar; Oil temperature = 700.05 C; Temperature = -232.19 C; Alternator Potential = 244.15 V
1970-01-01-00:00:00.354 2   1 255 127489 Engine Parameters, Dynamic:  Instance = 98; Oil pressure = 0.009 bar; Oil temperature = -273.15 C; Temperature = -266.60 C; Alternator Potential = 326.39 V
1970-01-01-00:00:00.355 2   1 255 127489 Engine Parameters, Dynamic:  Instance = 99; Oil pressure = 0.000 bar; Oil temperature = -273.15 C; Temperature = 53.24 C; Alternator Potential = -0.01 V
1970-01-01-00:00:00.451 2   1 255 127489 Engine Parameters, Dynamic:  Instance = Single Engine or Dual Engine Port; Oil pressure = 0.282 bar; Oil temperature = -81.25 C; Temperature = -238.84 C; Alternator Potential = 172.19 V
1970-01-01-00:00:00.452 2   1 255 127489 Engine Parameters, Dynamic:  Instance = Dual Engine Starboard; Oil pressure = 45.189 bar; Oil temperature = 623.25 C; Temperature = -119.55 C; Alternator Potential = -138.79 V
1970-01-01-00:00:00.453 2   1 255 127489 Engine Parameters, Dynamic:  Instance = 2; Oil pressure = 0.009 bar; Oil temperature = -273.15 C; Temperature = -266.60 C; Alternator Potential = 326.39 V
1970-01-01-00:00:00.454 2   1 255 127489 Engine Parameters, Dynamic:  Instance = 3; Oil pressure = 0.000 bar; Oil temperature = -273.15 C; Temperature = 53.24 C; Alternator Potential = -0.01 V
1970-01-01-00:00:00.856 2   1 255 127489 Engine Parameters, Dynamic:  Instance = 128; Oil pressure = 0.026 bar; Oil temperature = -101.25 C; Temperature = -238.84 C; Alternator Potential = 28.27 V
1970-01-01-00:00:00.857 2   1 255 127489 Engine Parameters, Dynamic:  Instance = 129; Oil pressure = 47.750 bar; Oil temperature = 648.85 C; Temperature = -232.19 C; Alternator Potential = 244.15 V
1970-01-01-00:00:00.861 2   1 255 127489 Engine Parameters, Dynamic:  Instance = 130; Oil pressure = 0.009 bar; Oil temperature = -273.15 C; Temperature = -266.60 C; Alternator Potential = 326.39 V
1970-01-01-00:00:00.862 2   1 255 127489 Engine Parameters, Dynamic:  Instance = 131; Oil pressure = 0.000 bar; Oil temperature = -273.15 C; Temperature = 53.24 C; Alternator Potential = -0.01 V
1970-01-01-00:00:00.951 2   1 255 127489 Engine Parameters, Dynamic:  Instance = 32; Oil pressure = 0.282 bar; Oil temperature = -81.25 C; Temperature = -238.84 C; Alternator Potential = 172.19 V
1970-01-01-00:00:00.952 2   1 255 127489 Engine Parameters, Dynamic:  Instance = 33; Oil pressure = 45.189 bar; Oil temperature = 572.05 C; Temperature = -119.55 C; Alternator Potential = -138.79 V
1970-01-01-00:00:00.953 2   1 255 127489 Engine Parameters, Dynamic:  Instance = 34; Oil pressure = 0.009 bar; Oil temperature = -273.15 C; Temperature = -266.60 C; Alternator Potential = 326.39 V
1970-01-01-00:00:00.954 2   1 255 127489 Engine Parameters, Dynamic:  Instance = 35; Oil pressure = 0.000 bar; Oil temperature = -273.15 C; Temperature = 53.24 C; Alternator Potential = -0.01 V
1970-01-01-00:00:01.404 2   1 255 127489 Engine Parameters, Dynamic:  Instance = 160; Oil pressure = 0.026 bar; Oil temperature = -113.15 C; Temperature = -238.84 C; Alternator Potential = 28.27 V
1970-01-01-00:00:01.405 2   1 255 127489 Engine Parameters, Dynamic:  Instance = 161; Oil pressure = 46.470 bar; Oil temperature = 623.25 C; Temperature = -232.19 C; Alternator Potential = 244.15 V
1970-01-01-00:00:01.406 2   1 255 127489 Engine Parameters, Dynamic:  Instance = 162; Oil pressure = 0.009 bar; Oil temperature = -273.15 C; Temperature = -266.60 C; Alternator Potential = 326.39 V
1970-01-01-00:00:01.407 2   1 255 127489 Engine Parameters, Dynamic:  Instance = 163; Oil pressure = 0.000 bar; Oil temperature = -273.15 C; Temperature = 53.24 C; Alternator Potential = -0.01 V
1970-01-01-00:00:01.501 2   1 255 127489 Engine Parameters, Dynamic:  Instance = 64; Oil pressure = 0.282 bar; Oil temperature = -101.25 C; Temperature = -238.84 C; Alternator Potential = 172.19 V
1970-01-01-00:00:01.502 2   1 255 127489 Engine Parameters, Dynamic:  Instance = 65; Oil pressure = 45.189 bar; Oil temperature = 546.45 C; Temperature = -119.55 C; Alternator Potential = -138.79 V
1970-01-01-00:00:01.502 2   1 255 127489 Engine Parameters, Dynamic:  Instance = 66; Oil pressure = 0.009 bar; Oil temperature = -273.15 C; Temperature = -266.60 C; Alternator Potential = 326.39 V
```

127489.log contents:
```
(0000000000.352000) vcan0 09F20101#601A00B706670D0B
(0000000000.353000) vcan0 09F20101#6186BA042600105F
(0000000000.354000) vcan0 09F20101#62090000008F027F
(0000000000.355000) vcan0 09F20101#63000000007F7FFF
(0000000000.451000) vcan0 09F20101#001A017F07670D43
(0000000000.452000) vcan0 09F20101#0185B00423003CC9
(0000000000.453000) vcan0 09F20101#02090000008F027F
(0000000000.454000) vcan0 09F20101#03000000007F7FFF
(0000000000.856000) vcan0 09F20101#801A00B706670D0B
(0000000000.857000) vcan0 09F20101#8186BA042400105F
(0000000000.861000) vcan0 09F20101#82090000008F027F
(0000000000.862000) vcan0 09F20101#83000000007F7FFF
(0000000000.951000) vcan0 09F20101#201A017F07670D43
(0000000000.952000) vcan0 09F20101#2185B00421003CC9
(0000000000.953000) vcan0 09F20101#22090000008F027F
(0000000000.954000) vcan0 09F20101#23000000007F7FFF
(0000000001.404000) vcan0 09F20101#A01A004006670D0B
(0000000001.405000) vcan0 09F20101#A186B5042300105F
(0000000001.406000) vcan0 09F20101#A2090000008F027F
(0000000001.407000) vcan0 09F20101#A3000000007F7FFF
(0000000001.501000) vcan0 09F20101#401A01B706670D43
(0000000001.502000) vcan0 09F20101#4185B00420003CC9
(0000000001.502000) vcan0 09F20101#42090000008F027F
```

ba05, 2024-08-07:
Upon further inspection: I believe this is because `candump2analyzer` does not recognize NMEA 2000 fast packet pgns (pgns where data is spread over multiple lines, where the first byte of each data payload is the sequential page byte.). `analyzer` wants all the data bytes to be on a single line. Ideally `candump2analyzer` would read the `Type` field defined in `pgns.json` for each PGN, and if `Type == Fast` handle appropriately. End result should be like [https://github.com/canboat/canboat/blob/ad8f22e75204d79f0c14411e27fa730811f11539/samples/pgn127489.txt](https://github.com/canboat/canboat/blob/ad8f22e75204d79f0c14411e27fa730811f11539/samples/pgn127489.txt)

sbender9, 2024-08-07:
Unless there is a recent bug. It definitely supports fast packets.

sbender9, 2024-08-07:
How did you get that log file?

ba05, 2024-08-07:
Log file was recorded with a [https://www.csselectronics.com/products/can-bus-logger-interface-cl2000](https://www.csselectronics.com/products/can-bus-logger-interface-cl2000)

Almost like analyzer is not recognizing the PGN as a fast packet.

Issues seems related to https://github.com/canboat/canboat/issues/446 and https://github.com/canboat/canboat/issues/448

sbender9, 2024-08-07:
Here's what I get when I run it:

```
$ ./rel/darwin-arm64/candump2analyzer < /tmp/cdump.log  | ./rel/darwin-arm64/analyzer
INFO 2024-08-07T23:15:26.282Z [analyzer] N2K packet analyzer
CANboat version v5.1.0

(C) 2009-2023, Kees Verruijt, Harlingen, The Netherlands.
For more information see https://github.com/canboat/canboat

This file is part of CANboat.

Licensed under the Apache License, Version 2.0 (the "License");
you may not use this file except in compliance with the License.
You may obtain a copy of the License at

    http://www.apache.org/licenses/LICENSE-2.0

Unless required by applicable law or agreed to in writing, software
distributed under the License is distributed on an "AS IS" BASIS,
WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
See the License for the specific language governing permissions and
limitations under the License.

INFO 2024-08-07T23:15:26.287Z [analyzer] Assuming PLAIN_OR_FAST format with one line per frame or one line per message
1970-01-01-00:00:00.355 2   1 255 127489 Engine Parameters, Dynamic:  Instance = Single Engine or Dual Engine Port; Oil pressure = 1.719 bar; Oil temperature = 69.95 C; Temperature = 70.00 C; Alternator Potential = 12.10 V; Fuel Rate = 3.8 L/h; Total Engine hours = 170:36:00; Coolant Pressure = 0.000 bar; Fuel Pressure = 6.550 bar; Reserved = 7F; Discrete Status 1 = None; Discrete Status 2 = None; Engine Load = Unknown; Engine Torque = Unknown
1970-01-01-00:00:00.454 2   1 255 127489 Engine Parameters, Dynamic:  Instance = Dual Engine Starboard; Oil pressure = 1.919 bar; Oil temperature = 69.95 C; Temperature = 68.00 C; Alternator Potential = 12.00 V; Fuel Rate = 3.5 L/h; Total Engine hours = 178:09:00; Coolant Pressure = 0.000 bar; Fuel Pressure = 6.550 bar; Reserved = 7F; Discrete Status 1 = None; Discrete Status 2 = None; Engine Load = Unknown; Engine Torque = Unknown
1970-01-01-00:00:00.862 2   1 255 127489 Engine Parameters, Dynamic:  Instance = Single Engine or Dual Engine Port; Oil pressure = 1.719 bar; Oil temperature = 69.95 C; Temperature = 70.00 C; Alternator Potential = 12.10 V; Fuel Rate = 3.6 L/h; Total Engine hours = 170:36:00; Coolant Pressure = 0.000 bar; Fuel Pressure = 6.550 bar; Reserved = 7F; Discrete Status 1 = None; Discrete Status 2 = None; Engine Load = Unknown; Engine Torque = Unknown
1970-01-01-00:00:00.954 2   1 255 127489 Engine Parameters, Dynamic:  Instance = Dual Engine Starboard; Oil pressure = 1.919 bar; Oil temperature = 69.95 C; Temperature = 68.00 C; Alternator Potential = 12.00 V; Fuel Rate = 3.3 L/h; Total Engine hours = 178:09:00; Coolant Pressure = 0.000 bar; Fuel Pressure = 6.550 bar; Reserved = 7F; Discrete Status 1 = None; Discrete Status 2 = None; Engine Load = Unknown; Engine Torque = Unknown
1970-01-01-00:00:01.407 2   1 255 127489 Engine Parameters, Dynamic:  Instance = Single Engine or Dual Engine Port; Oil pressure = 1.600 bar; Oil temperature = 69.95 C; Temperature = 70.00 C; Alternator Potential = 12.05 V; Fuel Rate = 3.5 L/h; Total Engine hours = 170:36:00; Coolant Pressure = 0.000 bar; Fuel Pressure = 6.550 bar; Reserved = 7F; Discrete Status 1 = None; Discrete Status 2 = None; Engine Load = Unknown; Engine Torque = Unknown
```

sbender9, 2024-08-07:
This could actually be an issue with the windows build?

sbender9, 2024-08-07:
candump2analyzer.exe output should look like?

```
1970-01-01-00:00:00.352,2,127489,1,255,8,60,1a,00,b7,06,67,0d,0b
1970-01-01-00:00:00.353,2,127489,1,255,8,61,86,ba,04,26,00,10,5f
1970-01-01-00:00:00.354,2,127489,1,255,8,62,09,00,00,00,8f,02,7f
1970-01-01-00:00:00.355,2,127489,1,255,8,63,00,00,00,00,7f,7f,ff
1970-01-01-00:00:00.451,2,127489,1,255,8,00,1a,01,7f,07,67,0d,43
1970-01-01-00:00:00.452,2,127489,1,255,8,01,85,b0,04,23,00,3c,c9
1970-01-01-00:00:00.453,2,127489,1,255,8,02,09,00,00,00,8f,02,7f
1970-01-01-00:00:00.454,2,127489,1,255,8,03,00,00,00,00,7f,7f,ff
1970-01-01-00:00:00.856,2,127489,1,255,8,80,1a,00,b7,06,67,0d,0b
1970-01-01-00:00:00.857,2,127489,1,255,8,81,86,ba,04,24,00,10,5f
1970-01-01-00:00:00.861,2,127489,1,255,8,82,09,00,00,00,8f,02,7f
1970-01-01-00:00:00.862,2,127489,1,255,8,83,00,00,00,00,7f,7f,ff
1970-01-01-00:00:00.951,2,127489,1,255,8,20,1a,01,7f,07,67,0d,43
1970-01-01-00:00:00.952,2,127489,1,255,8,21,85,b0,04,21,00,3c,c9
1970-01-01-00:00:00.953,2,127489,1,255,8,22,09,00,00,00,8f,02,7f
1970-01-01-00:00:00.954,2,127489,1,255,8,23,00,00,00,00,7f,7f,ff
1970-01-01-00:00:01.404,2,127489,1,255,8,a0,1a,00,40,06,67,0d,0b
1970-01-01-00:00:01.405,2,127489,1,255,8,a1,86,b5,04,23,00,10,5f
1970-01-01-00:00:01.406,2,127489,1,255,8,a2,09,00,00,00,8f,02,7f
1970-01-01-00:00:01.407,2,127489,1,255,8,a3,00,00,00,00,7f,7f,ff
1970-01-01-00:00:01.501,2,127489,1,255,8,40,1a,01,b7,06,67,0d,43
1970-01-01-00:00:01.502,2,127489,1,255,8,41,85,b0,04,20,00,3c,c9
1970-01-01-00:00:01.502,2,127489,1,255,8,42,09,00,00,00,8f,02,7f
```

ba05, 2024-08-07:
I determined the root cause. The file I have has Windows 'CRLF' line endings. `analyzer` only works correctly (at least with this fast packet PGN) with UNIX 'LF' line endings. I am using the windows build. 

Could this be classified as a bug?

sbender9, 2024-08-07:
Up to @keesverruijt ...

commit f121e9b6 (2024-08-08): Fix #451
closed 2024-08-08 by keesverruijt