canboat/canboat #453: NGX-1 USB Support

Ben93kie, 2024-08-27:
Hi guys, nice repo, thx!

Do you happen to support the NGX-1 as well or just the NGT-1?

NGX-1
https://productimageserver.com/literature/brochure/102318BR.pdf

Thanks!

keesverruijt, 2024-08-28:
Unknown at this time, but given that the software config product "Toolkit" works with both and looks familiar, probably.

It looks like a simple rebuild with newer hardware as the NGT-1 is quite old now, > 10 years.

Ben93kie, 2024-09-02:
Thanks!
Just want to add the note that 

"the NGT-1-USB is an end of life product and has been replaced with the NGX-USB" 

(according to Actisense). 

I don't have one right now, but will give an update here whenver I do have, thx!

Ben93kie, 2024-09-11:
Actisense team confirmed, it's not going to be changed 👍

Ben93kie, 2024-09-13:
Tested it and seems to be working the same

Ben93kie, 2024-09-13:
I don't really. If I find a free minute, I might. 
But, rest assured, actisense dev team said nothing is changed software wise. Only the USB-friendly name is different.

sbender9, 2024-09-13:
No worries! Chances are that it will work fine if it works with canboat. Just would like confirmation before I add it to the list of supported devices.